zephyrproject-rtos/zephyr #46414: mcuboot: rt1060: confirmed image causes usage fault

bbilas, 2022-06-09:
**Describe the bug**
The bootloader encounters a usage fault when a signed confirmed image is flashed. A signed unconfirmed image seems to work well:

**To Reproduce**
Steps to reproduce the behavior:
1.` west build -p -b mimxrt1060_evk -d build-mcuboot bootloader/mcuboot/boot/zephyr`
2. `west flash -d build-mcuboot`
3.` west build -p -b mimxrt1060_evk -d build-hello-world zephyr/samples/hello_world/ -- -DCONFIG_BOOTLOADER_MCUBOOT=y -DCONFIG_MCUBOOT_GENERATE_CONFIRMED_IMAGE=y -DCONFIG_BUILD_OUTPUT_HEX=y -DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE=\"bootloader/mcuboot/root-rsa-2048.pem\"`
4. `west flash -d build-hello-world --skip-rebuild --hex-file build-hello-world/zephyr/zephyr.signed.confirmed.hex`

**Expected behavior**
The bootloader should load confirmed and unconfirmed images.

**Impact**
Unable to use confirmed images in a production scenario.

**Logs and console output**
Confirmed image:

<pre>
*** Booting Zephyr OS build zephyr-v3.1.0  ***
I: Starting bootloader
I: Primary image: magic=good, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: primary slot
I: Starting swap using scratch algorithm.
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x88000000
E: r0/a1:  0x0000002a  r1/a2:  0x00fff439  r2/a3:  0x00fff439
E: r3/a4:  0x87fffffc r12/ip:  0xbb8d9a02 r14/lr:  0x568046a8
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x600043b6
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x80000138 (unknown)
E: Halting system
</pre>

Unconfirmed image:

<pre>
*** Booting Zephyr OS build zephyr-v3.1.0  ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Scratch: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: primary slot
I: Swap type: none
I: Bootloader chainload address offset: 0x10000
I: Jumping to the first image slot
*** Booting Zephyr OS build zephyr-v3.1.0  ***
Hello World! mimxrt1060_evk
</pre>

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.13.1
 - V3.1.0 (2ddd73feafd3316af2c547c34d6969bea637d5c6)

Laczen, 2022-06-10:
@bbilas, shoildn't this be reported in as an mcuboot issue? Or is it a zephyr issue?

bbilas, 2022-06-13:
> @bbilas, shoildn't this be reported as an mcuboot issue? Or is it a zephyr issue?

mcuboot seems to work well itself, IMO. there is an issue with a confirmed image generated by Zephyr.

Laczen, 2022-06-13:
> > @bbilas, shoildn't this be reported as an mcuboot issue? Or is it a zephyr issue?
> 
> mcuboot seems to work well itself, IMO. there is an issue with a confirmed image generated by Zephyr.

@bbilas, as the generation of the confirmed image is calling imgtool (part of mcuboot) the problem would either be in the options passed to imgtool or the device just isn't supported (e.g. to large alignment requirement) by mcuboot.

bbilas, 2022-06-13:
> @bbilas, as the generation of the confirmed image is calling imgtool (part of mcuboot) the problem would either be in the options passed to imgtool or the device just isn't supported (e.g. to large [...]

Let's wait a while to see what others think about that. Personally I have no objections to moving that to the MCUBoot repository.

DerekSnell, 2022-06-15:
Hi @bbilas ,

We will test this further, but I wanted to share I followed your steps, and I did not see this issue.  The confirmed image booted just fine for me, same as the unconfirmed image.  I used Zephyr v3.1.0 with your steps.  I did build it and flash from Windows, not Linux, so we can also try that.  I also used the JLink runner to program the flash.

I did want to ask, because I have seen you have a custom RT1060 board.  Are you seeing this issue on NXP's EVK?  Or only on your board?  And for the EVK, can you confirm the full part number printed on the PCB?  I don't see why the difference would matter for this, but there is a newer EVK in production now, so your part number will end as -EVK or -EVKB.

Thanks

DerekSnell, 2022-06-15:
Hi @bbilas ,

A quick update, we found the newer version of imgtool is causing this issue.  I was originally using v1.7.2 which worked fine, both on Windows and Ubuntu.  But upgrading to the latest v1.9.0 causes the issue you see.

The issue looks to be in the padding and trailer that imgtool adds to the end of the image slot.  Now that we can replicate this, I'm going to try manipulating the hex files to better show the issue.

Thanks

bbilas, 2022-06-15:
Hi @DerekSnell,
I see the issue on my custom board and EVK board too. I've been using imgtool v1.8.0 then I tried to downgrade to the v1.7.2 but the following error occurs: 

`FATAL ERROR: DT zephyr,flash node has no write-block-size; can't determine imgtool write alignment
`
when I build the `hello-world` sample with MCUBoot enabled.

OS: Arch Linux

EVK details:
`700-47858 REV X1` `SCH-47858 REV B1` `MIMXRT1060-EVKB`

bbilas, 2022-06-15:
I've applied that workaround to our build machines to use imgtool v1.7.2 and at the first look our custom board works well using a confirmed image as well as unconfirmed.

@d3zd3z @nvlsianpu Would you mind looking at it?

mmahadevan108, 2022-07-18:
@bbilas, can this issue be closed?

bbilas, 2022-07-18:
> @bbilas, can this issue be closed?

We have created an issue in the mcuboot repository nevertheless the root issue is not found yet. I don't know what is the right way to manage such an issue but it seems that it's related to the mcuboot rather than zephyr itself so IMO it can be closed.

dleach02, 2022-08-17:
Issue moved to https://github.com/mcu-tools/mcuboot/issues/1390

closed 2022-08-17 by dleach02