zephyrproject-rtos/zephyr #46446: lvgl: Using sw_rotate with SSD1306 shield causes memory fault

richesonk, 2022-06-10:
**Describe the bug**
Using the new `sw_rotate` feature from LVGL with the `ssd1306_128x64` shield causes a memory fault.

The issue seems to be related to the `draw_buf_rotate...()` functions not accounting for the possibility that a custom `set_px_cb` (such as `lvgl_set_px_cb_mono()` function maybe packing the data as bits instead of full bytes. This results in out of bounds array accesses in the rotation functions and causes data corruption in other (likely display related) areas of memory.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit modules/lib/gui/lvgl/zephyr/lvgl.c adding the following lines immediately after the call to `lv_disp_drv_init(&disp_drv);` in `lvgl_init()`:
```c
	disp_drv.sw_rotate = 1;
	disp_drv.rotated = LV_DISP_ROT_90;
```
2. Build the LVGL display sample for a supported board that also supports the `ssd1306_128x64` shield. I am using a Nordic development kit and the Zephyr based nRF Connect SDK, so my build command is:
```
west build zephyr/samples/subsys/display/lvgl -b nrf5340dk_nrf5340_cpuapp  -- -DSHIELD=ssd1306_128x64
```
3. Flash the built code to the board (with and SSD1306 shield attached)
4. Observer the failure (memfault shown on the console output, or obvious rebooting causing display to flash)

**Expected behavior**
Normal display of the LVGL display sample rotated on the display by 90 degrees (a.k.a. portrait mode instead of landscape)

**Impact**
This is a major impact to my current project as we need to use the display in portrait mode and the SSD1306 display controller does not seem to support 90 degree rotation in HW.

**Logs and console output**
No real information to capture/add here. It's a standard memfault report in the console output. The actual fault happens for me on exit from `draw_buf_rotate_90_sqr()` but is due to the data corruption caused during the execution of that function.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: Zephyr SDK
 - v3.0.99-ncs1 (via the nRF Connect SDK from Nordic)

**Additional context**
Needed to create an samples/subsys/display/lvgl/boards/nrf5340dk_nrf5340_cpuapp.overlay file with the following snippet for the sample to work with the nRF5340 dev kit.
```
&arduino_i2c {
	status = "okay";
	clock-frequency = <I2C_BITRATE_FAST>;
	zephyr,concat-buf-size = <1025>;
};
```

fabiobaltieri, 2022-06-15:
Can you clarify if this is a Zephyr specific issue? From how you describe it it seems to be all contained in the LVGL module.

brgl, 2022-06-15:
@kisvegabor could you take a look? You helped me with `set_px_cb` before so maybe it rings a bell for you.

kisvegabor, 2022-06-15:
This issue is really related to LVGL. @richesonk please open the same issue in [LVGL's repo](https://github.com/lvgl/lvgl/issues) to track it in the right place. (Feel free to just copy paste your post)

> The issue seems to be related to the draw_buf_rotate...() functions not accounting for the possibility that a custom set_px_cb (such as lvgl_set_px_cb_mono() function maybe packing the data as bits [...]

Probably you are right, and this is the root of the problem.

richesonk, 2022-06-15:
> This issue is really related to LVGL. @richesonk please open the same issue in [LVGL's repo](https://github.com/lvgl/lvgl/issues) to track it in the right place. (Feel free to just copy paste your post)

Done. I have opened https://github.com/lvgl/lvgl/issues/3414 in the LVGL repo. Shall we close this issue, or leave it open to track picking up any changes/fixes from LVGL?

brgl, 2022-06-15:
@richesonk the fix will have to make its way into the zephyr fork too so let's keep it open until that happens.

richesonk, 2022-06-22:
@brgl I posted a patch in the linked LVGL issue that implements rotation support as recommended by @kisvegabor this patch to Zephyr enables the SW rotation support by default for the SSD1306 and adds an overlay file for building the LVGL sample on the nRF5340 dev kit.
```
diff --git a/boards/shields/ssd1306/Kconfig.defconfig b/boards/shields/ssd1306/Kconfig.defconfig
index fb71784857..4e00e2d432 100644
--- a/boards/shields/ssd1306/Kconfig.defconfig
+++ b/boards/shields/ssd1306/Kconfig.defconfig
@@ -42,6 +42,9 @@ choice LV_COLOR_DEPTH
 	default LV_COLOR_DEPTH_1
 endchoice
 
+config LV_DISP_SW_ROTATE
+	default y
+
 endif # LVGL
 
 endif # DISPLAY
diff --git a/samples/subsys/display/lvgl/boards/nrf5340dk_nrf5340_cpuapp.overlay b/samples/subsys/display/lvgl/boards/nrf5340dk_nrf5340_cpuapp.overlay
new file mode 100644
index 0000000000..20f3c5eafc
--- /dev/null
+++ b/samples/subsys/display/lvgl/boards/nrf5340dk_nrf5340_cpuapp.overlay
@@ -0,0 +1,5 @@
+&arduino_i2c {
+	status = "okay";
+	clock-frequency = <I2C_BITRATE_FAST>;
+	zephyr,concat-buf-size = <1025>;
+};
```
This along with the LVGL patch should enable the LVGL sample to be built for rotation using
`west build zephyr/samples/subsys/display/lvgl -b nrf5340dk_nrf5340_cpuapp -d build/display_nrf5340dk -- -DCONFIG_LV_DISP_ROT_90=y -DCONFIG_LV_COLOR_DEPTH_8=y -DCONFIG_LV_Z_BITS_PER_PIXEL=8`

The sample will indeed build and run with the text rotated, but because of a check on the coordinates, the sample throws ` <err> ssd1306: Unsupported origin` errors when attempting to updated the counter that is written to the screen. Can you think of a reason that these changes would cause that error to get thrown? I will continue to investigate, but I wanted to post my progress here in case you (or others) are interested.

brgl, 2022-06-23:
@richesonk this is still concerning upstream lvgl though, right?

richesonk, 2022-06-23:
> @richesonk this is still concerning upstream lvgl though, right?

Hmmmm, not exactly. The patch that I posted over in LVGL, is derived from the LVGL snapshot included with nRF Connect SDK v2.0, so I guess these patches are not technically applied to the upstream LVGL, but any future solution would need to include those patches in (and pick up) the upstream LVGL.

kisvegabor, 2022-06-25:
I agree that (for now) it can be solved on the SSD1306 driver's level without modifying LVGL.

brgl, 2022-06-27:
> Hmmmm, not exactly. The patch that I posted over in LVGL, is derived from the LVGL snapshot included with nRF Connect SDK v2.0, so I guess these patches are not technically applied to the upstream [...]

That's fine, can you submit it as a proper PR in that case?

richesonk, 2022-06-27:
> That's fine, can you submit it as a proper PR in that case?

Unfortunately, I am doing this as part of a work project and my company is very protective of its IP. So, even though this code does not contain any of that IP, there is always the concern that contributing to an open source code base will create an exposure to some code that does contain IP.

Contributing to LVGL looks fairly straightforward so I might be able to get this one through, but I will have to check with the company and I have run into issues before with things like Contributor License Agreements and such. I'll let you know once I have an answer from the company.

richesonk, 2022-07-07:
> Unfortunately, I am doing this as part of a work project and my company is very protective of its IP. So, even though this code does not contain any of that IP, there is always the concern that [...]

Sorry for the delay on this, but, as expected, I have heard back from my company and I am not allowed to contribute directly to the LVGL (or Zephyr) projects. As I mentioned above, my company is highly protective of our IP and tends to error on the side of being overly cautious. So, I will not be able to submit a proper PR for the changes over in the LVGL project.

richesonk, 2022-07-11:
What is the relationship between this team and the LVGL team? I am confused, b/c the issue in LVGL has been closed citing the need for this to be addressed in "the Zephyr driver", but it seems to me that at least some portion of "the Zephyr driver" is packaged inside the LVGL project. Does this team maintain the code in `modules/lib/gui/lvgl/zephyr` just within the LVGL project, or are those files owned/maintained by the LVGL team?

Since the related issue in the LVGL project has been closed, it seems like this issue should probably also be closed. I will just have to maintain my fixes as local patches. Hopefully, others who run into similar issues will find these tickets and be able to pick up the patches too. It's unfortunate that my company's policies prevent me from officially submitting the changes, and no one from these projects has the bandwidth to walk the changes through.

brgl, 2022-07-11:
@richesonk "this" team takes care of zephyr and WRT LVGL - the zephyr<->LVGL glue code specifically. We track upstream LVGL and try to keep downstream changes to a minimum so any general LVGL problems are handled over the LVGL github.

fabiobaltieri, 2022-07-11:
> @brgl thanks for the clarification. So do you consider the zephyr<->LVGL glue code to be code that lives here in the Zephyr repo, or is that the code in the `zephyr` directory of the upstream LVGL project? The LVGL related changes I've implemented to work around this issue are Zephyr specific, they just happened to be contained in the `zephyr` portion of the LVGL project.

If you check the commit history for the repository/code you are working with it should be pretty clear where the code is coming from, the upstream LVGL repo does not have a "zephyr" directory at all (https://github.com/lvgl/lvgl), and if you look at the commit history of the Zephyr copy (https://github.com/zephyrproject-rtos/lvgl) you can see that it gets uprev'd periodically.

The diff you posted in the LVGL issue is for the Zephyr driver, so it looks like it does belongs to this repo (@brgl to confirm). That said, I think it would be a much better use of everyone's time if you were posting those as pull requests directly since it looks like you have the issue already identified and resolved. Then we could comment code on the code rather than doing back and forth in snippets. Can we somehow help solve the concern your company has about contributing code? The conributions would be much appreciated, feel free to get in touch on Discord if you want to talk about it.

richesonk, 2022-07-11:
@fabiobaltieri Thank you for the clarifications and my apologies on not digging that relationship out myself. I don't have a ton of experience with the these types of projects that pull in and maintain/supplement another upstream project.

Unfortunately, I don't think there is anything we can do to alleviate my company's concern about contributing. As I mentioned above, the concern is over exposing IP (a large portion of the company's value is derived from our patent portfolio) and the company takes a very (perhaps overly) cautious approach to protecting those patents/that IP.

closed 2022-09-24 by github-actions[bot]

doctorseus, 2023-02-02:
This issue is not fixed, so it should be reopened.