zephyrproject-rtos/zephyr #46478: mimxrt1050_evk_qspi freeze when erasing flash

jeppenodgaard, 2022-06-13:
**Describe the bug**
mimxrt1050_evk_qspi freeze when erasing flash

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b mimxrt1050_evk_qspi zephyr/samples/drivers/flash_shell/
1. west flash
1. See logs

**Expected behavior**
`flash page_erase` should not freeze when erasing pages in parition "image-1".

**Impact**
Application freeze. Seems to occasionally work.
`flash page_erase 784 10` never works.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.1.0  ***
Found flash controller IS25WP064.
Flash I/O commands can be run.

uart:~$ flash page_erase 784 1
Erasing page 784 (start offset 0x310000, size
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - v3.1.0

**Additional context**
Maybe related:
https://github.com/zephyrproject-rtos/zephyr/issues/44676

jeppenodgaard, 2022-06-13:
`LOG_MODE_IMMEDIATE=y` and `LOG_MODE_IMMEDIATE=y` seems to be a work around.
`flash page_erase 784 100` runs successfully with those enabled.

fabiobaltieri, 2022-06-15:
@dleach02 could you look into this?

danieldegrasse, 2022-06-15:
@jeppenodgaard What version of zephyr did you reproduce this issue on? I have tried the flash shell on both Zephyr 3.1.0 (2ddd73feafd3316af2c547c34d6969bea637d5c6) and a recent copy of the main branch (85f202744dd30baa1d42de4baca03a75e5afaa6c), and have been able to erase pages 784-793 on both revisions.

jeppenodgaard, 2022-06-15:
I used also used https://github.com/zephyrproject-rtos/zephyr/commit/2ddd73feafd3316af2c547c34d6969bea637d5c6 and I cannot reproduce it today either. Looking at the code, would it make sense to add k_shed_lock before calling the flash api?

danieldegrasse, 2022-06-15:
The flash driver already does something similar under the hood- see [line 419 of NXP's flash driver](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/flash_mcux_flexspi_nor.c#L419). The `irq_lock()` call here will block preemption by anything besides zero-latency interrupts while the flash is being erased.

jeppenodgaard, 2022-06-16:
I see. Wasn't aware it also prevented thread switching.

To sum up where I experienced the issue:

1. With my own application using flash_img_buffered_write() in Zephyr 3.0.0.
2. zephyr/samples/drivers/flash_shell/ in Zephyr 3.0.0
3. zephyr/samples/drivers/flash_shell/ in Zephyr 3.1.0
4. On another evaluation kit.

I wonder if the `west flash -r pyocd` did something.
Still not able to reproduce today. I will close it for now since there's no lead.