zephyrproject-rtos/zephyr #46483: Update RISC-V ISA configs

stephanosio, 2022-06-13:
**Is your enhancement proposal related to a problem? Please describe.**

From https://www.mail-archive.com/u-boot@lists.denx.de/msg430287.html:

> From version 2.38, binutils default to ISA spec version 20191213. This 
> means that the csr read/write (csrr*/csrw*) instructions and fence.i
> instruction has separated from the `I` extension, become two standalone
> extensions: Zicsr and Zifencei.

The `csr*` instructions are no longer part of the "I" extension and require a separate extension called "Zicsr" to be specified to the `-march` flag.

Without this, build will fail with the following error message:
```
/home/runner/work/sdk-ng/sdk-ng/zephyrproject/zephyr/soc/riscv/riscv-privilege/common/soc_irq.S: Assembler messages:
/home/runner/work/sdk-ng/sdk-ng/zephyrproject/zephyr/soc/riscv/riscv-privilege/common/soc_irq.S:31: Error: unrecognized opcode `csrrc t1,mip,t0'
/home/runner/work/sdk-ng/sdk-ng/zephyrproject/zephyr/soc/riscv/riscv-privilege/common/soc_irq.S:50: Error: unrecognized opcode `csrr t0,mcause'
```

For more details, refer to the following log attempting to build a Zephyr RISC-V target using the Binutils 2.38:
https://github.com/zephyrproject-rtos/sdk-ng/runs/6862707569?check_suite_focus=true#step:11:226

**Describe the solution you'd like**

Add the ISA extension Kconfig symbols for the new (missing) extensions:

* Zicsr
* Zifencei
* Za, Zb, Zc, Zs (formerly the "B" extension which, to my knowledge, is no longer a thing)

**Describe alternatives you've considered**

We could build the Binutils targeting an older ISA spec (for example, with `--with-isa-spec=2.2`), but this does not reflect the latest RISC-V standard and will likely become unmaintainable in the near future (it is already sort of deprecated).

**Additional context**

Discovered while upgrading the version of the Binutils included in the Zephyr SDK from 2.35.1 to 2.38 -- this enhancement will be required in order to support the upcoming Zephyr SDK 0.15.0 release.

stephanosio, 2022-06-13:
Test Zephyr SDK RISC-V toolchain build based on Binutils 2.38 can be downloaded here:
https://github.com/zephyrproject-rtos/sdk-ng/suites/6907912608/artifacts/268357069

stephanosio, 2022-06-13:
See also https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/aE1ZeHHCYf4

mmahadevan108, 2022-09-12:
@stephanosio I have cleared the milestone field as we are past the feature freeze date for the 3.2 release. Kindly pick a new milestone for a future release.

stephanosio, 2022-09-12:
This was already merged as part of https://github.com/zephyrproject-rtos/zephyr/issues/46491

closed 2022-09-12 by stephanosio