zephyrproject-rtos/zephyr #46497: Modbus: Add support for FC03 without floating-point extension

danielskatt, 2022-06-13:
**Is your enhancement proposal related to a problem? Please describe.**
When running Modbus as a client on a motor drive, for example Altivar Machine ATV320, there is a mismatch in the number of registers since Zephyr uses a function with floating-point extension which the Altavar does not so the Altar expects for example 2 bytes and Zephyr expects 4 bytes.  

**Describe the solution you'd like**
Add a function in the modbus_client.c called mbc_validate_fc03_response which does not use floating-point extension so instead of typecasting the data to (float *) use a typecast of (uint16_t *)

**Describe alternatives you've considered**
Since there is not way of changing that parameter in the motor driver and modbus_server.c has both FC03 support for uint16_t and float it should also be in the modbus_client.c. I can't see another alternative that is not adding the function.  

**Additional context**
No additional context

danielskatt, 2022-06-20:
There is an exciting pull request regarding this issue waiting for review here
https://github.com/zephyrproject-rtos/zephyr/pull/46498

jfischer-no, 2022-06-29:
floating-point extension applies only to register addresses above 5000 (MODBUS_FP_EXTENSIONS_ADDR). If there is a conflict and floating-point extension is not desired at all, it can be disabled by Kconfig option MODBUS_FP_EXTENSIONS (CONFIG_MODBUS_FP_EXTENSIONS=n). In client implementation [FC03 falls through to FC04](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/modbus/modbus_client.c#L128-L133) if one of the conditions above do not apply.

danielskatt, 2022-07-01:
@jfischer-no Thanks for the reply, I was not aware of that it was intended like that. I'll remove the pull request

closed 2022-07-01 by danielskatt

lanwer50, 2023-11-22:
@jfischer-no: Thanks for clarifying. From what I understand: if we need to read only non-float values, then the right solution is to disable the MODBUS_FP_EXTENSIONS to not bother with register count mismatches. 

My question is: What if we need to read both non float (e.g. uint16_t with reg addr >= 5000) and a float registers. We still then need to use the FP_EXTENSIONS which brings us back to the main problem. Correct? Or is there already a solution that deals with this situation?

KristjanZigart, 2024-10-08:
@jfischer-no Why exactly do floating-point extensions apply to registers between 5000 and uint16_t max? I apologize if I'm not familiar with some specifics.

My current use case is reading floating point and non-floating point registers. As soon as you enable `MODBUS_FP_EXTENSIONS` it becomes impossible to do both without patching Zephyr.

jfischer-no, 2024-10-08:
It follows how uC/Modbus supports FP registers. Currently I think it is better to remove it completely and let the user do the conversion on the server side. From the spec there are only two types, single-bit and 16-bit word, but no floats. Btw, it is broken (https://github.com/zephyrproject-rtos/zephyr/pull/78556).