zephyrproject-rtos/zephyr #46555: test: tests/drivers/adc/adc_api twister result wrong

hunterhu3000, 2022-06-15:
**Describe the bug**
While running daily test the test case of tests/drivers/adc/adc_api is failed. 
The logs in email "Jenkins auto test result: Github main" -> HTML-> sample.drivers.sample.drivers.adc  displays the log of samples/basic/threads/. 
Actually tests/drivers/adc/adc_api can pass. But it displayed fail. 
samples/basic/threads/ can also pass when running seperately. But the log on the website looks abnormal.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?  mec172xevb_assy6906

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p mec172xevb_assy6906 --device-testing --device-serial /dev/ttyUSB0 -v -T tests/ -e cmsis_dsp -x=CONFIG_BOOT_DELAY=500 -T samples/

**Expected behavior**
The tests/drivers/adc/adc_api should pass

**Impact**
The twister is not generating the report correctly in some cases

**Logs and console output**
See the "Jenkins auto test result: Github main" of June.14

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Ubuntu 20.04
 - Toolchain (e.g Zephyr SDK, ...): 0.14.1
 - Commit SHA or Version used:  3f8d5c49b3fb54878504f50e87902eab113fc5e2

nashif, 2022-06-21:
> **Logs and console output**
> See the "Jenkins auto test result: Github main" of June.14

what is that? Please provide all information in the bug report. It is not clear what the results you are getting and what you are expecting and why this is an issue with twister and not with the test/platform itself.

hunterhu3000, 2022-06-24:
The issue is not reproduced after 6/16. So the issue can be closed. When it happens again, we will submit a new issue. Thanks

closed 2022-06-24 by hunterhu3000