zephyrproject-rtos/zephyr #46556: Doc inconsistency

caco3, 2022-06-15:
There seem to be some documentation inconsistency:

1. `CONFIG_BT_CTLR_ULL_HIGH_PRIO` can be found in the docs for Zephyr `3.0.0`: https://docs.zephyrproject.org/3.0.0/search.html?q=BT_CTLR_ULL_HIGH_PRIO but can not be found in 3.1.0: https://docs.zephyrproject.org/3.1.0/search.html?q=BT_CTLR_ULL_HIGH_PRIO, see also https://github.com/zephyrproject-rtos/zephyr/discussions/46529
2. In `3.1.0` it only can be found through the `Kconfig Search` at https://docs.zephyrproject.org/3.1.0/kconfig.html
3. Searching there for `CONFIG_BT_CTLR_ULL_HIGH_PRIO` shows a dependency for its default value to `CONFIG_BT_CTLR_ULL_LLL_PRIO_SUPPORT`. How ever that flag itself can not be found: https://docs.zephyrproject.org/3.1.0/kconfig.html#CONFIG_BT_CTLR_ULL_LLL_PRIO_SUPPORT

carlescufi, 2022-06-16:
I am not sure if this is a Kconfig doc issue or an actual Bluetooth controller problem.

cvinayak, 2022-06-17:
This is not Bluetooth Controller problem, the Kconfigs are common to both v3.0.0 and v3.1.0 releases.

1. The documentation search no longer finds Kconfigs in the v3.1.0 documentation.
2. The Kconfig options section in documentation only find the non-hidden Kconfigs in v3.1.0 documentation.

In comparison, v3.0.0 can find all Kconfigs (non-hidden and hidden) in the documentation search available in the left side margin.

Also, the Kconfig options page is missing in v3.1.0 release:
[image]

Whereas v3.0.0 release displays this:
[image]

carlescufi, 2022-06-17:
Closing with explanation from @cvinayak and @gmarull, thanks both!

closed 2022-06-17 by carlescufi

caco3, 2022-06-17:
Thanks for the explanations!

Personally I think it is rater odd that a parameter shows references to other parameters with a link but then the new page does not provide new information. It would be more consistent if those hidden parameters would have no link.
But I understand that this might not be trivial to improve.