zephyrproject-rtos/zephyr #46558: crash on bt_le_adv_start() when using CONFIG_BT_CTLR_ADVANCED_FEATURES

caco3, 2022-06-15:
We need to enable `CONFIG_BT_CTLR_ADVANCED_FEATURES` because we need to adjust `CONFIG_BT_CTLR_ULL_HIGH_PRIO`, see https://github.com/zephyrproject-rtos/zephyr/discussions/41401#discussioncomment-1858153

In Zephyr `3.1.0` this causes a crash in `bt_le_adv_start()`:

```[00:00:00.411,926] <inf> bt: Starting Advertisement...
[00:00:00.422,210] <err> os: ***** HARD FAULT *****
[00:00:00.422,210] <err> os:   Fault escalation (see below)
[00:00:00.422,241] <err> os: ARCH_EXCEPT with reason 3

[00:00:00.422,241] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000001  r2/a3:  0x00000000
[00:00:00.422,241] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x0003ead7
[00:00:00.422,271] <err> os:  xpsr:  0x61000028
[00:00:00.422,271] <err> os: Faulting instruction address (r15/pc): 0x0003eae2
[00:00:00.422,302] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:00.422,332] <err> os: Fault during interrupt handling

[00:00:00.422,363] <err> os: Current thread: 0x20015f90 (unknown)
[00:00:01.841,278] <err> os: Halting system
```

Note: In Zephyr `2.7.2` same code/config worked fine. I think also on Zephyr `3.0.0` it was still ok (tbc.).

cvinayak, 2022-06-15:
Use `arm-none-eabi-addr2line -e <build folder>/zephyr/zephyr.elf 0x0003eae2` to get the Faulting filename and line number.

I still suspect that some threads in your application or subsys need their stack sizes increased.

cvinayak, 2022-06-15:
@caco3 please try with the `LL_ASSERT(!retval);` commented out in the lll_adv.c at line 802.

caco3, 2022-06-15:
> Use `arm-none-eabi-addr2line -e <build folder>/zephyr/zephyr.elf 0x0003eae2` to get the Faulting filename and line number.
> 
> I still suspect that some threads in your application or subsys need their stack sizes increased.

That returns `/opt/zephyrproject-v3.1.0/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv.c:490`!

cvinayak, 2022-06-15:
> @caco3 please try with the `LL_ASSERT(!retval);` commented out in the lll_adv.c at line 802.

Did you try with the assertion commented out?

caco3, 2022-06-15:
yes, just tried it now.

With the line 802 disabled, I can successfully start the advertisement.
Also I then can connect to the device and disconnect again.
How ever, after the disconnect it is supposed to start the advertisement again, which then again leads to a segfault:
```
[00:01:07.027,374] <inf> bt: Starting Advertisement...
[00:01:07.036,743] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000001  r2/a3:  0x00000000
[00:01:07.036,743] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x0003e931
[00:01:07.036,743] <err> os:  xpsr:  0x61000000
[00:01:07.036,773] <err> os: Faulting instruction address (r15/pc): 0x0003e93c
[00:01:07.036,834] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:01:07.036,865] <err> os: Current thread: 0x20015fa8 (unknown)
[00:01:07.144,836] <err> os: Halting system
```
```
arm-none-eabi-addr2line -e zephyr.elf 0x0003e93c
/opt/zephyrproject-v3.1.0/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv.c:410 (discriminator 1)```

caco3, 2022-06-15:
Stack usage in first advertisement:
```
Thread analyze:
 0x20016770          : STACK: unused 7984 usage 208 / 8192 (2 %); CPU: 0 %      : Total CPU cycles used: 0
 0x200166c8          : STACK: unused 7984 usage 208 / 8192 (2 %); CPU: 0 %      : Total CPU cycles used: 1
 0x20016140          : STACK: unused 140 usage 1004 / 1144 (87 %); CPU: 0 %      : Total CPU cycles used: 6861 => /zephyr/subsys/bluetooth/host/hci_ecc.c:40
 0x200161e8          : STACK: unused 7720 usage 472 / 8192 (5 %); CPU: 0 %      : Total CPU cycles used: 92
 0x200138a8          : STACK: unused 500 usage 524 / 1024 (51 %); CPU: 3 %      : Total CPU cycles used: 22409 => zephyr/subsys/debug/thread_analyzer.c:180
 0x20001b28          : STACK: unused 840 usage 184 / 1024 (17 %); CPU: 0 %      : Total CPU cycles used: 3
 0x20001d28          : STACK: unused 864 usage 160 / 1024 (15 %); CPU: 0 %      : Total CPU cycles used: 1
 0x20001dd0          : STACK: unused 580 usage 444 / 1024 (43 %); CPU: 0 %      : Total CPU cycles used: 61 => /zephyr/drivers/i2c/i2c_common.c:19
 0x20001bd0          : STACK: unused 864 usage 160 / 1024 (15 %); CPU: 0 %      : Total CPU cycles used: 4
 0x20001c78          : STACK: unused 484 usage 540 / 1024 (52 %); CPU: 0 %      : Total CPU cycles used: 6 => dataStreaming.c:162
 0x20001e80          : STACK: unused 824 usage 200 / 1024 (19 %); CPU: 0 %      : Total CPU cycles used: 19
 0x20016bf0          : STACK: unused 7876 usage 316 / 8192 (3 %); CPU: 0 %      : Total CPU cycles used: 4
 0x20016098          : STACK: unused 1556 usage 492 / 2048 (24 %); CPU: 6 %      : Total CPU cycles used: 48777
 0x20016aa0          : STACK: unused 8032 usage 160 / 8192 (1 %); CPU: 88 %      : Total CPU cycles used: 651704
 0x20016b48          : STACK: unused 7168 usage 1024 / 8192 (12 %); CPU: 0 %      : Total CPU cycles used: 1282
 ISR0                : STACK: unused 7528 usage 664 / 8192 (8 %)
```

Stack usage while connected:
```
Thread analyze:
 0x20016770          : STACK: unused 7436 usage 756 / 8192 (9 %); CPU: 0 %      : Total CPU cycles used: 200
 0x200166c8          : STACK: unused 7880 usage 312 / 8192 (3 %); CPU: 0 %      : Total CPU cycles used: 131
 0x20016140          : STACK: unused 140 usage 1004 / 1144 (87 %); CPU: 0 %      : Total CPU cycles used: 6861
 0x200161e8          : STACK: unused 7720 usage 472 / 8192 (5 %); CPU: 0 %      : Total CPU cycles used: 180
 0x200138a8          : STACK: unused 500 usage 524 / 1024 (51 %); CPU: 3 %      : Total CPU cycles used: 85770
 0x20001b28          : STACK: unused 728 usage 296 / 1024 (28 %); CPU: 0 %      : Total CPU cycles used: 13
 0x20001d28          : STACK: unused 588 usage 436 / 1024 (42 %); CPU: 0 %      : Total CPU cycles used: 4
 0x20001dd0          : STACK: unused 580 usage 444 / 1024 (43 %); CPU: 0 %      : Total CPU cycles used: 199
 0x20001bd0          : STACK: unused 864 usage 160 / 1024 (15 %); CPU: 0 %      : Total CPU cycles used: 16
 0x20001c78          : STACK: unused 484 usage 540 / 1024 (52 %); CPU: 0 %      : Total CPU cycles used: 16
 0x20001e80          : STACK: unused 812 usage 212 / 1024 (20 %); CPU: 0 %      : Total CPU cycles used: 82
 0x20016bf0          : STACK: unused 7856 usage 336 / 8192 (4 %); CPU: 0 %      : Total CPU cycles used: 38
 0x20016098          : STACK: unused 1556 usage 492 / 2048 (24 %); CPU: 2 %      : Total CPU cycles used: 63478
 0x20016aa0          : STACK: unused 8032 usage 160 / 8192 (1 %); CPU: 94 %      : Total CPU cycles used: 2600872
 0x20016b48          : STACK: unused 7168 usage 1024 / 8192 (12 %); CPU: 0 %      : Total CPU cycles used: 1952
 ISR0                : STACK: unused 7528 usage 664 / 8192 (8 %)
```

caco3, 2022-06-15:
Note that I stop advertisement after the connection got established: `bt_le_adv_stop();`.

If I do **not** stop it, and after the disconnect also do **not** start it again, it does not crash!

cvinayak, 2022-06-15:
If you use the below you will get the analysis every 5 seconds and with the thread names:

```
CONFIG_THREAD_ANALYZER=y
CONFIG_THREAD_ANALYZER_AUTO=y
CONFIG_THREAD_ANALYZER_AUTO_INTERVAL=5
CONFIG_THREAD_NAME=y
```

Could you re-run with these and review if all the threads are listed. If any thread is not listed, I have seen those threads to have overflowed and cause troubles.

caco3, 2022-06-15:
ah, nice!

Stack usage in first advertisement (all my own threads are disabled):
```
Thread analyze:
 BT RX               : STACK: unused 8000 usage 192 / 8192 (2 %); CPU: 0 %      : Total CPU cycles used: 1
 BT RX pri           : STACK: unused 8000 usage 192 / 8192 (2 %); CPU: 0 %      : Total CPU cycles used: 1
 BT ECC              : STACK: unused 7188 usage 1004 / 8192 (12 %); CPU: 2 %      : Total CPU cycles used: 6465
 BT TX               : STACK: unused 7740 usage 452 / 8192 (5 %); CPU: 0 %      : Total CPU cycles used: 62
 thread_analyzer     : STACK: unused 560 usage 464 / 1024 (45 %); CPU: 2 %      : Total CPU cycles used: 4834
 sysworkq            : STACK: unused 7884 usage 308 / 8192 (3 %); CPU: 0 %      : Total CPU cycles used: 1
 logging             : STACK: unused 276 usage 492 / 768 (64 %); CPU: 24 %      : Total CPU cycles used: 58327
 idle                : STACK: unused 8032 usage 160 / 8192 (1 %); CPU: 69 %      : Total CPU cycles used: 163744
 main                : STACK: unused 7184 usage 1008 / 8192 (12 %); CPU: 0 %      : Total CPU cycles used: 899
 ISR0                : STACK: unused 7528 usage 664 / 8192 (8 %)
```

caco3, 2022-06-15:
Log just before starting the advertisement the 2nd time:
```
[00:00:20.314,483] <inf> bt: waiting 10s before starting advertisement...
[00:00:21.972,229] <inf> thread_analyzer: Thread analyze:
[00:00:21.973,419] <inf> thread_analyzer:  BT RX               : STACK: unused 7452 usage 740 / 8192 (9 %); CPU: 0 %
[00:00:21.973,419] <inf> thread_analyzer:       : Total CPU cycles used: 190
[00:00:21.974,700] <inf> thread_analyzer:  BT RX pri           : STACK: unused 7920 usage 272 / 8192 (3 %); CPU: 0 %
[00:00:21.974,700] <inf> thread_analyzer:       : Total CPU cycles used: 68
[00:00:21.975,860] <inf> thread_analyzer:  BT ECC              : STACK: unused 7188 usage 1004 / 8192 (12 %); CPU: 0 %
[00:00:21.975,860] <inf> thread_analyzer:       : Total CPU cycles used: 6449
[00:00:21.977,081] <inf> thread_analyzer:  BT TX               : STACK: unused 7748 usage 444 / 8192 (5 %); CPU: 0 %
[00:00:21.977,111] <inf> thread_analyzer:       : Total CPU cycles used: 101
[00:00:21.977,233] <inf> thread_analyzer:  thread_analyzer     : STACK: unused 508 usage 516 / 1024 (50 %); CPU: 0 %
[00:00:21.977,233] <inf> thread_analyzer:       : Total CPU cycles used: 1510
[00:00:21.978,454] <inf> thread_analyzer:  sysworkq            : STACK: unused 7548 usage 644 / 8192 (7 %); CPU: 0 %
[00:00:21.978,454] <inf> thread_analyzer:       : Total CPU cycles used: 40
[00:00:21.978,546] <inf> thread_analyzer:  logging             : STACK: unused 276 usage 492 / 768 (64 %); CPU: 12 %
[00:00:21.978,546] <inf> thread_analyzer:       : Total CPU cycles used: 91631
[00:00:21.979,827] <inf> thread_analyzer:  idle                : STACK: unused 8032 usage 160 / 8192 (1 %); CPU: 85 %
[00:00:21.979,858] <inf> thread_analyzer:       : Total CPU cycles used: 606621
[00:00:21.980,987] <inf> thread_analyzer:  main                : STACK: unused 7184 usage 1008 / 8192 (12 %); CPU: 0 %
[00:00:21.981,018] <inf> thread_analyzer:       : Total CPU cycles used: 949
[00:00:21.981,964] <inf> thread_analyzer:  ISR0                : STACK: unused 7496 usage 696 / 8192 (8 %)
[00:00:26.982,055] <inf> thread_analyzer: Thread analyze:
[00:00:26.983,245] <inf> thread_analyzer:  BT RX               : STACK: unused 7452 usage 740 / 8192 (9 %); CPU: 0 %
[00:00:26.983,245] <inf> thread_analyzer:       : Total CPU cycles used: 190
[00:00:26.984,527] <inf> thread_analyzer:  BT RX pri           : STACK: unused 7920 usage 272 / 8192 (3 %); CPU: 0 %
[00:00:26.984,527] <inf> thread_analyzer:       : Total CPU cycles used: 68
[00:00:26.985,687] <inf> thread_analyzer:  BT ECC              : STACK: unused 7188 usage 1004 / 8192 (12 %); CPU: 0 %
[00:00:26.985,687] <inf> thread_analyzer:       : Total CPU cycles used: 6449
[00:00:26.986,907] <inf> thread_analyzer:  BT TX               : STACK: unused 7748 usage 444 / 8192 (5 %); CPU: 0 %
[00:00:26.986,938] <inf> thread_analyzer:       : Total CPU cycles used: 101
[00:00:26.987,060] <inf> thread_analyzer:  thread_analyzer     : STACK: unused 508 usage 516 / 1024 (50 %); CPU: 0 %
[00:00:26.987,060] <inf> thread_analyzer:       : Total CPU cycles used: 1830
[00:00:26.988,281] <inf> thread_analyzer:  sysworkq            : STACK: unused 7548 usage 644 / 8192 (7 %); CPU: 0 %
[00:00:26.988,281] <inf> thread_analyzer:       : Total CPU cycles used: 40
[00:00:26.988,372] <inf> thread_analyzer:  logging             : STACK: unused 276 usage 492 / 768 (64 %); CPU: 11 %
[00:00:26.988,372] <inf> thread_analyzer:       : Total CPU cycles used: 97729
[00:00:26.989,654] <inf> thread_analyzer:  idle                : STACK: unused 8032 usage 160 / 8192 (1 %); CPU: 87 %
[00:00:26.989,685] <inf> thread_analyzer:       : Total CPU cycles used: 764267
[00:00:26.990,814] <inf> thread_analyzer:  main                : STACK: unused 7184 usage 1008 / 8192 (12 %); CPU: 0 %
[00:00:26.990,844] <inf> thread_analyzer:       : Total CPU cycles used: 999
[00:00:26.991,790] <inf> thread_analyzer:  ISR0                : STACK: unused 7496 usage 696 / 8192 (8 %)
ASSERTION FAIL [!err] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv.c:410
[00:00:30.314,575] <inf> bt: Starting Advertisement...
[00:00:30.323,730] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000001  r2/a3:  0x00000000
[00:00:30.323,760] <err> os: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x0003afe1
[00:00:30.323,760] <err> os:  xpsr:  0x61000000
[00:00:30.323,760] <err> os: Faulting instruction address (r15/pc): 0x0003afec
[00:00:30.323,791] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:00:30.323,822] <err> os: Current thread: 0x20015cc8 (BT TX)
[00:00:30.375,061] <err> os: Halting system
```

cvinayak, 2022-06-15:
ok. This is a bug related to fdda91d4. I need to think on how to handle synchronizing the k_sem_give with k_sem_take when mayfly is used to deferred the k_sem_give to ULL_HIGH execution context.

I will follow up, and ping back here when I have something.

Thank you again on reporting the issue.

caco3, 2022-06-15:
Thank you for looking into it!

cvinayak, 2022-06-17:
@caco3 Please the fix in this PR https://github.com/zephyrproject-rtos/zephyr/pull/46633 and reverrt back if there are further issues

PR #46633 (2022-06-20): Bluetooth: Controller: Fix pdu_free_sem_give assertion under ZLI use
Fix assertion due to multiple mayfly_enqueue calls used
under ZLI when pdu_free_sem_give is invoked from the LLL.

Fixes #46558.
closed 2022-06-20 by carlescufi