zephyrproject-rtos/zephyr #46573: raspberry pi pico always in mass storage mode

jeronimoagullo, 2022-06-15:
**Describe the bug**
I am trying to use the raspberry pi pico (rpi_pico board) with Zephyr. I managed to build the blink and hello_world projects. However, when it is flash using the UF2 file (`build/zephyr/zephyr.uf2`), the raspberry pi pico resets and appears again as a mass storage device (**wihtout pressing the BOOTSEL botton**). It seems that the app or the boot2 were not flashed properly.

**Environment**
- Zephyr version: 3.1.99
- Zephyr SDK: 0.14.2
- cmd console in Windows 11

**To Reproduce**
Steps to reproduce the behavior:
1. go to hello_world or blink example folder
2. `west build -b rpi_pico -c`
3. copy the file `/build/zephyr/zephyr.uf2` to the RPI_RP2 mass storage device
4. The raspberry pi pico restarts but it appears again as a mass storage device and neither the hello_world example prints data in the UARTS (I tested with a TTL-USB converter) nor the LED blinks (I checked the gpio pin in the dts file). I tend to think that the reason is that it is in "Bootloader mode" and the app doesn't start.

**LOGS**
I would like to share also the output of the build process. The `BOOT_FLASH` uses 0GB, which I don't know if it is the origin of the issue. However, it builds the bootloader since it generates the file `\build\bootloader\boot_stage2.bin`.

```
[47/47] Linking C executable zephyr\zephyr.elf
Memory region         Used Size  Region Size  %age Used
      BOOT_FLASH:          0 GB        256 B      0.00%
           FLASH:       11220 B    2096896 B      0.54%
            SRAM:        3736 B       264 KB      1.38%
        IDT_LIST:          0 GB         2 KB      0.00%
Converting to uf2, output size: 22528, start address: 0x10000000
Wrote 22528 bytes to zephyr.uf2
``` 
**More Info**
I have tested also the raspberry pi pico with Arduino in which I managed easily to flash a hello_world example. Then, the raspberry pi pico restarted and appeared as a COM device instead of a mass storage device (It was needed to press the BOOTSEL botton to go back to mass storage state). However, when I use zephyr the app does nothing and the board is always a mass storage device (even without pressing the BOOTSEL botton).

ThomasFike, 2022-06-16:
I also have this issue but on popos 22.04 LTS. The blinky example just never actually runs.

ThomasFike, 2022-06-16:
I tried to run the elf through picotool to see if there was any info and it returned none for all fields
`thomasfike@pop-os:~/pico/picotool/build$ ./picotool info -a ~/zephyrproject/zephyr/build/zephyr/zephyr.uf2
File /home/thomasfike/zephyrproject/zephyr/build/zephyr/zephyr.uf2:

Program Information
 none

Fixed Pin Information
 none

Build Information
 none
`

ThomasFike, 2022-06-16:
In looking at the discord I found that this issue was introducted in PR #46062. This change will likely need to be reverted and solved correctly to fix this  issue for all users. I reverted the commit locally and the UF2 now works for blinky.

ThomasFike, 2022-06-16:
In looking at the discord I found that this issue was introduced in #46062. This change will likely need to be reverted and solved correctly to fix this issue for all users. I reverted the commit locally and the UF2 now works for blinky.

yonsch, 2022-06-16:
Hi @jeronimoagullo and @ThomasFike , this bug was introduced very recently, as you said. #46616 should fix it.

fabiobaltieri, 2022-06-21:
Maybe related to https://github.com/zephyrproject-rtos/zephyr/pull/46616?

jeronimoagullo, 2022-06-22:
Hi @ThomasFike, @yonsch and @fabiobaltieri 

I have just tested the #46616 locally (copying the changes to the repective CMakeLists.txt files) and it works smoothly. I will wait to the approval of this commit. Once, it occurs, I will close this issue.

Thank you very much for the support.

PR #46616 (2022-06-23): rpi_pico: Fix bootloader linking
A recent change to hal_rpi_pico's cmake (#46062), intended to fix the compilation via Unix Makefiles, ended up not linking the BL to the final ELF. This commit fixes both problems.
This was tested with and without `-G"Unix Makefiles"`, and was tested on hardware.

Fixes #46573
closed 2022-06-23 by carlescufi