zephyrproject-rtos/zephyr #46580: Suggestion for additional configuration of `twister --coverage` gcovr formats

jpwright, 2022-06-15:
**Is your enhancement proposal related to a problem? Please describe.**
I want to use twister to generate coverage reports for unit tests using gcovr in xml (cobertura) formatted output. This is supported by gcovr using the `--xml` flag, but not accessible when using `twister --coverage` which is hard-coded to call gcovr with `--json` and `--html` flags.

**Describe the solution you'd like**
An additional twister option, such as
1. `--coverage-formats=json,xml,html` (default would be `json,html`) allowing one or more output formats to be produced by gcovr, or
2. `--coverage-options` allowing arbitrary flags to be passed to calls to gcovr

**Describe alternatives you've considered**
Looked for a tool to convert the gcovr json reports to other formats, but this doesn't seem to exist

**Additional context**
I am happy to submit a PR with the change if the additional option is acceptable.

carlescufi, 2022-06-15:
@nashif @PerMac @gopiotr feedback?

PerMac, 2022-06-20:
I am not experienced with gcovr. If you consider there are other useful option which a user might want to use then 2.) seems the way to go. If not really, 1.) might be enough.

nashif, 2022-07-04:
@jpwright  you are welcome to submit a PR introducing this.

jpwright, 2022-07-06:
Thanks for the feedback - I submitted a PR that just covers output report formats rather than any arbitrary gcovr option.

PR #47427 (2022-07-18): twister: add `--coverage-formats` option for use with gcovr
Adds an `--coverage-formats` option to twister, which (if gcovr is used as the coverage tool) allows generation of alternate report formats such as coberatura xml, gcc txt, etc.

Closes #46580.
closed 2022-07-18 by nashif