zephyrproject-rtos/zephyr #46596: STM32F74X RMII interface does not work

easternblack, 2022-06-16:
**Describe the bug**
Hello.

I'm testing MII & RMII controllers with stm32f745.
I have a custom board that has stm32f745 + lan9303 mii controller. lan9303 is run in un-managed state, and I tested ping to my desktop PC via NETWORK_SHELL. in this case, ping works very well.
And then, I make a test in same condition with stm32f745 + rtl8304 which is rmii controller, but it does not work.

I thought there may be hardware issue, so continued with NUCLEO-F746zg, and got same result.

I'm originally using Zephyr 2.2.99 but also tested this on 2.7.0 and 3.1.99.
Of course I checked "CONFIG_ETH_STM32_HAL_MII=n" for using RMII, and pinmux. and tested on various network configs.

**To Reproduce**

I tested on simple 'hello world' sample. because I thought NET_SHELL is enough to test ethernet controller.
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=nucleo_f746zg ..
3. make
4. See error
```
*** Booting Zephyr OS build zephyr-v3.1.0-469-g92bc3e47c0f8  ***
Hello World! nucleo_f746zg
[00:00:02.463,000] <inf> net_config: Initializing network
[00:00:02.463,000] <inf> net_config: Waiting interface 1 (0x20010b10) to be up...
[00:00:02.482,000] <err> eth_stm32_hal: Failed to enqueue frame into RX queue: -115
[00:00:02.482,000] <inf> net_config: Interface 1 (0x20010b10) coming up
[00:00:02.482,000] <inf> net_config: IPv4 address: 192.168.65.125
uart:~$
uart:~$
uart:~$ net ping 192.168.65.16
PING 192.168.65.16
Ping timeout
```

**Expected behavior**
This is the suceessful case with MII(stm32f745 + lan9303)
```
uart:~$ net ping 192.168.65.125
PING 192.168.65.125
28 bytes from 192.168.65.125 to 192.168.65.125: icmp_seq=0 ttl=64 time=0 ms
28 bytes from 192.168.65.125 to 192.168.65.125: icmp_seq=1 ttl=64 time=0 ms
28 bytes from 192.168.65.125 to 192.168.65.125: icmp_seq=2 ttl=64 time=0 ms
```

**Environment (please complete the following information):**
 - OS: Linux ubuntu 18.04
 - zephyr-sdk-0.13.2
 - Tested on Zephyr 2.2.99, 2.7.0, 3.1.99.

erwango, 2022-12-19:
@easternblack A new version of STM32 ethernet HAL was delivered earlier this year.
It's purpose if to fix a large number of known issues and limitations. Among the issues fixed, there were issues related to RMII i/f:
108675 Ethernet erroneous data received in RMII configuration
108674 Ethernet erroneous data received in RMII configuration
108670 Ethernet erroneous data received in RMII configuration 

This new HAL version was integrated in Zephyr few weeks ago https://github.com/zephyrproject-rtos/zephyr/pull/45770 (only available on F4/H7)
It can now be also used on F7 with this PR: https://github.com/zephyrproject-rtos/zephyr/pull/53193

To use it:
- fetch PR #53193
- run `west update`
- Compile application with flag: `CONFIG_ETH_STM32_HAL_API_V2=y`

For more info about this new API:
https://community.st.com/s/question/0D53W00001Gi9BoSAJ/ethernet-hal-driver-reworked-by-st-and-available-in-22q1-preview-now-available-on-github

Please have a check and get back to us.

Desvauxm-st, 2023-01-04:
I try to reproduce the issue:STM32F74X RMII interface does not work #46596
I'm testing MII & RMII controllers with nucleo_f746zg and I use a cisco Router LINKSYS WRT54G-TM (4-Port 10/100 Mbits)( see file  attachment)

conclusion :

when CONFIG_ETH_STM32_HAL_MII =n 

ping is OK in my case

[STM32F74X RMII interface does not work.docx](https://github.com/zephyrproject-rtos/zephyr/files/10342494/STM32F74X.RMII.interface.does.not.work.docx)

erwango, 2023-02-08:
@easternblack Any update on this issue ?

easternblack, 2023-02-09:
I changed ethernet interface, so I has no plan for test RMII on STM32F7 so far.   
If there's issue on this topic, I'll reopen.   
Thanks for your help.

easternblack, 2023-02-17:
@erwango I think this is out of topic, but I'll tell you what I did and consequence.    
   
I have nucleo-f746zg, and I confirmed @Desvauxm-st 's example works to me too.   
But after that, I enabled ETH_STM32_HAL_API_V2, and nucleo-f746g did not ethernet.    
**ETH_STM32_HAL_API_V2=n**
```
*** Booting Zephyr OS build 7796596e9bfd ***
[00:00:04.032,000] <inf> net_config: Initializing network
[00:00:04.033,000] <inf> net_config: Waiting interface 1 (0x20010bd8) to be up...
[00:00:04.042,000] <inf> net_config: Interface 1 (0x20010bd8) coming up
[00:00:04.042,000] <inf> net_config: IPv4 address: 192.168.65.125
Hello World! nucleo_f746zg
uart:~$
uart:~$
uart:~$ net ping 192.168.65.16
PING 192.168.65.16
28 bytes from 192.168.65.16 to 192.168.65.125: icmp_seq=0 ttl=128 time=0 ms
28 bytes from 192.168.65.16 to 192.168.65.125: icmp_seq=1 ttl=128 time=0 ms
28 bytes from 192.168.65.16 to 192.168.65.125: icmp_seq=2 ttl=128 time=0 ms
```
**ETH_STM32_HAL_API_V2=y**
```
*** Booting Zephyr OS build 7796596e9bfd ***
[00:00:00.010,000] <inf> net_config: Initializing network
[00:00:00.010,000] <inf> net_config: Waiting interface 1 (0x20010c58) to be up...
[00:00:01.700,000] <inf> net_config: Interface 1 (0x20010c58) coming up
[00:00:01.701,000] <inf> net_config: IPv4 address: 192.168.65.125
Hello World! nucleo_f746zg
[00:00:03.102,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.102,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.102,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.102,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.102,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.102,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.102,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.102,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.102,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:03.103,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:05.253,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
[00:00:06.303,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080
uart:~$ net ping 192.168.65.16
PING 192.168.65.16
Ping timeout
[00:00:20.458,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8480
[00:00:20.458,000] <err> eth_stm32_hal: HAL_ETH_ErrorCallback errorcode:8 dmaerror:8080 <-- This msg repeats infinitly.
uart:~$
```
The version of Zephyr I using is 3.2.99.   
prj.conf what I used is very similar to @Desvauxm-st 's one. I just tweaked IP address and removed IPv6 things.   
I expected ethernet working on either legacy or V2.   

Is there anything I can do?

erwango, 2023-02-21:
Reopenning. @Desvauxm-st Can you have another look ?

Desvauxm-st, 2023-02-22:
Hi @easternblack

STM32F74X RMII interface does not work #46596

I have in charge to reproduce the issue you faced.
I compile the samples/hello_word with the following prj.conf:

# nothing here
CONFIG_ETH_STM32_HAL_MII=n
CONFIG_NETWORKING=y

CONFIG_NET_LOG=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=n
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_STATISTICS=y

CONFIG_NET_PKT_RX_COUNT=28
CONFIG_NET_PKT_TX_COUNT=14
CONFIG_NET_BUF_RX_COUNT=48
CONFIG_NET_BUF_TX_COUNT=28
CONFIG_NET_BUF_DATA_SIZE=1500
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=4
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=5
CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=1
CONFIG_NET_MAX_CONTEXTS=5
#CONFIG_NET_CONTEXT_RCVTIMEO=y
CONFIG_NET_TC_TX_COUNT=1
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_POLL_MAX=4
CONFIG_POSIX_MAX_FDS=8

CONFIG_INIT_STACKS=y
CONFIG_TEST_RANDOM_GENERATOR=y

CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_SHELL=y

CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.0.111"
CONFIG_NET_CONFIG_PEER_IPV4_ADDR="192.168.0.110"

CONFIG_LOG=y
CONFIG_SHELL_CMDS_RESIZE=n
CONFIG_ETH_STM32_HW_CHECKSUM=y
CONFIG_ETH_STM32_HAL_API_V2=y

And the result is like that : see file attachement:
[STM32F74X RMII interface does not wor1.docx](https://github.com/zephyrproject-rtos/zephyr/files/10803953/STM32F74X.RMII.interface.does.not.wor1.docx)

 

So I can’t reproduce the issue  until now
Could you give me your prj.conf you use please ?

easternblack, 2023-02-23:
Hello. @Desvauxm-st 
   
Here's my prj.conf what I used.   
I tested RMII on nucleo-f746zg.   
## prj.conf
```
CONFIG_ETH_STM32_HAL_MII=n
CONFIG_NETWORKING=y

CONFIG_NET_LOG=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=y
CONFIG_NET_DHCPV4=n
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y
CONFIG_NET_STATISTICS=y

CONFIG_NET_PKT_RX_COUNT=28
CONFIG_NET_PKT_TX_COUNT=14
CONFIG_NET_BUF_RX_COUNT=48
CONFIG_NET_BUF_TX_COUNT=28
CONFIG_NET_BUF_DATA_SIZE=1500
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=4
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=5
CONFIG_NET_IF_UNICAST_IPV4_ADDR_COUNT=1
CONFIG_NET_MAX_CONTEXTS=5
#CONFIG_NET_CONTEXT_RCVTIMEO=y
CONFIG_NET_TC_TX_COUNT=1
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
CONFIG_NET_SOCKETS_POLL_MAX=4
CONFIG_POSIX_MAX_FDS=8

CONFIG_INIT_STACKS=y
CONFIG_TEST_RANDOM_GENERATOR=y

CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_SHELL=y

CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_PEER_IPV6_ADDR="2001:db8::2"
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.65.125"
CONFIG_NET_CONFIG_MY_IPV4_GW="192.168.1.254"

CONFIG_LOG=y
CONFIG_SHELL_CMDS_RESIZE=n
#CONFIG_ETH_STM32_HW_CHECKSUM=y    <-- When this option lives, error occurs while Make.
CONFIG_ETH_STM32_HAL_API_V2=y
```
   
Here is specific VERSION of my zephyr.   
```
easternblack@easternblack-VirtualBox:~/zephyrproject/zephyr$ cat VERSION 
VERSION_MAJOR = 3
VERSION_MINOR = 2
PATCHLEVEL = 99
VERSION_TWEAK = 0
EXTRAVERSION =
```

The result of test is same as tesed above(https://github.com/zephyrproject-rtos/zephyr/issues/46596#issuecomment-1433961193)    
And I checked west.yml that if I used correct STM32 HAL V2 Driver. I followed this MR(https://github.com/zephyrproject-rtos/zephyr/pull/53193)   
```
    - name: hal_stm32
      revision: 83cf59e4309e5f1758a76c0513c437e0fbcee40b
      path: modules/hal/stm32
```
    
When I tweak option `ETH_STM32_HAL_API_V2=n`, It works well.
This problem occurs only when I use V2 API.

erwango, 2023-02-23:
> Here is specific VERSION of my zephyr.

This is is not precise enough. This could be any commit from 3.2.0 to 3.3.0. Please provide SHA1 of the base commit you're using from Zephyr tree.

easternblack, 2023-02-23:
My Zephyr is from my team's local git repo. unfortunately, I cannot get SHA of official Zephyr github's one. sorry.😢    
   
Instead of that, I tested same thing on Zephyr 3.3, it's SHA1 is e086d9d5d629388f5203a9f3225bd675b83bfd6c.   
   
And the consequence of the test is like below. I used same NUCLEO-F746ZG and prj.conf.
```
*** Booting Zephyr OS build zephyr-v3.3.0-375-ge086d9d5d629 ***
[00:00:00.011,000] <inf> net_config: Initializing network
[00:00:00.011,000] <inf> net_config: Waiting interface 1 (0x200113f0) to be up...
[00:00:01.505,000] <inf> net_config: Interface 1 (0x200113f0) coming up
[00:00:01.505,000] <inf> net_config: IPv4 address: 192.168.65.125
Hello World! nucleo_f746zg
[00:00:01.605,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:01.605,000] <inf> net_config: IPv6 address: 2001:db8::1
uart:~$
uart:~$
uart:~$ net ping 192.168.65.16
PING 192.168.65.16
Ping timeout
```
The repeating error msg is gone, but it's still not working.    
   
To verify this test, I also tested with `ETH_STM32_HAL_API_V2=n`   
```
*** Booting Zephyr OS build zephyr-v3.3.0-375-ge086d9d5d629 ***
[00:00:01.699,000] <inf> net_config: Initializing network
[00:00:01.699,000] <inf> net_config: Waiting interface 1 (0x20011370) to be up...
[00:00:01.724,000] <inf> net_config: Interface 1 (0x20011370) coming up
[00:00:01.724,000] <inf> net_config: IPv4 address: 192.168.65.125
Hello World! nucleo_f746zg
[00:00:01.824,000] <inf> net_config: IPv6 address: 2001:db8::1
[00:00:01.824,000] <inf> net_config: IPv6 address: 2001:db8::1
uart:~$
uart:~$
uart:~$ net ping 192.168.65.16
PING 192.168.65.16
28 bytes from 192.168.65.16 to 192.168.65.125: icmp_seq=1 ttl=128 time=0 ms
28 bytes from 192.168.65.16 to 192.168.65.125: icmp_seq=2 ttl=128 time=0 ms
28 bytes from 192.168.65.16 to 192.168.65.125: icmp_seq=3 ttl=128 time=0 ms
uart:~
```
And I successed on ping to desktop PC.

driechers, 2023-05-03:
I am having the same issue on nucleo_f756zg using the dhcp v4 sample. I have confirmed that switching back from v2 hal to v1 hal fixes my issue. The specific problem with the packet is that the icmp checksum is zero. I have traced through the net subsystem icmp code and found that the icmp checksum calculation and packet serialization has the correct checksum value. I have yet to trace through the HAL. I'm not sure however the correct path forward as I am still a zephyr noob. Is the HAL from STM or is the HAL created by zephyr community? If from STM do we carry patches to fix bugs? I'm happy to investigate further and submit a fix I just want to make the most helpful, forward facing fix.

erwango, 2023-05-03:
> I am having the same issue on nucleo_f756zg using the dhcp v4 sample. I have confirmed that switching back from v2 hal to v1 hal fixes my issue. The specific problem with the packet is that the icmp [...]

Thanks @driechers for investigating this.
Both HAL versions are STM. Rational for the new version is provided [here](https://community.st.com/s/question/0D53W00001Gi9BoSAJ/ethernet-hal-driver-reworked-by-st-and-available-in-22q1-preview-now-available-on-github).
We do backport issues from HAL. To be more specific we regularly update STM32 HALs from https://github.com/STMicroelectronics/ to https://github.com/zephyrproject-rtos/hal_stm32.

Also if we actually identify genuine HAL issues (and the issue is acknowledged by ST support), we can fix hal_stm32 waiting the fix to be delivered in official STM32Cube repos.

erwango, 2023-05-24:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/57639