zephyrproject-rtos/zephyr #46602: BLE paring/connection issue on Windows (Zephyr 3.1.0)

tgagneret-embedded, 2022-06-16:
**Describe the bug**
On Windows, it is impossible to reconnect to a BLE device previously paired on Zephyr 3.1.0. It was working with Zephyr 3.0.

I'm compiling the **samples/bluetooth/peripheral_hids** project on NRF5340DK.

**To Reproduce**
Steps to reproduce the behavior:
1.west build -b nrf5340_nrf5340dk_cpuapp -d build_app samples/bluetooth/peripheral_hids
2. west flash -d build_app
3. west build -b nrf5340dk_nrf5340_cpunet -d build_net samples/bluetooth/hci_rpmsg
4. west flash -d build_net
5. Try to pair your device on Windows
6. Reset the board (or disable and enable again bluetooth on your Windows)
7. The Windows PC is trying to connect indefinitely to the NRF5340 using BLE without success

**Expected behavior**
Windows PC should connect to the NRF5340 using BLE.

**Logs and console output**
```
Connected 9C:DA:3E:E5:E6:9B (public)
Security failed: 9C:DA:3E:E5:E6:9B (public) level 1 err 2
Disconnected from 9C:DA:3E:E5:E6:9B (public) (reason 0x13)
Connected 9C:DA:3E:E5:E6:9B (public)
Security failed: 9C:DA:3E:E5:E6:9B (public) level 1 err 2
Disconnected from 9C:DA:3E:E5:E6:9B (public) (reason 0x13)
Connected 9C:DA:3E:E5:E6:9B (public)
Security failed: 9C:DA:3E:E5:E6:9B (public) level 1 err 2
Disconnected from 9C:DA:3E:E5:E6:9B (public) (reason 0x13)
```

**Environment:**
 - Windows
 - Zephyr 3.1
 - 2ddd73feafd3316af2c547c34d6969bea637d5c6

**Additional context**
I found that the commit e11ff7df48bce7c98b05ff1626448cd8ab3d82c0 is causing the issue. Revert it allows the Windows PC to connect  successfully to the NRF5340 using BLE.

carlescufi, 2022-06-16:
@tgagneret-embedded could you please try with the latest main? I think this was fixed between 3.1.0 and the current main.

carlescufi, 2022-06-16:
@sjanc any idea?

carlescufi, 2022-06-16:
@tgagneret-embedded if this is still broken in `main`, could you try and bisect to find the commit that introduced the issue please?

tgagneret-embedded, 2022-06-16:
The issue is still there on main (416dd6cd369786dad096522715d16eb8712b2f68).

I didn't bissect, but debugging I saw that the code in smp_ident_addr_info could cause the issue.

For Windows, zephyr find a key when calling bt_keys_find_addr. So it calls bt_keys_clear. Thus when calling bt_keys_store, key is wrong. Using "git blame", reverting e11ff7df48bce7c98b05ff1626448cd8ab3d82c0 makes it work.

One weird thing, is that "SMP_FLAG_PAIRING" is set in smp flags when paring using Windows, but not when pairing with Android (I didn't check Linux, but behavior is probably the same than Android). The consequence is that when doing pairing with Windows, "bt_smp_update_keys" is executed but not with Android. 
So, when smp_ident_addr_info is called, bt_keys_find_addr finds the keys populated by bt_smp_update_keys, so it clears it.

I hope it helps.

PR #46505 (2022-06-17): bluetooth: host: Revert "bluetooth: host: Avoid sending duplicate dev…
…ice to resolving list."

This reverts commit e11ff7df48bce7c98b05ff1626448cd8ab3d82c0.
This patch causes some PTS failure (see https://github.com/zephyrproject-rtos/zephyr/pull/45510#issuecomment-1153606013)
Another patch to address duplicate device will follow.

Fixes #46602.
closed 2022-06-17 by carlescufi