zephyrproject-rtos/zephyr #46623: Promote user "tari" to traige permission level

keith-zephyr, 2022-06-16:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for an elevated permission level in
the Contributor role.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Peter Marheine
1. tari
1. Google

## Supporting Documents

[Zephyr authored PRs](https://github.com/pulls?q=is%3Apr+author%3Atari+archived%3Afalse+repo%3Azephyrproject-rtos%2Fzephyr+)
[Zephyr reviewed PRS](https://github.com/pulls?q=is%3Apr+reviewed-by%3Atari+archived%3Afalse+repo%3Azephyrproject-rtos%2Fzephyr+)

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-permission-levels-for-an-organization>

keith-zephyr, 2022-06-16:
@tari for visibility.

stephanosio, 2022-06-16:
Added as a contributor with triage permission.

closed 2022-06-16 by stephanosio

keith-zephyr, 2022-06-17:
Thanks for the quick response.  However, I am still unable to add @tari as a reviewer on an existing PR.  Are there other permissions that must be granted?

stephanosio, 2022-06-17:
> Are there other permissions that must be granted?

It seems @tari has not accepted the invitation yet.

keith-zephyr, 2022-06-17:
> > Are there other permissions that must be granted?
> 
> It seems @tari has not accepted the invitation yet.

Ok, thank you for the clarification.