zephyrproject-rtos/zephyr #46632: Intel CAVS: Assertion failures in tests/boards/intel_adsp/hda

yerabolu, 2022-06-16:
**Describe the bug**
Assertion failures on boards.intel_adsp.hda_host_out_smoke and boards.intel_adsp.1cpu.hda_host_out_smoke tests. 

Please also mention any information which could help others to understand
the problem you're facing:
 - intel_adsp_cavs15, intel_adsp_cavs18 and intel_adsp_cavs25

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W --hardware-map /home/ztest/cavs/cavs.map --device-testing -x=CONFIG_BOOT_DELAY=500 -T tests/boards/intel_adsp/hda/ --no-skipped-report -vvvv
2. see error

**Logs and console output**
```
START - test_hda_host_out_smoke
smoke testing hda with fifo buffer at address 0x9e021000, size 256
Using buffer of size 256 at addr 0x9e021000
dsp init: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:131 host_out(3:0x0x728c0), dgcs: 0x830000, dgbba 0x0, dgbs 0, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 0, dgbllpi 0x0, dglpibi 0x0
host reset: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:135 host_out(3:0x0x728c0), dgcs: 0x830000, dgbba 0x0, dgbs 0, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 0, dgbllpi 0x0, dglpibi 0x0
host config: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:141 host_out(3:0x0x728c0), dgcs: 0x830000, dgbba 0x0, dgbs 0, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 0, dgbllpi 0x0, dglpibi 0x0
dsp set buffer: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:144 host_out(3:0x0x728c0), dgcs: 0x830000, dgbba 0x21000, dgbs 256, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 256, dgbllpi 0x0, dglpibi 0x0
host start: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:149 host_out(3:0x0x728c0), dgcs: 0x830000, dgbba 0x21000, dgbs 256, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 256, dgbllpi 0x0, dglpibi 0x0
dsp enable: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:153 host_out(3:0x0x728c0), dgcs: 0x4830020, dgbba 0x21000, dgbs 256, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 256, dgbllpi 0x0, dglpibi 0x0
host send: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:163 host_out(3:0x0x728c0), dgcs: 0x4830b20, dgbba 0x21000, dgbs 256, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 256, dgbllpi 0x0, dglpibi 0x0
dsp wait for full: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:168 host_out(3:0x0x728c0), dgcs: 0x4830b20, dgbba 0x21000, dgbs 256, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 256, dgbllpi 0x0, dglpibi 0x0

Assertion failed at WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:187: test_hda_host_out_smoke: (is_ramp is false)
Expected data to be a ramp
FAIL - test_hda_host_out_smoke in 0.243 seconds
```

```
START - test_hda_host_out_smoke
smoke testing hda with fifo buffer at address 0xbe01e000, size 256
Using buffer of size 256 at addr 0xbe01e000
dsp init: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:131 host_out(3:0x0x728c0), dgcs: 0x830000, dgbba 0x0, dgbs 0, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 0, dgbllpi 0x0, dglpibi 0x0
host reset: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:135 host_out(3:0x0x728c0), dgcs: 0x830000, dgbba 0x0, dgbs 0, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 0, dgbllpi 0x0, dglpibi 0x0
host config: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:141 host_out(3:0x0x728c0), dgcs: 0x830000, dgbba 0x0, dgbs 0, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 0, dgbllpi 0x0, dglpibi 0x0
dsp set buffer: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:144 host_out(3:0x0x728c0), dgcs: 0x830000, dgbba 0x1e000, dgbs 256, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 256, dgbllpi 0x0, dglpibi 0x0
host start: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:149 host_out(3:0x0x728c0), dgcs: 0x830000, dgbba 0x1e000, dgbs 256, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 256, dgbllpi 0x0, dglpibi 0x0
dsp enable: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:153 host_out(3:0x0x728c0), dgcs: 0x4830020, dgbba 0x1e000, dgbs 256, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 256, dgbllpi 0x0, dglpibi 0x0
host send: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:163 host_out(3:0x0x728c0), dgcs: 0x4830b20, dgbba 0x1e000, dgbs 256, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 256, dgbllpi 0x0, dglpibi 0x0
dsp wait for full: WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:168 host_out(3:0x0x728c0), dgcs: 0x4830b20, dgbba 0x1e000, dgbs 256, dgbrp 0, dgbwp 0, dgbsp 0, dgmbs 256, dgbllpi 0x0, dglpibi 0x0

Assertion failed at WEST_TOPDIR/zephyr/tests/boards/intel_adsp/hda/src/smoke.c:187: test_hda_host_out_smoke: (is_ramp is false)
Expected data to be a ramp
FAIL - test_hda_host_out_smoke in 0.244 seconds
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) - Fedora
 - Toolchain (e.g Zephyr SDK, ...) - SDK 14.1
 - Commit SHA: 92bc3e47c0f8

smrtos, 2022-06-29:
The hda_buf[] data is expected to be a ramp after host sending data. But the dumped value is all 0.
Investigating why.

```
hda_buf[0] = 0
hda_buf[1] = 0
hda_buf[2] = 0
hda_buf[3] = 0
hda_buf[4] = 0
hda_buf[5] = 0
hda_buf[6] = 0
hda_buf[7] = 0
hda_buf[8] = 0

```

smrtos, 2022-06-30:
The cavstool.py and the tests/boards/intel_adsp/hda code interact through the IPC and DMA interfaces to complete the test.
There may be some misbehavior in cavstool.py when operating the register interface. Investigating the register specs.

smrtos, 2022-07-07:
Some details about the test scenario: a host side script cavstool.py is used to serve as both Intel HD Audio driver and audio app. It communicates with the test code running on the DSP. And sends a data[256] to DSP via the host DMA (not link DMA).

According to the cavs spec, the DGMBS is meant for scenario where SPIB (Stream Position In Buffer) operation is disabled, which means DMA will operate in legacy mode with the minimum FIFO size. 

This legacy mode is unreliable which shouldn't be used. In the cavsetool.py, we do enable the SPIB: https://github.com/zephyrproject-rtos/zephyr/blob/8d4bd6be3a15f3bdfae12a8b8f1005ce67fba635/soc/xtensa/intel_adsp/tools/cavstool.py#L129-L130

And the DGMBS should just be an informational register which is effective only when SPIB is disabled.

But it seems it goes beyond being just informational and block the host DMA transfer.

Still investigating this with the audio team.

smrtos, 2022-07-08:
Maybe fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/47548

commit e243122c (2022-07-11): dma/cavs_hda: Configure DGMBS only for host DMA configuration

The DGMBS should be only set for the host side and not on the link DMA.

Fixes: #46632