zephyrproject-rtos/zephyr #46635: twister: test blocks on tests/subsys/modbus

shaoanxli, 2022-06-17:
**Describe the bug**
When running tests/subsys/modbus with twister command , the twister process blocks and shows "Waiting for device frdm_k64f to become available".

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. twister -p frdm_k64f --device-testing --hardware-map map.yml -T tests/subsys/modbus/   -vv
3. See the blocked process.

**Expected behavior**
The test should pass or fail.

**Impact**
The test blocks so that the test can not finish automatically.

**Logs and console output**
```
DEBUG   - run test: frdm_k64f/tests/subsys/modbus/modbus.rtu
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available
DEBUG   - Waiting for device frdm_k64f to become available

```

**Environment:**
OS: Linux,
Toolchain Zephyr SDK 0.14.1
Commit SHA or Version: f232db463078ff3e8a85ceeb9e9fcc0c14e141e3

shaoanxli, 2022-06-22:
> @shaoanxli Would you like to reopen the PR.

https://github.com/zephyrproject-rtos/zephyr/pull/46599

This is not the root cause of this twister bug , so I close it. I think if we do not add fixture to hardware-map , the twister may log warning "should add fixture to hardware-map file", and skip this test.

hakehuang, 2022-06-22:
@shaoanxli  shall we close this issue? it is the test setting issue. or will you create a poll request to add the harness?

shaoanxli, 2022-06-22:
> @shaoanxli shall we close this issue? it is the test setting issue. or will you create a poll request to add the harness?

I think it is fine to close this issue. This pr may fix this #45263 .

closed 2022-06-22 by shaoanxli