zephyrproject-rtos/zephyr #46717: L2CAP ECRED feature should support response with result code 0x000D defined in core stack 5.3 spec

yajunzhi, 2022-06-21:
About 
For test case L2CAP/ECFC/BV-43-C and L2CAP/ECFC/BV-44-C,  it need to response with result 0x000F or 0x000D, and then perform user authorization after sending connection response. 
After authorization, also need send response packet. And the request sending device should  waiting for connection response.
The procedure is like BR/EDR connection.

If zephyr protocol stack can consider adding this process, it will be able to pass the case certification.

hermabe, 2022-06-21:
Do note that there is an open TSE for these to tests: https://www.bluetooth.org/tse/errata_view.cfm?errata_id=18351.

yajunzhi, 2022-06-22:
Got it. 
That is to say, the code could be modified when the TSE is closed.

jori-nordic, 2024-02-09:
@hermabe is this still valid?

hermabe, 2024-02-09:
> @hermabe is this still valid?

Looks like the errata is not released in a TCRL yet. But the "pending" status codes are not relevant for LE, so this can probably be closed.

closed 2024-02-09 by jori-nordic