zephyrproject-rtos/zephyr #46721: HAL module request: hal_renesas

andrzej-kaczmarek, 2022-06-21:
## Origin

New repository that will contain required parts of Renesas SDKs (currently
available for registered user only) for supported MCUs.

## Purpose

Provides HAL and/or supporting code for Renesas MCUs.

## Mode of integration

Integrate as a HAL module, i.e. hal_renesas.

## Pull Request

n/a

## Description

This provides relevant SDK files to support Renesas MCUs in Zephyr.
Initially, registers definitions for DA1469x family will be stored here that
already have compatible licenses and are also used by e.g. Apache Mynewt:

```
~/.../modules/hal/renesas$ tree
.
├── CMakeLists.txt
├── da1469x
│   ├── CMakeLists.txt
│   └── sdk
│       └── bsp
│           └── include
│               ├── DA1469xAB.h
│               └── system_DA1469x.h
└── zephyr
    └── module.yml
```

More files may be added in future if required and relicensed by Renesas, e.g.
for peripherals support or new Renesas MCU families support.

# Dependencies

None.

## Revision

None, initial push with separate PR.

## License

``Apache-2.0``
``BSD-3-Clause``

andrzej-kaczmarek, 2022-06-21:
PR to add support for Renesas DA1469x: https://github.com/zephyrproject-rtos/zephyr/pull/46720

andrzej-kaczmarek, 2022-06-21:
@blauret @NiekIlmer FYI

carlescufi, 2022-06-21:
Thanks, can you confirm that you will maintain this HAL for the foreseeable future?

andrzej-kaczmarek, 2022-06-27:
@carlescufi yes, I will maintain and extend both HAL and SoC port, there will be also Renesas engineers involved later at some point to take over.

andrzej-kaczmarek, 2022-06-29:
Yes, at the moment this is what initial port for DA1469x requires and this part of SDK was already relicensed to 3BSD. The other parts will need to be relicensed if they are required for subsequent features and then added here, I assume.

As I understand 3BSD license prohibits those SoC register definitions from being included directly into main Zephyr tree, thus I reuqested a HAL module for current and future work on Renesas SoCs.

carlescufi, 2022-06-29:
> Yes, at the moment this is what initial port for DA1469x requires and this part of SDK was already relicensed to 3BSD. The other parts will need to be relicensed if they are required for subsequent [...]

That is actually not accurate, as per the [external components contribution guide](https://docs.zephyrproject.org/latest/contribute/external.html) it is possible to contribute 3-clause BSD code into the main tree, so I'd recommend to re-evaluate the fact you want to use a module here:

- Are you gonna add more files later to the HAL?
- Or will you just be writing Zephyr-specific drivers?

Depending on what you plan for the future, you should probably consider just pulling `DA1469xAB.h` directly into the tree?

andrzej-kaczmarek, 2022-07-04:
At the moment there are only Zephyr-specific drivers, but I expect that code from SDK can be reused if possible at later stage (they will need to be relicensed first). Also the plan is to enable CMAC (i.e. integrated Bluetooth controller) and this will require firmware binaries so I think HAL module is the place (?) to store such files.

So I agree that there are not much files here to keep in HAL module, and I cannot say that there will be many more files added there soon-ish, but perhaps it's good to have a place to store those files.

Also one problem with e.g. `DA1469xAB.h` is that it is sourced directly from SDK, so it does not really conform to any Zephyr coding guidelines. Perhaps it would need to be generated in some other way if integrating into Zephyr tree is the only viable option to proceed here.

carlescufi, 2022-07-06:
@andrzej-kaczmarek could you please update the body of this issue with the URL and instructions to download the SDK from the Renesas website?

carlescufi, 2022-07-06:
TSC: no objections, repo will be created.

andrzej-kaczmarek, 2022-07-12:
@carlescufi overlooked that comment before, updated now

carlescufi, 2022-08-16:
https://github.com/zephyrproject-rtos/hal_renesas

closed 2022-08-16 by carlescufi