zephyrproject-rtos/zephyr #46759: Process: low priority bug management

mbolivar-nordic, 2022-06-22:
The purpose of this issue is to continue discussion begun at the TSC about how to handle low priority bugs.

We have been unable to hit our low priority bug targets for a few releases now, and have agreed to do the following things as a project at the TSC level:

1. Ask maintainers for help triaging bugs into "will fix this release" vs. "known issue with no plans for a fix" categories (this applies to all bugs, not just low priority bugs)
2. Establish new criteria for low priority bugs in a release; likely just counting those in the "will fix in this release" category (this does **not** apply to medium and high priority bugs; the total amount of those open will still affect release readiness as it does today, regardless of the category from 1.)
3. Establish timelines and processes for when we do this triaging
4. Establish timelines and processes for when we revisit bug counts from the the "known issue" category, so things don't get out of hand

The details of how this will work were left to the process WG to sort out. This issue will track the resulting discussion and document the resulting recommendations for the TSC.

mbolivar-nordic, 2022-06-22:
Process WG:

- @mbolivar-nordic : review of issue and request for additional items
- @nashif: this relates to a more general discussion of issue management in general, and release criteria in particular. Not sure if having numbers in place for release criteria gives value. E.g. if you have 51 lows and you need 50, how do you decide what to fix? This is not how quality should be managed. We introduced an idea of a 'blocker' issue which hasn't been moved into processes. We also haven't established clear criteria for what qualifies as 'low' priority - low priority for the project as a whole because it only affects one platform ignores the possibility it may be a serious issue for that platform.
- @mbolivar-nordic : my goal here is to start a process-based feedback loop between project level questions and maintainer level triaging, which we can iteratively improve to tackle the larger issues you mention
- @nashif : sounds fine to me
- @mbolivar-nordic : propose to use 'milestone' for bugs with a target fix, and label 'known-issue' for latter category
- @nashif : 'will fix in next release' should be handled via priority. E.g. 'medium' must be addressed via next release. 'low' could mean 'address later'.
- @keith-zephyr : does using milestone vnow.next remove it from our metrics?
- @mbolivar-nordic not right now. @nashif can you summarize your proposal?
- @nashif : I don't want low priority bugs to be ignored, but they should not block the release. I also do not want the count of mediums to block the release either. We need to shift to a mode where we fix critical issues based on what the users or the developers put in front of us. Want to make bugs a blocker regardless of their priority.
- @stephanosio at this point you are talking about severity as well as priority
- @mbolivar-nordic still unclear on the proposal - remove all counts?
- @nashif yes: could release with a high, for example.
- @mbolivar-nordic then I don't see the value of priority; seems like needless overhead for maintainers; would want to switch to severity only for release criteria
- @nashif priority could be for the vendor or maintainer. e.g. if there is a bug that only affects an intel board, then the priority could still be high for the users and maintainers of that board.
- @stephanosio priorities could be used as a reference for maintainers of a specific area, but severity for release criteria
- @mbolivar-nordic not sure what the value to the project is of processes around priorities if they do not affect releases
- @nashif asking people to maintain priority and severity is challenging. Maybe we need to recognize that priority is being misused, switch to severity, and see what evolves. E.g. say at individual milestones within a release, allow all bug fixes by milestone N, only high severity fixes by milestone N+1, etc.
- @nashif my main point is that we need to move away from quotas and move to tracking bugs by severity

AIs:

- @nashif to open slides 
- all to review async and discuss on tsc channel on discord
- regroup next week

marc-hb, 2022-06-29:
> asking people to maintain priority and severity is challenging. 

Indeed most people can't seem to make the difference. I think that's because they usually go together despite being defined very differently. I force myself to remember some rare counter-examples where they don't go together:

- A complete lockup in a feature that no customer uses yet (high severity, low priority)
- Some branding issue with no technical or legal consequence (low severity, high priority)
- A "VIP" bug = low impact defect but reported by someone popular on social media.

> Maybe we need to recognize that priority is being misused, switch to severity, and see what evolves

Interesting, that would make a lot of sense to me:

- Priorities have nothing to do with quality; they're a project management tool and only relative to each other. As in: "when everything is high priority then nothing is".
- Is Zephyr project management across all companies really a thing? Inside each contributing company yes of course but globally? I'm trying to imagine someone from company A telling someone in company B to stop working on something because it's lower priority than something else.

Afraid priority and severity will keep being frequently confused as long as both are in use.

My 2 cents.

> Ask maintainers for help triaging bugs into "will fix this release" vs. "known issue with no plans for a fix" categories

In some trackers, project managers want open issues to represent only work that they expect to be done in the not too distant future, closing all the other ones as "wontfix". With the help of tools like this:

https://docs.github.com/en/actions/managing-issues-and-pull-requests/closing-inactive-issues

In simpler terms: open issues = TODO list in those trackers.

I read the quoted sentence as a clarification that Zephyr bugs will NOT be managed as a TODO list. In other words, a low severity Zephyr bug stays open as long as it can be reproduced, even when no one has any plan to fix it currently?

mbolivar-nordic, 2022-07-20:
Process WG:

- @mbolivar-nordic it looks like the "Release blockers" slide is the main proposal for this issue: https://docs.google.com/presentation/d/10Q-B0Upv1bBaoQRNM5n9yUEMeXO6NINnFNDkA86RCsM/edit#slide=id.ga4343a65c2_1_0
- @nashif that and slide 6 "release criteria"
- @stephanosio my concern is assigning blockers will take lengthy discussions on a per-subsystem basis
- @dleach02 I also do like the metrics; we've just been discovering that we have problems with lows
- @stephanosio I agree having separate support tiers would help
- @nashif let me summarize: testing for v3.2, keep the criteria, but change the numbers for low priority bugs to not perpetually make exceptions for release criteria. right?
- @mbolivar-nordic that's my priority number one
- @nashif ok, and second thing, the scripts you've worked on, to try to look into more detailed bug breakdowns, probably at TSC level, and sending emails and notifications to try to bring bugs down, right?
- @mbolivar-nordic not exactly; I want to get understanding of what bugs are targeted for this release, and which aren't

Outcome:

- for v3.2, process WG will recommend to TSC to double the low priority bug count criterion to <100 for release
- target v3.3 for wider scope of handling severity as well as priority, reference platforms, per-area quality criteria, etc

mbolivar-nordic, 2022-07-27:
Process WG:

 - @nashif will add recommendation to move to 100 for v3.2 to TSC next week
 - @dleach02 : Anas noted that we are now adding release blockers to v3.2 using the corresponding GitHub label
 - @mbolivar-nordic : should we also note to the TSC that we're adding release blockers as part of the recommendations to the TSC?
 - @nashif: I think that's worth it. We can also mark low priority bugs as release blockers.
 - @dleach02 to some extent the counts are arbitrary
 - @nashif: for historical context, these numbers came out of an internal policy from Intel
- @mbolivar-nordic I'd be ok with using this time to talk about 3.3 release criteria
- discussion moved to https://github.com/zephyrproject-rtos/zephyr/issues/38566

mbolivar-nordic, 2022-08-10:
Process WG:

- @mbolivar-nordic at TSC, we concluded that for v3.2, the change is to stop counting "future" milestone low priority bugs, but the release criteria counts of bugs don't change
- @mbolivar-nordic other v3.3 changes?
- @nashif yes, dropping bug requirements, introducing release blockers and severity vs priority, etc. Will come up with a modified proposal for August 31.

mbolivar-nordic, 2022-08-17:
Process WG: nothing to add

mbolivar-nordic, 2022-09-07:
Process WG:

- presentation from @nashif, "Copy of Release Criteria and Issue Management V2"
- @keith-zephyr additional release criterion we're not tracking is reproducible builds
- @mbolivar-nordic I can take the tier 1 and subsystem specific test cases discussion internally
- @MaureenHelm proposal makes a lot of sense, but I'm still concerned that we'll end up with a lot of bugs that don't get fixed because they aren't blockers

marc-hb, 2022-09-07:
> presentation from @nashif, "Copy of Release Criteria and Issue Management V2"

In case anyone is looking for this: it's still the same URL as the one at the very top: https://docs.google.com/presentation/d/10Q-B0Upv1bBaoQRNM5n9yUEMeXO6NINnFNDkA86RCsM/

I don't remember whether it started as V1 or Vanything but right now it's definitely called "Copy of Release Criteria and Issue Management V2"

mbolivar-nordic, 2022-09-21:
Process WG:

- OK to move to "awaiting TSC approval" based on content of @nashif's presentation given 2 weeks ago

keith-zephyr, 2022-09-21:
RFC for adding reproducible build check to the CI is on https://github.com/zephyrproject-rtos/zephyr/issues/50205

fabiobaltieri, 2022-09-29:
Hey was going through the slides: since the new criteria have a strong depenency on tests passing on Tier 1 hardware, but the Tier 1 testing infrastructure would be owned by the vendor (unless I misunderstood), should we have some more detailed and explicit requirement on what's expected from the vendor team to keep the platform at tier? Trying to be defensive about unresponsive vendor maintainers who may be the only one with access to the specific testing infrastructure, even for problems like flaky tests. Maybe we could demand to have at least N ponit of contacts that can access those boards, or asking that the release team is given access to run/rerun tests remotely.

fabiobaltieri, 2022-09-29:
About Severity/Priority, I understand that severity would do a better job to classify the issue, but then what do we do with the information? At the end of the day an issue either blocks the release or doesn't, and we already have two tags that do that (priority:high and release blocker, I'd argue we shoudl get rid of one). What do do with the severity bit of information? I'm afraid we'd end up wasting time arguing whether something is "major" or "critical" and then do nothing with it. (my background: I worked with bug tracking system with priority and severity, and haven't seen severity used once)

mbolivar-nordic, 2022-09-29:
>  Trying to be defensive about unresponsive vendor maintainers who may be the only one with access to the specific testing infrastructure, even for problems like flaky tests

Commitment to responsiveness is a requirement for your board to be in tier 1.

The "stick" we can use as a project to enforce responsiveness that goes along with the "carrot" of being featured more prominently in tier 1 is that you can drop to tier 3 if you're not responsive.

mbolivar-nordic, 2023-03-08:
We've had TSC approval for this in principle for a while, and have now moved the implementation work to the testing WG's domain

closed 2023-03-08 by mbolivar-nordic