zephyrproject-rtos/zephyr #46765: test-ci: kernel.timer: test_timer_remaining asserts

hakehuang, 2022-06-22:
**Describe the bug**
test_timer_remaining test fails on mimxrt595_evk_cm33

    ===================================================================
    START - test_timer_remaining
    
        Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:718: test_timer_remaining: (abs(delta_ticks) <= MAX(slew_ticks, 1U) is false)
    tick/busy slew -2 larger than test threshold 0
     FAIL - test_timer_remaining in 0.71 seconds

 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mimxrt595_evk_cm33 ..
3. make
4. See error

**Expected behavior**
Test PASS

**Impact**
kernel timer

**Logs and console output**

```
*** Booting Zephyr OS build zephyr-v3.1.0-583-g83e4f5dd5866  ***
    Running TESTSUITE timer_api
    ===================================================================
    START - test_time_conversions
     PASS - test_time_conversions in 0.146 seconds
    ===================================================================
    START - test_timer_duration_period
     PASS - test_timer_duration_period in 0.327 seconds
    ===================================================================
    START - test_timer_restart
     PASS - test_timer_restart in 0.277 seconds
    ===================================================================
    START - test_timer_period_0
     PASS - test_timer_period_0 in 0.203 seconds
    ===================================================================
    START - test_timer_period_k_forever
     PASS - test_timer_period_k_forever in 0.203 seconds
    ===================================================================
    START - test_timer_expirefn_null
     PASS - test_timer_expirefn_null in 0.327 seconds
    ===================================================================
    START - test_timer_periodicity
     PASS - test_timer_periodicity in 0.252 seconds
    ===================================================================
    START - test_timer_status_get
     PASS - test_timer_status_get in 0.2 seconds
    ===================================================================
    START - test_timer_status_get_anytime
     PASS - test_timer_status_get_anytime in 0.327 seconds
    ===================================================================
    START - test_timer_status_sync
     PASS - test_timer_status_sync in 0.352 seconds
    ===================================================================
    START - test_timer_k_define
     PASS - test_timer_k_define in 0.702 seconds
    ===================================================================
    START - test_timer_user_data
     PASS - test_timer_user_data in 0.302 seconds
    ===================================================================
    START - test_timer_remaining
    
        Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:718: test_timer_remaining: (abs(delta_ticks) <= MAX(slew_ticks, 1U) is false)
    tick/busy slew -2 larger than test threshold 0
     FAIL - test_timer_remaining in 0.71 seconds
    ===================================================================
    START - test_timeout_abs
     PASS - test_timeout_abs in 0.2 seconds
    ===================================================================
    START - test_sleep_abs
     PASS - test_sleep_abs in 0.7 seconds
    ===================================================================
    TESTSUITE timer_api failed.
    ===================================================================
    PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: zephyr-v3.1.0-583-g83e4f5dd5866

mmahadevan108, 2022-09-15:
@hakehuang, can you please try this on `3.2-rc1`. I cannot reproduce this failure.

hakehuang, 2022-09-16:
@mmahadevan108 , I can still  repoduce this issue on my board with latest rc1 weekly build

```
  *** Booting Zephyr OS build v3.2.0-rc1-18-g53af1ba59e29  ***

    Running TESTSUITE timer_api

    ===================================================================

    START - test_sleep_abs

     PASS - test_sleep_abs in 0.007 seconds

    ===================================================================

    START - test_time_conversions

     PASS - test_time_conversions in 0.151 seconds

    ===================================================================

    START - test_timeout_abs

     PASS - test_timeout_abs in 0.002 seconds

    ===================================================================

    START - test_timer_duration_period

     PASS - test_timer_duration_period in 0.327 seconds

    ===================================================================

    START - test_timer_expirefn_null

     PASS - test_timer_expirefn_null in 0.327 seconds

    ===================================================================

    START - test_timer_k_define

     PASS - test_timer_k_define in 0.702 seconds

    ===================================================================

    START - test_timer_period_0

     PASS - test_timer_period_0 in 0.203 seconds

    ===================================================================

    START - test_timer_period_k_forever

     PASS - test_timer_period_k_forever in 0.203 seconds

    ===================================================================

    START - test_timer_periodicity

     PASS - test_timer_periodicity in 0.252 seconds

    ===================================================================

    START - test_timer_remaining

    

        Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:716: timer_api_test_timer_remaining: (abs(delta_ticks) <= MAX(slew_ticks, 1U) is false)

    tick/busy slew -2 larger than test threshold 0

     FAIL - test_timer_remaining in 0.072 seconds

    ===================================================================

    START - test_timer_restart

     PASS - test_timer_restart in 0.277 seconds

    ===================================================================

    START - test_timer_status_get

     PASS - test_timer_status_get in 0.002 seconds

    ===================================================================

    START - test_timer_status_get_anytime

     PASS - test_timer_status_get_anytime in 0.327 seconds

    ===================================================================

    START - test_timer_status_sync

     PASS - test_timer_status_sync in 0.352 seconds

    ===================================================================

    START - test_timer_user_data

     PASS - test_timer_user_data in 0.302 seconds

    ===================================================================

    TESTSUITE timer_api failed.
```

mmahadevan108, 2022-09-16:
Thats strange, I am seeing the below on my RT595 board:
```
*** Booting Zephyr OS build v3.2.0-rc1  ***
Running TESTSUITE timer_api
===================================================================
START - test_sleep_abs
 PASS - test_sleep_abs in 0.007 seconds
===================================================================
START - test_time_conversions
 PASS - test_time_conversions in 0.118 seconds
===================================================================
START - test_timeout_abs
 PASS - test_timeout_abs in 0.002 seconds
===================================================================
START - test_timer_duration_period
 PASS - test_timer_duration_period in 0.327 seconds
===================================================================
START - test_timer_expirefn_null
 PASS - test_timer_expirefn_null in 0.327 seconds
===================================================================
START - test_timer_k_define
 PASS - test_timer_k_define in 0.702 seconds
===================================================================
START - test_timer_period_0
 PASS - test_timer_period_0 in 0.202 seconds
===================================================================
START - test_timer_period_k_forever
 PASS - test_timer_period_k_forever in 0.203 seconds
===================================================================
START - test_timer_periodicity
 PASS - test_timer_periodicity in 0.252 seconds
===================================================================
START - test_timer_remaining
 PASS - test_timer_remaining in 0.052 seconds
===================================================================
START - test_timer_restart
 PASS - test_timer_restart in 0.277 seconds
===================================================================
START - test_timer_status_get
 PASS - test_timer_status_get in 0.002 seconds
===================================================================
START - test_timer_status_get_anytime
 PASS - test_timer_status_get_anytime in 0.327 seconds
===================================================================
START - test_timer_status_sync
 PASS - test_timer_status_sync in 0.352 seconds
===================================================================
START - test_timer_user_data
 PASS - test_timer_user_data in 0.302 seconds
===================================================================
TESTSUITE timer_api succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [timer_api]: pass = 15, fail = 0, skip = 0, total = 15 duration = 3.452 seconds
 - PASS - [timer_api.test_sleep_abs] duration = 0.007 seconds
 - PASS - [timer_api.test_time_conversions] duration = 0.118 seconds
 - PASS - [timer_api.test_timeout_abs] duration = 0.002 seconds
 - PASS - [timer_api.test_timer_duration_period] duration = 0.327 seconds
 - PASS - [timer_api.test_timer_expirefn_null] duration = 0.327 seconds
 - PASS - [timer_api.test_timer_k_define] duration = 0.702 seconds
 - PASS - [timer_api.test_timer_period_0] duration = 0.202 seconds
 - PASS - [timer_api.test_timer_period_k_forever] duration = 0.203 seconds
 - PASS - [timer_api.test_timer_periodicity] duration = 0.252 seconds
 - PASS - [timer_api.test_timer_remaining] duration = 0.052 seconds
 - PASS - [timer_api.test_timer_restart] duration = 0.277 seconds
 - PASS - [timer_api.test_timer_status_get] duration = 0.002 seconds
 - PASS - [timer_api.test_timer_status_get_anytime] duration = 0.327 seconds
 - PASS - [timer_api.test_timer_status_sync] duration = 0.352 seconds
 - PASS - [timer_api.test_timer_user_data] duration = 0.302 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

hakehuang, 2022-09-19:
maybe we can close it, and let me find a solid way to reproduce it.

closed 2022-09-19 by hakehuang