zephyrproject-rtos/zephyr #46823: L2CAP the param of function bt_l2cap_ecred_chan_reconfigure()

yajunzhi, 2022-06-23:
**Current define:**
int bt_l2cap_ecred_chan_reconfigure(struct bt_l2cap_chan **chans, uint16_t mtu);
Expect:
int bt_l2cap_ecred_chan_reconfigure(struct bt_l2cap_chan **chans, uint16_t mtu, uint16_t mps) 

**considered**
If the reconfigure interface could support mps param, some verification from PTS could be passed. Because some test cases need IUT send different MTU/MPS value parameter. There is no way to send the reconfigure cmd with the modifiable MPS.

The related test case is L2CAP/ECFC/BV-24-C.

hermabe, 2022-06-23:
L2CAP/ECFC/BV-24-C passed in latest weekly PTS run. From the TS:
> If TSPX_l2ca_cbmps_min equals TSPX_l2ca_cbmps_max, the test ends with a Pass verdict

This is done [here](https://github.com/intel/auto-pts/blob/683b4272ddcbfa011a5e1087070ec31db10077be/autopts/ptsprojects/zephyr/l2cap.py#L140), allowing the test to pass.

Adding the mps parameter would be a stable API change and would need to follow the process described [here](https://docs.zephyrproject.org/latest/develop/api/api_lifecycle.html#introducing-incompatible-changes).

yajunzhi, 2022-06-23:
I modified the SPX_l2ca_cbmps_min and TSPX_l2ca_cbmps_max as the same,  and tested. PTS ends with PASS as you mentioned.
About the API, who submit the RFC better? Current, the passed verification is our requirement. 
Thank you.

theob-pro, 2023-02-16:
@yajunzhi doesn´t the comment from @hermabe address your concerns? ie, is there a real need for an API change, except to pass a test that already seems to be passing?

yajunzhi, 2023-02-17:
Hi, 
I think the L2CAP_CREDIT_BASED_RECONFIGURE_REQ packe should provide in zephyr for some scenarios and application. And EATT may use the API to change MTU size. Actually, it is not one essential requirement currently.
Thank you.

jori-nordic, 2023-02-21:
Alright, closing the issue then.

closed 2023-02-21 by jori-nordic