zephyrproject-rtos/zephyr #46844: Timer drivers likely have off-by-one in rapidly-presented timeouts

andyross, 2022-06-23:
(This one is sort of half way between a bug and an enhancement, labelling it a bug for early triage but feel free to redefine.)

Discussion in #46639 showed that the arm_arch_timer had an edge case in the rounding logic such that a system that got into a timer interrupt and back out into `sys_clock_set_timeout()` before the hardware counter had advanced would end up setting the resulting interrupts a cycle too early.  Even that is mostly benign (as it just produces a spurious interrupt that then reschedules the correct timeout at the next tick), but for the case of "next interrupt" (i.e. a ticks argument of 0 or 1) it would compute an expiration time of "now", which the hardware can't do.

There's nothing particularly weird about that driver, it's doing the same rounding logic almost all our other drivers are doing.  So it's likely that almost all our other drivers are subject to the same glitch, even if most of them have clock rates or interrupt latencies that prevent them from ever hitting it in practice.

In the near term, we should probably organize an audit of the existing drivers to make sure they all do the right thing (maybe a white-boxy unit test framework might help, but it would have to be able to indirect the hardware counter read, which is fiddly).

Longer term, it's clear that the bulk of our drivers are implementing the same basic architecture, with a free running up-counter and a comparator register.   We should probably look at providing a framework for them all to use so they can share the same cycle counting and tick rounding logic.

carlescufi, 2022-06-23:
@nordic-krch FYI

andyross, 2022-08-15:
This is definitely a "future" thing.  The more I think about it the more it seems like we need a single framework for a "standard up-counter with comparator" driver to cover these situations.  The hardware would provide the MIN_CYCLES parameter (how close to "now" can a timeout be set), API implementations for "current_cycles()" and "set_timeout()", and a build-time parameter for whether the counter was 64 or 32 bits wide (probably don't need to do smaller masked counters, as the only one in tree currently -- SysTick -- is non-standard anyway).  Then the framework would handle everything else in a standard way.

andyross, 2022-10-15:
Un-stale.  This still needs to happen in some form.

closed 2022-12-30 by github-actions[bot]