zephyrproject-rtos/zephyr #46864: Intel CAVS: cavstool_client.py sporadically fails

laurenmurphyx64, 2022-06-23:
**Describe the bug**
cavstool_client.py randomly sends garbage (?) to cavstool_server.py (in Zephyr codebase, it's named cavstool.py), which tries to flash the garbage to the ADSP. West fails with no console output. Twister receives no console output for alternating tests, causing every Twister run to fail.

**To Reproduce**

Use West to flash `hello_world` or run multiple tests with Twister. Observed on all CAVS platforms with APL, WHL and TGL DUTs.

I initially thought that if I retried West or used Twister to run West multiple times in a row too fast, I could reliably trigger this bug. However, I've also seen it happen the very first time I start the client and server.

**Expected behavior**
Successful West / Twister run.

**Impact**
Showstopper. Twister does not work on our shared developer CAVS hardware.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 (on DUT server and client developer machine)
 - Toolchain: Zephyr SDK
 - (Base) commit SHA: 7f6b762a734b793a031103e12bbce865645d4f90

**Logs and console output**

On developer machine:
```
$ west build -p -b intel_adsp_cavs15 samples/hello_world
...
$ west flash --remote-host <DUT IP>
...
INFO:cavs-client:Download process
INFO:cavs-client:Sent:     download
INFO:cavs-client:Receive: download
INFO:cavs-client:Sent fw:     zephyr.ri.f1c78357c1445ea0e9f3bc78877e6ba3, /home/herbert/zephyr-intel/zephyr/build/zephyr/zephyr.ri.f1c78357c1445ea0e9f3bc78877e6ba3
INFO:cavs-client:Sending...
INFO:cavs-client:Done Sending (65552).
INFO:cavs-client:Monitor process
INFO:cavs-client:Sent:     start_log
INFO:cavs-client:Receive: start_log
INFO:cavs-client:Start to monitor log output...
--
```

On APL DUT:
```
sudo ./cavs_server.p -s <DUT IP>
INFO:cavs-fw:Req server start...
INFO:cavs-fw:Log server start...
INFO:cavs-fw:192.168.0.200 wrote: b'download'
INFO:cavs-fw:192.168.0.200 wrote: b'zephyr.ri.f1c78357c1445ea0e9f3bc78877e6ba3XMan\x10\x00\x00\x00\x10\x00\x00\x00\x00\x00\x00\x01$CPD\x03\x00\x00\x00\x01\x01\x10\xa6ADSPADSP.man\x00\x00\x00\x00X\x00\x00\x00x\x03\x00\x00\x00\x00\x00\x00cavs0015.met\x00\x04\x00\x00`\x00\x00\x00\x00\x00\x00\x00cavs0015\x00\x00\x00\x00\x80\x04\x00\x00\x80\xfb\x00\x00\x00\x00\x00\x00\x04\x00\x00\x00\xa1\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x86\x80\x00\x00#\x06" \xde\x00\x00\x00$MN2\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00@\x00\x00\x00\x01\x00\x00\x00\x85\x00\xe1h\xaa\xeb\xd2\x07\x1b|^\xed\xd6\xe7\xe5\xf9\xc1\x0eG\xd4L\xab\x8c\xf0\xe8\xee\x8b@65X\x8f\xf4o\xfc\xfd\x0f\xddU\x8bE\x8c\xf0G\xdc\xb4\xac!;K \xe6\x81\xb3\xcc\x90\xd4^\xf1\xa4\x9bhR\xc8\xf1-\xf9\xc4w\xc6M\xa9\x90\xc7\x10\xfdC\xc8Kk#^\x92\xf5\x8f\xac\xd5}`\'6|!N!\x99\xde\xcb\xc0E\xf3\x04"\xb8}\x16h@\xf9\\\xf0\xb9~\x8c\x05\xb6\xfc(\xbb=\xd8\xff\xb6\xa4\xd4T\';\x1aBN\xf5\xa6\xa8^D\xe2\x9e\xedhj\'`\x13\x8d/\'p\xcdW\xc9\x18\xa3\xb00\xa1\xf4\xe62\x12\x89*\xaf@\xa5\xfdR\xf1\xaa\x8a\xa4\xef =\x10\xa3p\xf29\xc5\x05\x99"\x10\x81\x83nE\xa4\xf3Z\x9dj\xb8\x88\xfei@\xd1\xb1\xcb*\xdb(\x05\xdeT\xbf=\x86_9\x8b\xc1\xf4\xaf\x00a\x86\x01\xfa"\xac\xf6,\xa4\x17j\xa7\xd8\n\x8c\x9f\xbf\x1fb\xb2.hR?\x82\x8f\xe5(M\xdb\xb5Z\x96(\'\x19\xafC\xb9\x01\x00\x01\x00I\xe7\xd0\xe0*\x99\xc8M\x14\xf6{(Z\\B\xbd\xa1E\xed\xc1E\xd5*z\xc9Z!n\xac\\\xce\x03\tN \\\xc8WB\xf02\x92,\xae\x8f{yr;w\xdd\x17P(\xb5js@\x97\x97V?\x93\x16\x86\xfd\xbbVz\xf9\x89;F\xbb\x01Hz\xfa\xa0\xa9\x84!fL,\xa4\xf3\x8d\xf3\xb3\x9e\xc0h\xfd\x0b\x11\xe3\\\xb3\xa0\xbb\x19\x06\x01\xa8n~\xe2\xe4\x1c[J\tM3}7)fv\x8c=Y\xae\xd5\xe9\xc5\xd7I\xb4\xdfO\x89\xa4\xeex\x00]\x03\xf2V\x05x\xaa\x16g\xab}\xbb\xfb\xe6\xc6*!\x00\xa5^\xc9\xcc\x1c"#\xd0)\xc3x\x10>2\x13\x0c\xb4U\x80\x1d\xd0Eog\xc2~~\xa9\xad+sz\x16\x93+p\x17}\xd6-N5\x0cW\x05\xde\x19&v.\xd3q\x7f\xd4\x83p\x84\nrQkw\xa1H\xa8\ro\xfb\xd2\x91\x8f\xc3\xa1q\xaf/Q\x9b\xfe\x8e/\x8f\xb4V\xb7\xbev;\x93}\xafD\xaa\\Je^]\xac\\\xb4\x0f\x00\x00\x00h\x00\x00\x00ADSP\x00\x00\x00\x00\x00\x00\x00\x00\x08\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00cavs0015.met\x03\x02 \x00`\x00\x00\x00\xc3\xf5\x04d\t\xe2\xfaa_\xf6L.]\xb3\xa2\xb98+\xd9\x9aR+\x96U\x90\xa6r\xf8\xe7\xb9\xee\xbe\x03\x00\x00\x00\x8c\x00\x00\x00ADSP\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x10\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xffcavs0015.met\x03\x00\xff\xff`\x00\x00\x00\xc3\xf5\x04d\t\xe2\xfaa_\xf6L.]\xb3\xa2\xb98+\xd9\x9aR+\x96U\x90\xa6r\xf8\xe7\xb9\xee\xbe\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11\x00\x00\x00`\x00\x00\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x95.,\xc9\'\xb1\xb1_\x13\xfdU\x03\x8bVu\xb1K,\xb6\x05\x1c\xa6,\xed"\';\xa4\xb7\x12@$\x00 \x00\x00\x80\x1b\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x [... cut ...]
INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
INFO:cavs-fw:Starting DSP, ADSPCS = 0x303
INFO:cavs-fw:Unresetting DSP cores, ADSPCS = 0x3030000
INFO:cavs-fw:Running DSP cores, ADSPCS = 0x3030000
INFO:cavs-fw:Wait for ROM startup, ADSPCS = 0x3030000
INFO:cavs-fw:Sending IPC command, HIPIDR = 0x81004e00
INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
INFO:cavs-fw:Waiting for firmware handoff, FW_STATUS = 0x5000001
WARNING:cavs-fw:Load failed?  FW_STATUS = 0x80000006
INFO:cavs-fw:cAVS firmware load complete
```

enjiamai, 2022-06-24:
The PR https://github.com/zephyrproject-rtos/zephyr/pull/46880 also partially fix this hang up problem when the garbage happens, at least twister won't hang up in this case.

The garbage is the "Filename" + "part of bytes the ADSP firmware" which are both transferred by socket.sendall/sendfile. In the code, they are sent separately but we saw them being received together.  This happens frequently (10~15%) in some environments.

Anyway, the file transfer should be more robust, I am trying on improving the send/receive file logic. I tried the use the lock but it seems like the symptom could be mitigated but cannot be fully solved.  Consider doing check and resending the firmware, or just use SCP to transfer the firmware binary.

laurenmurphyx64, 2022-06-24:
@enjiamai Is there a chance we could invoke [SCP in the Python script](https://stackoverflow.com/a/68365) to transfer the firmware binary? I don't mind re-running West, that's totally fine, but every Twister run I've tried fails and that seems like a really mission-critical problem that needs to be solved. Maybe it could look something like this:

CLIENT, OVER TCP: Hey, I'm going to send the binary
SERVER, OVER TCP: OK, send it
CLIENT, USING PYTHON SUBPROCESS TO INVOKE SCP: *Sends the file*
CLIENT, OVER TCP: I sent the binary, did you get it?
SERVER, OVER TCP: I got it, gonna go ahead and flash it, thanks

enjiamai, 2022-06-24:
Ok, @laurenmurphyx64 , I think SCP is a good choice. go!

enjiamai, 2022-06-26:
Hi @laurenmurphyx64 , #46880 can address all these problem. I refine the way we transfer FW files and the whole logic, and I tested the new logic work quite stable now, no matter sending FW file or hanging up problem, and after applying it I rarely see a false-positive test. 

I could add scp later if you think we still need it.  If we add scp, I think it will look this way: (we need to specify the username and ssh port both)
```
west flash --remote-host={host_name}:{req_port}:{log_port}   --scp={user}:{ssh_port}
```

I appreciate all your help these days!

marc-hb, 2022-06-27:
I spent a few hours debugging these crashes (without being aware of this github issue) and it looks like the protocol was simply missing delimiters or lengths. So different messages were merged together (TCP has no boundaries).

Repeating what I just wrote in #46880, longer term low-level serialization should not be re-invented here. This is a very common need and there are off-the-shelf solutions for this that avoid typical bugs and save a lot of time. After a 5 minutes search I found:

Python remote objects https://pyro4.readthedocs.io/en/stable/tutorials.html
https://docs.jsonrpc.ru/quickstart.html#a-minimal-application

There may be more options.

closed 2022-07-07 by nashif

marc-hb, 2022-09-12:
Yet another raw socket issue:
- https://github.com/zephyrproject-rtos/zephyr/pull/50071#issuecomment-1242722148

Raw sockets: just say no. Don't design your own network protocol, always re-use some existing one. Just like parsers.

cc: @smrtos @teburd

PR #46880 (2022-07-07): soc: xtensa: partial fix of socket misuse and refine the code
This PR fix bug and refine the client-server CAVS tool, includes:

- Improve the firmware transfer reliability by fixing the misuse of the socket.  Fix the most frequent occurence of the common `recv()` bug described here: https://docs.python.org/3/howto/sockets.html#using-a-socket
The longer term fix is to switch to a higher level API like Python Remote Objects.

- Not rely on the client's command to disconnect. Previously we rely on the SIGINT to send stop_command to the server, but it does not work well in some environments. Refi [... cut]
commit faff3f7e (2022-07-07): soc: xtensa: partial fix of socket misuse and refine the code

1. Improve the firmware transfer reliability by fixing the misuse
of the socket. Fix the most frequent occurence of the common `recv()`
bug described here:

https://docs.python.org/3/howto/sockets.html#using-a-socket

The longer term fix is to switch to a higher level API like Python
Remote Objects.

2. Not rely on the client's command to disconnect. Previously we
rely on the SIGINT to send stop_command to the server, but it does
not work well in some environments. Refine the whole logic and the
sever disconnect service by checking [... cut]