zephyrproject-rtos/zephyr #46910: lib: libc: picolibc: time.h definitions and declarations

cfriedt, 2022-06-26:
**Is your enhancement proposal related to a problem? Please describe.**
Due to the slight overlap between ISO C and POSIX standards when it comes to the `<time.h>` header, some additional declarations are needed by the libc.

The main motivation for this is to eliminate the ~fragile workaround [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/posix/time.h#L109), which also does some weird things with `#include_next <time.h>` to account for the `<posix/time.h>` header being included before `<time.h>` and vice-versa.
```cpp
#else /* ZEPHYR_INCLUDE_POSIX_TIME_H_ */
/* Read the toolchain header when <posix/time.h> finds itself on the
 * first attempt.
 */
#include_next <time.h>
#endif /* ZEPHYR_INCLUDE_POSIX_TIME_H_ */
```

Since `<time.h>` is primarily a libc header, the simpler solution is to *not* maintain a conflicting header with `<posix/time.h>` and instead to minimally declare some POSIX-related constructs in the libc header.

**Describe the solution you'd like**
As shown in https://github.com/zephyrproject-rtos/zephyr/pull/43987, the minimal libc has some additional declarations in `<time.h>` to enable `CONFIG_POSIX_CLOCK`.
```
#ifdef CONFIG_POSIX_CLOCK
#define TIMER_ABSTIME 4
typedef enum {
	CLOCK_MONOTONIC,
	CLOCK_REALTIME,
} clockid_t;
typedef unsigned long timer_t;

struct itimerspec {
	struct timespec it_interval;
	struct timespec it_value;
};

int clock_gettime(clockid_t clockid, struct timespec *res);
int clock_settime(clockid_t clockid, const struct timespec *tp);

int timer_create(clockid_t clockid, struct sigevent *sevp, timer_t *timerid);
int timer_delete(timer_t timerid);
int timer_gettime(timer_t timerid, struct itimerspec *its);
int timer_settime(timer_t timerid, int flags, const struct itimerspec *value,
	struct itimerspec *ovalue);

int nanosleep(const struct timespec *req, struct timespec *rem);
#endif
```

**Describe alternatives you've considered**
The status quo, with the fragile `time.h` workaround and conflicting `<posix/time.h>` header.

**Additional context**

keith-packard, 2022-06-26:
Looks like picolibc could use a section in its sys/features.h about zephyr -- picolibc has a huge list of optional system APIs that it can declare, including these time-related ones. These, in particular, are declared if  _POSIX_TIMERS is #define'd.

Here's the list of things which the cygwin section includes; we can add a zephyr section that sets the correct value for each one.

```
#define _POSIX_ADVISORY_INFO			200809L
#define _POSIX_ASYNCHRONOUS_IO			200809L
#define _POSIX_BARRIERS				200809L
#define _POSIX_CHOWN_RESTRICTED			     1
#define _POSIX_CLOCK_SELECTION			200809L
#define _POSIX_CPUTIME				200809L
#define _POSIX_FSYNC				200809L
#define _POSIX_IPV6				200809L
#define _POSIX_JOB_CONTROL			     1
#define _POSIX_MAPPED_FILES			200809L
/* #define _POSIX_MEMLOCK			    -1 */
#define _POSIX_MEMLOCK_RANGE			200809L
#define _POSIX_MEMORY_PROTECTION		200809L
#define _POSIX_MESSAGE_PASSING			200809L
#define _POSIX_MONOTONIC_CLOCK			200809L
#define _POSIX_NO_TRUNC				     1
/* #define _POSIX_PRIORITIZED_IO		    -1 */
#define _POSIX_PRIORITY_SCHEDULING		200809L
#define _POSIX_RAW_SOCKETS			200809L
#define _POSIX_READER_WRITER_LOCKS		200809L
#define _POSIX_REALTIME_SIGNALS			200809L
#define _POSIX_REGEXP				     1
#define _POSIX_SAVED_IDS			     1
#define _POSIX_SEMAPHORES			200809L
#define _POSIX_SHARED_MEMORY_OBJECTS		200809L
#define _POSIX_SHELL				     1
#define _POSIX_SPAWN				200809L
#define _POSIX_SPIN_LOCKS			200809L
/* #define _POSIX_SPORADIC_SERVER		    -1 */
#define _POSIX_SYNCHRONIZED_IO			200809L
#define _POSIX_THREAD_ATTR_STACKADDR		200809L
#define _POSIX_THREAD_ATTR_STACKSIZE		200809L
#define _POSIX_THREAD_CPUTIME			200809L
/* #define _POSIX_THREAD_PRIO_INHERIT		    -1 */
/* #define _POSIX_THREAD_PRIO_PROTECT		    -1 */
#define _POSIX_THREAD_PRIORITY_SCHEDULING	200809L
#define _POSIX_THREAD_PROCESS_SHARED		200809L
#define _POSIX_THREAD_SAFE_FUNCTIONS		200809L
/* #define _POSIX_THREAD_SPORADIC_SERVER	    -1 */
#define _POSIX_THREADS				200809L
#define _POSIX_TIMEOUTS				200809L
#define _POSIX_TIMERS				200809L
/* #define _POSIX_TRACE				    -1 */
/* #define _POSIX_TRACE_EVENT_FILTER		    -1 */
/* #define _POSIX_TRACE_INHERIT			    -1 */
/* #define _POSIX_TRACE_LOG			    -1 */
/* #define _POSIX_TYPED_MEMORY_OBJECTS		    -1 */
#define _POSIX_VDISABLE				   '\0'
```

keith-packard, 2022-06-26:
oh look, there's already a document which has what we need -- doc/services/portability/posix.rst:
```
    _POSIX_CLOCK_SELECTION,
    _POSIX_FSYNC,
    _POSIX_MEMLOCK,
    _POSIX_MEMLOCK_RANGE,
    _POSIX_MONOTONIC_CLOCK,
    _POSIX_NO_TRUNC,
    _POSIX_REALTIME_SIGNALS,
    _POSIX_SEMAPHORES,yes
    _POSIX_SHARED_MEMORY_OBJECTS,
    _POSIX_SYNCHRONIZED_IO,
    _POSIX_THREAD_ATTR_STACKADDR,
    _POSIX_THREAD_ATTR_STACKSIZE,
    _POSIX_THREAD_CPUTIME,
    _POSIX_THREAD_PRIO_INHERIT,yes
    _POSIX_THREAD_PRIO_PROTECT,
    _POSIX_THREAD_PRIORITY_SCHEDULING,yes
    _POSIX_THREAD_SPORADIC_SERVER,
    _POSIX_TIMEOUTS,
    _POSIX_TIMERS,
    _POSIX2_C_DEV,
    _POSIX2_SW_DEV,
```

keith-packard, 2022-06-26:
So, this leaves me with the (seemingly simple) question -- what cpp symbol should picolibc use to detect when it is being used within zephyr? `#ifdef ZEPHYR_VERSION`?

keith-packard, 2022-06-26:
And, looks like I can answer that myself -- the build system adds -D__ZEPHYR__=1 on the command line.

keith-packard, 2022-06-26:
hrm. well, that's not going to work -- clock_gettime is defined with __syscall linkage in the zephyr header files. Suggestions on what picolibc should do are welcome here; do we need a magic __syscall thingy in the picolibc headers? I'd have to define that as '' by default to make things work, ofc.

cfriedt, 2022-06-26:
> So, this leaves me with the (seemingly simple) question -- what cpp symbol should picolibc use to detect when it is being used within zephyr? `#ifdef ZEPHYR_VERSION`?

I think `__ZEPHYR__` is typical

cfriedt, 2022-06-26:
> hrm. well, that's not going to work -- clock_gettime is defined with __syscall linkage in the zephyr header files. Suggestions on what picolibc should do are welcome here; do we need a magic __syscall thingy in the picolibc headers? I'd have to define that as '' by default to make things work, ofc.

That's a good question. Perhaps @stephanosio and @tejlmand might want to chime in.

keith-packard, 2022-06-27:
ok, so I've got a picolibc PR that enables all of the zephyr-supported POSIX interfaces here: [Picolibc #305](https://github.com/picolibc/picolibc/pull/305). This is also in the zephyr picolibc repo on the[ zephyr-posix branch](https://github.com/zephyrproject-rtos/picolibc/tree/zephyr-posix). That'll get merged to picolibc once zephyr is ready to adapt its interfaces, so you're driving this one.

cfriedt, 2022-06-27:
Perfect!

closed 2025-06-18 by cfriedt