zephyrproject-rtos/zephyr #46917: frdm_k64f : run test tests/drivers/gpio/gpio_get_direction failed

shaoanxli, 2022-06-27:
**Describe the bug**
Two bugs here , in test_disconnect and test_input_output function , shows "gpio_pin_configure() failed: -134".
It seems that these two flags are not recognized: "GPIO_INPUT | GPIO_OUTPUT | GPIO_OUTPUT_INIT_LOW"  , "GPIO_DISCONNECTED" .

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. twister -p frdm_k64f --device-testing  --hardware-map map.yml -T tests/drivers/gpio/gpio_get_direction/ -vv
3. See error

or

2. west build -p always -b frdm_k64f tests/drivers/gpio/gpio_get_direction/
3. screen /dev/ttyACM0 115200
4. west flash
5. See error

**Expected behavior**
Test case should pass as expected.

**Impact**
The testsuite gpio_get_direction failed.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.1.0-812-g88ca3aca98c0  ***
Running TESTSUITE gpio_get_direction
===================================================================
START - test_disconnect
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_get_direction/src/main.c:20: test_disconnect: (0 not equal to rv)
gpio_pin_configure() failed: -134
FAIL - test_disconnect in 0.15 seconds
===================================================================
START - test_input
PASS - test_input in 0.1 seconds
===================================================================
START - test_output
PASS - test_output in 0.1 seconds
===================================================================
START - test_input_output
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_get_direction/src/main.c:92: test_input_output: (0 not equal to rv)
gpio_pin_configure() failed: -134
FAIL - test_input_output in 0.16 seconds
===================================================================
TESTSUITE gpio_get_direction failed.
===================================================================
RunID: c8e7e0ad19f305178da49e64f6b23b25
PROJECT EXECUTION FAILED
```
**Environment :**
OS: Linux
Toolchain Zephyr SDK 0.14.1
Commit SHA or Version: 88ca3aca98c0788e2c45db5c3fb9c8523293a8c0

shaoanxli, 2022-06-28:
> @shaoanxli - can you please check out and test the fix in #46960?

The test went fine after adding this modification : 26e551d7db4dc98cc935ac0e4e162e47c57cb1bd . But there are still some issues with this commit : e26d4f3adf9006348b7efaff82420ed6f85f7611 .

[image]

PerMac, 2022-06-28:
I also see this test failing on all nrf boards on our farm. In my case the output is
```
DEBUG   - DEVICE: START - test_disconnect
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/gpio/gpio_get_direction/src/main.c:20: test_disconnect: (0 not equal to rv)
DEBUG   - DEVICE: gpio_pin_configure() failed: -5
DEBUG   - DEVICE: FAIL - test_disconnect in 0.15 seconds

```

cfriedt, 2022-06-28:
> I also see this test failing on all nrf boards on our farm. In my case the output is
> ```
> DEBUG   - DEVICE: START - test_disconnect
> DEBUG   - DEVICE:
> DEBUG   - DEVICE: Assertion failed at [...]

@PerMac - you or someone at Nordic should add get direction support to the nRF driver. I only implemented it for a few platforms.

shaoanxli, 2022-06-29:
@cfriedt The latest commit can not skip the test "test_input_output" .

cfriedt, 2022-06-29:
> @cfriedt The latest commit can not skip the test "test_input_output" .

Ack- copy-paste error. It went into the work g function. Will fix

cfriedt, 2022-06-29:
Actually, @PerMac it seems that `gpio_nrfx.c` (at least the driver, not sure about the hardware) does not support setting gpio disconnect (at least). Do you know whether it supports setting simultaneous input-output?

It seems rather than reporting `-ENOTSUP`, the `gpio_nrfx.c` driver returns `-EIO`, which is likely a driver bug.
```
 * @retval 0 If successful.
 * @retval -ENOTSUP if any of the configuration options is not supported
 *                  (unless otherwise directed by flag documentation).
 * @retval -EINVAL Invalid argument.
 * @retval -EIO I/O error when accessing an external GPIO chip.
 * @retval -EWOULDBLOCK if operation would block.
 */
__syscall int gpio_pin_configure(const struct device *port,
```

See https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/gpio.h#L681

PerMac, 2022-06-29:
@anangl can you please chip in? It is beyond my scope :(

anangl, 2022-06-29:
> Actually, @PerMac it seems that `gpio_nrfx.c` (at least the driver, not sure about the hardware) does not support setting gpio disconnect (at least). Do you know whether it supports setting simultaneous input-output?

@cfriedt The gpio_nrfx driver supports simultaneous input-output pin configuration. It also supports the `GPIO_DISCONNECTED` flag, not perfectly though, as it incorrectly returns the `-EIO` error if the pin to be configured with `GPIO_DISCONNECTED` was not earlier configured as input or output. I can provide a fix for that - you can assign #47030 to me, unless you prefer to solve it yourself.

PR #46960 (2022-07-02): tests: drivers: gpio: skip unsupported tests
Some drivers may not support simultaneous input-output configuration or disconnect.

In such cases, the driver should return `-ENOTSUP` and the test should be skipped.

Fixes #46917

* Also update the suite to use the new ztest API
* Also correct error codes for related functions in `gpio.h`
commit e46afe75 (2022-07-02): tests: drivers: gpio: skip unsupported tests / drivers

Some drivers may not support simultaneous input-output
configuration or disconnect.

In such cases, the driver should return `-ENOTSUP` and the
test should be skipped.

Fixes #46917
closed 2022-07-02 by cfriedt