zephyrproject-rtos/zephyr #46925: Intel CAVS: tests/lib/mem_block/ failed, possibly caused by winstream buffer size

LixinGuoX, 2022-06-27:
**Describe the bug**
On Intel_adsp_cavs platform, `tests/lib/mem_block/` failed due to missing the early log like below:

We manually increase the `HP_SRAM_WIN3_SIZE` to from `0x2000 `to `0x4000` can resolve this issue temporarily, but we are not sure if this new value is safe, need further reference to spec or other.

```
Prefixing /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs18/tests/lib/mem_blocks/lib.mem_blocks/zephyr/zephyr.ri with manifest /home/ztest/zephyrproject/zephyr/twister-out/intel_adsp_cavs18/tests/lib/mem_blocks/lib.mem_blocks/zephyr/zephyr.ri.xman

DEBUG   - DEVICE: --
DEBUG   - DEVICE: 4  A3 0x9e01f560
DEBUG   - DEVICE: E:  **  A4 0x1  A5 0x60020  A6 0x9e01f628  A7 0x60c20
DEBUG   - DEVICE: E:  **  A8 0x1  A9 (nil) A10 0x1 A11 0x9e01f5dc
DEBUG   - DEVICE: E:  ** A12 0x9e01f000 A13 0x57 A14 0xbe01c07b A15 0x10
DEBUG   - DEVICE: E:  ** LBEG (nil) LEND (nil) LCOUNT (nil)
DEBUG   - DEVICE: E:  ** SAR 0x4
DEBUG   - DEVICE:
DEBUG   - DEVICE:
DEBUG   - DEVICE: Backtrace:0xBE0138EF:0xBE01C140 0xBE012E45:0xBE01C150 0xBE013402:0xBE01C170 0xBE010BD9:0xBE01C1A0 0xBE0144DA:0xBE01C1D0 0xBE011E4B:0xBE01C1F0
DEBUG   - DEVICE:
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
DEBUG   - DEVICE: E: Current thread: 0x9e01f000 (test_mem_block_invalid_params_p)
DEBUG   - DEVICE: Caught system error -- reason 4
DEBUG   - DEVICE: PASS - test_mem_block_invalid_params_panic_1 in 0.3 seconds
```

**To Reproduce**
Steps to reproduce the behavior:
1. `twister -W --hardware-map ~/cavs/cavs.map --device-testing -v  -T tests/lib/mem_blocks/ -vv`

**Expected behavior**
Test should pass.

**Environment (please complete the following information):**
 - OS: Linux
 - Commit b034711f595ec51f2055cc7a623d4668c116d169

smrtos, 2022-07-13:
Checking the spec to decide if the increased size is safe.
And we also tried to increase the frequency of reading logs from winstream buffer in hope that logs can be extracted out before wrapping. But no obvious effect.

smrtos, 2022-07-14:
We decided to remove the excessive log output. May be fixed #47779.

nashif, 2022-07-28:
@smrtos Can you please confirm that this test actually passes and only caputering the results is not working correctly, if so, then it can be closed as a non-bug in the test itself, and point to the CAVS issue and the test enahancement. But please you mark this internally and do not open the same bug again.

smrtos, 2022-07-29:
Yes, this is a non-bug in the test itself. I am closing this one and track it with https://github.com/zephyrproject-rtos/zephyr/issues/47833. Thanks.

closed 2022-07-29 by smrtos

smrtos, 2022-08-31:
Fixed by:
https://github.com/zephyrproject-rtos/zephyr/pull/49721
https://github.com/zephyrproject-rtos/zephyr/pull/49546