zephyrproject-rtos/zephyr #46954: Binaries found in hal_nxp without conspicuous license information

rahulmohang, 2022-06-27:
Hello, 

There are a number of binaries in some of the folders. e.g. 

https://github.com/zephyrproject-rtos/hal_nxp/blob/master/mcux/mcux-sdk/middleware/mmcau/asm-cm0p/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/middleware/mmcau/asm-cm4-cm7/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/middleware/issdk/algorithms/pedometer/lib/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/CMSIS/RTOS2/RTX/Library/GCC/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/CMSIS/DSP/Lib/GCC/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54018/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54018/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC55S28/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC55S28/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC5514/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC5514/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54607/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54607/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC5512/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC5512/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54S018M/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54S018M/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54016/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54016/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54618/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54618/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54005/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54005/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54606/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54606/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC55S26/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC55S26/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC55S66/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC55S66/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54628/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54628/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC5526/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC5526/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54018M/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54018M/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54608/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54608/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54616/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54616/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC55S14/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC55S14/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54605/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54605/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC51U68/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC51U68/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54S018/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54S018/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC55S69/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC55S69/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC5516/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC5516/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC5528/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC5528/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC55S16/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC55S16/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54S005/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54S005/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54114/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54114/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54113/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54113/mcuxpresso/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54S016/gcc/*.a
https://github.com/zephyrproject-rtos/hal_nxp/tree/master/mcux/mcux-sdk/devices/LPC54S016/mcuxpresso/*.a

In general, I have noticed the statement, 'SPDX-License-Identifier: BSD-3-Clause' in many files in the folders. Does BSD-3-Clause apply to the binaries? If yes, is the license a BSD-3-Clause with a mention of NXP in the disclaimer?

fabiobaltieri, 2022-06-28:
This should be moved to the hal_nxp repository.

rahulmohang, 2022-06-28:
Thank you @fabiobaltieri and @dleach02.

closed 2022-07-05 by mmahadevan108