zephyrproject-rtos/zephyr #46985: uOSCORE/uEDHOC integration as a Zephyr module

rlubos, 2022-06-28:
## Origin

uOSCORE / uEDHOC

https://github.com/eriptic/uoscore-uedhoc/tree/dev

## Purpose

The repository contains C implmentations of [OSCORE ](https://datatracker.ietf.org/doc/html/rfc8613) and [EDHOC ](https://datatracker.ietf.org/doc/html/draft-ietf-lake-edhoc-12) protocols.

OSCORE is an end-to-end security protocol for CoAP. It can be considered as an alternative for DTLS security for CoAP. What makes it different from DTLS though is that OSCORE provides a full end-to-end security, even if CoAP/HTTP proxy is in use. Contrary to DTLS, OSCORE does not requisite for a packet to be decrypted at the proxy node.

EDHOC is a lightweight key exchange protocol. According to the RFC, a main objective for EDHOC is to be a lightweight authenticated key exchange for OSCORE.

## Mode of integration

I'd like to integrate the libraries as a module. Both uOSCORE and uEDHOC were developed with Zephyr as one of the target platforms in mind, so the repository integrates with Zephyr seamlessly - i. e. no downstream patches are needed to integrate libraries with Zephyr. As the libraries (especially uEDHOC) may still be a subject of change (i. e. aligning with newer spec versions), it'd be easier to bring updates to Zephyr if we retain the repository structure, instead of integrating the libraries directly into Zephyr tree.
I suggest to keep the original repository name, i. e. `uoscore-uedhoc`.

## Maintainership

I propose myself as the module maintainer. We can add the original author (@StefanHri) as a collaborator or a co-maintainer, if he agrees to become one. @StefanHri please let us know.

## Pull Request

https://github.com/zephyrproject-rtos/zephyr/pull/46983

## Description

We've been discussing at Nordic possible alternatives for DTLS security for LwM2M. As some background information, DTLS is not very effective in LTE/NB-IoT networks, where short NAT timeouts, resulting in IP address changes, invalidate DTLS sessions quickly, making this protocol not very friendly for low power devices.

One possible solution for this problem is use of the OSCORE protocol as an alternative to DTLS. When looking for viable options we've encountered the usocore-uedhoc library (https://github.com/eriptic/uoscore-uedhoc), back then developed as a project at Fraunhofer university, which has already been already developed to be compatible with Zephyr.

An alternative solution for the aforementioned problem is to use DTLS connection ID extension, ideally, we could support both options.

What is its primary functionality:
* OSCORE - end to end security for CoAP
* EDHOC - key exchange algorithm for OSCORE

What problem are you trying to solve?
* Need a OSCORE library sutiable for embedded Devices,
* LwM2M specifies OSCORE-based transport as an alternative to DTLS, the library could be used to implement it.

Why is this the right component to solve:
* Acceptable FLASH/RAM requirements, backed by [official paper](https://arxiv.org/pdf/2103.13832.pdf),
* Developed with Zephyr in mind - integrates seamlessly,
* Compatible license

## Dependencies

* mbed TLS (from Zephyr)
* zcbor (from Zephyr)
* [compact25519](https://github.com/DavyLandman/compact25519/tree/a5abe6fdf36a13237d674c848b6ccf73efcd5ad4) - optional, extends uEDHOC functinality with additional cipher suites. For now this functionality is disabled so no need to bring this in.

## Revision

Current state of the [dev branch](https://github.com/eriptic/uoscore-uedhoc/tree/dev) ([e8920192b66db4f909eb9cd3f155d5245c1ae825](https://github.com/eriptic/uoscore-uedhoc/commit/e8920192b66db4f909eb9cd3f155d5245c1ae825))

## License

Dual license:
* ``Apache-2.0``
* ``MIT``

StefanHri, 2022-06-28:
Great! Please add me as co-maintainer

carlescufi, 2022-07-01:
https://github.com/zephyrproject-rtos/uoscore-uedhoc

closed 2022-07-01 by carlescufi