zephyrproject-rtos/zephyr #46988: samples: net: openthread: coprocessor: RCP is missing required capabilities: tx-security tx-timing

ck-telecom, 2022-06-28:
**Describe the bug**
otbr-agent failed while check NCP caps

Please also mention any information which could help others to understand
get latest otbr-posix build and run on RPi 3B 
```
Jun 28 14:32:45 raspberrypi otbr-agent[866]: [NOTE]-AGENT---: Running 0.3.0-efc940cdaa
Jun 28 14:32:45 raspberrypi otbr-agent[866]: [NOTE]-AGENT---: Thread version: 1.2.0
Jun 28 14:32:45 raspberrypi otbr-agent[866]: [NOTE]-AGENT---: Thread interface: wpan0
Jun 28 14:32:45 raspberrypi otbr-agent[866]: [NOTE]-AGENT---: Backbone interface: eth0
Jun 28 14:32:45 raspberrypi otbr-agent[866]: [NOTE]-AGENT---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0
Jun 28 14:32:45 raspberrypi otbr-agent[866]: [INFO]-NCP-----: OpenThread log level changed to 4
Jun 28 14:32:45 raspberrypi otbr-agent[866]: 49d.17:04:41.378 [I] Platform------: RCP reset: RESET_POWER_ON
Jun 28 14:32:45 raspberrypi otbr-agent[866]: 49d.17:04:41.384 [N] Platform------: RCP API Version: 6
Jun 28 14:32:45 raspberrypi otbr-agent[866]: 49d.17:04:41.386 [C] Platform------: RCP is missing required capabilities: tx-security tx-timing 
Jun 28 14:32:45 raspberrypi otbr-agent[866]: 49d.17:04:41.386 [C] Platform------: CheckRadioCapabilities() at radio_spinel_impl.hpp:391: RadioSpinelIncompatible
Jun 28 14:32:45 raspberrypi systemd[1]: otbr-agent.service: Main process exited, code=exited, status=3/NOTIMPLEMENTED
Jun 28 14:32:45 raspberrypi systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
```

**To Reproduce**
Steps to reproduce the behavior:
```
west build -p auto -b nrf52840_mdk_usb_dongle samples/net/openthread/coprocessor/ -DOVERLAY_CONFIG="overlay-usb-nrf-br.conf overlay-rcp.conf" -DDTC_OVERLAY_FILE="usb.overlay"
```
4. See error

**Expected behavior**
RCP is working corrrectly

**Impact**

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA latest version

**Additional context**
even I add the following patch, still failed
```
diff --git a/samples/net/openthread/coprocessor/prj.conf b/samples/net/openthread/coprocessor/prj.conf
index da5ef9f637..68f55d6e77 100644
--- a/samples/net/openthread/coprocessor/prj.conf
+++ b/samples/net/openthread/coprocessor/prj.conf
@@ -1,5 +1,6 @@
 CONFIG_NETWORKING=y
 CONFIG_NET_L2_OPENTHREAD=y
+CONFIG_OPENTHREAD_THREAD_VERSION_1_2=y
```

rlubos, 2022-06-29:
It's not a bug but a missing capability of a driver you're using. I see you use nRF device, so:
* `tx-security` should be covered with `CONFIG_NRF_802154_ENCRYPTION=y` (I'm not sure if the option has been tested in vanilla Zephyr context though).
* `tx-timing` is not supported by the upstream nRF 802.15.4 driver. So you either need to disable this feature on the BR side somehow, or switch to nRF Connect SDK, where the driver comes with full feature set.

ck-telecom, 2022-06-29:
Thanks for your explain!
```
zephyr/subsys/net/lib/openthread/platform/radio.c

otRadioCaps otPlatRadioGetCaps(otInstance *aInstance)
{
    ...
#if defined(CONFIG_IEEE802154_2015)
        if (radio_caps & IEEE802154_HW_TX_SEC) {
                caps |= OT_RADIO_CAPS_TRANSMIT_SEC;
        }
#endif

#if defined(CONFIG_NET_PKT_TXTIME)
        if (radio_caps & IEEE802154_HW_TXTIME) {
                caps |= OT_RADIO_CAPS_TRANSMIT_TIMING;
        }
#endif
    ... 
}

rlubos, 2022-06-29:
This is only part of the picture. Yes you need to enable those configs to get certain cabailities. But those capabilites have to be reported by the radio driver in the first place:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/lib/openthread/platform/radio.c#L754
Capability handling in nRF 802.15.4 driver is a bit complex (https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/ieee802154/ieee802154_nrf5.c#L225) but the current state is as I mentioned in the previous post - to get full feature set you need to use driver shipped with NCS.

ck-telecom, 2022-07-03:
@rlubos
Unfortunately ncs can not build either:
```
west build -b nrf52840dongle_nrf52840 nrf/samples/openthread/coprocessor -DOVERYLAY_CONFIG="overylay-usb.conf" -DDTC_OVERLAY_FILE="usb.overlay"
```
```
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/jandy/workspace/ncs/nrf/samples/openthread/coprocessor
-- Found Python3: /usr/bin/python3.10 (found suitable exact version "3.10.4") found components: Interpreter 
-- Cache files will be written to: /home/jandy/.cache/zephyr
-- Zephyr version: 3.0.99 (/home/jandy/workspace/ncs/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: nrf52840dongle_nrf52840
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.14.2 (/home/jandy/zephyr-sdk-0.14.2)
-- Found dtc: /home/jandy/zephyr-sdk-0.14.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.14.2 (/home/jandy/zephyr-sdk-0.14.2)
-- Found BOARD.dts: /home/jandy/workspace/ncs/zephyr/boards/arm/nrf52840dongle_nrf52840/nrf52840dongle_nrf52840.dts
-- Found devicetree overlay: usb.overlay
-- Generated zephyr.dts: /home/jandy/workspace/ncs/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/jandy/workspace/ncs/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/jandy/workspace/ncs/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/jandy/workspace/ncs/build/zephyr/dts.cmake

warning: UART_INTERRUPT_DRIVEN (defined at boards/shields/sparkfun_sara_r4/Kconfig.defconfig:24, boards/shields/wnc_m14a2a/Kconfig.defconfig:17, soc/arm/quicklogic_eos_s3/Kconfig.defconfig:20, drivers/serial/Kconfig:72) has direct dependencies (NETWORKING && SHIELD_SPARKFUN_SARA_R4) || (NETWORKING && SHIELD_WNC_M14A2A) || (SERIAL && SOC_EOS_S3) || (SERIAL_SUPPORT_INTERRUPT && SERIAL) with value n, but is currently being y-selected by the following symbols:
 - OPENTHREAD_COPROCESSOR (defined at subsys/net/l2/openthread/Kconfig:217), with value y, direct dependencies NET_L2_OPENTHREAD && NETWORKING (value: y), and select condition NET_L2_OPENTHREAD && NETWORKING (value: y)
Parsing /home/jandy/workspace/ncs/nrf/samples/openthread/coprocessor/Kconfig
Loaded configuration '/home/jandy/workspace/ncs/zephyr/boards/arm/nrf52840dongle_nrf52840/nrf52840dongle_nrf52840_defconfig'
Merged configuration '/home/jandy/workspace/ncs/nrf/samples/openthread/coprocessor/prj.conf'

error: Aborting due to Kconfig warnings

CMake Error at /home/jandy/workspace/ncs/zephyr/cmake/modules/kconfig.cmake:290 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/jandy/workspace/ncs/zephyr/cmake/modules/zephyr_default.cmake:121 (include)
  /home/jandy/workspace/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:51 (include)
  /home/jandy/workspace/ncs/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:76 (include_boilerplate)
  CMakeLists.txt:11 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/jandy/workspace/ncs/build -S/home/jandy/workspace/ncs/nrf/samples/openthread/coprocessor -GNinja -DBOARD=nrf52840dongle_nrf52840 -DOVERYLAY_CONFIG=overylay-usb.conf -DDTC_OVERLAY_FILE=usb.overlay

```

rlubos, 2022-07-05:
@ck-telecom There is a typo in the command you've posted, it should be
```
west build -b nrf52840dongle_nrf52840 nrf/samples/openthread/coprocessor -DOVERLAY_CONFIG="overlay-usb.conf" -DDTC_OVERLAY_FILE="usb.overlay"
```

ck-telecom, 2022-07-06:
still failed with library depencies
```
jandy@ubuntu-desktop:~/workspace/ncs$ west build -b nrf52840dongle_nrf52840 nrf/samples/openthread/coprocessor -DOVERLAY_CONFIG="overlay-usb.conf" -DDTC_OVERLAY_FILE="usb.overlay"
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/jandy/workspace/ncs/nrf/samples/openthread/coprocessor
-- Cache files will be written to: /home/jandy/.cache/zephyr
-- Zephyr version: 3.0.99 (/home/jandy/workspace/ncs/zephyr)
-- Found west (found suitable version "0.13.1", minimum required is "0.7.1")
-- Board: nrf52840dongle_nrf52840
-- Found host-tools: zephyr 0.14.2 (/home/jandy/zephyr-sdk-0.14.2)
-- Found dtc: /home/jandy/zephyr-sdk-0.14.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.14.2 (/home/jandy/zephyr-sdk-0.14.2)
-- Found BOARD.dts: /home/jandy/workspace/ncs/zephyr/boards/arm/nrf52840dongle_nrf52840/nrf52840dongle_nrf52840.dts
-- Found devicetree overlay: usb.overlay
-- Generated zephyr.dts: /home/jandy/workspace/ncs/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/jandy/workspace/ncs/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/jandy/workspace/ncs/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/jandy/workspace/ncs/build/zephyr/dts.cmake
Parsing /home/jandy/workspace/ncs/nrf/samples/openthread/coprocessor/Kconfig
Loaded configuration '/home/jandy/workspace/ncs/zephyr/boards/arm/nrf52840dongle_nrf52840/nrf52840dongle_nrf52840_defconfig'
Merged configuration '/home/jandy/workspace/ncs/nrf/samples/openthread/coprocessor/prj.conf'
Merged configuration '/home/jandy/workspace/ncs/nrf/samples/openthread/coprocessor/overlay-usb.conf'
Configuration saved to '/home/jandy/workspace/ncs/build/zephyr/.config'
Kconfig header saved to '/home/jandy/workspace/ncs/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/jandy/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- OpenThread Source Directory: /home/jandy/workspace/ncs/modules/lib/openthread
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Searching 16 bit integer - Using unsigned short
-- Check if the system is big endian - little endian
-- Package Name: OPENTHREAD
-- Setting default package version: e2a765599
-- Package Version: e2a765599
CMake Warning at /home/jandy/workspace/ncs/zephyr/CMakeLists.txt:1803 (message):
  __ASSERT() statements are globally ENABLED

-- Configuring done
-- Generating done
-- Build files have been written to: /home/jandy/workspace/ncs/build
-- west build: building application
[1/418] Preparing syscall dependency handling

[2/418] Generating include/generated/version.h
-- Zephyr version: 3.0.99 (/home/jandy/workspace/ncs/zephyr), build: v3.0.99-ncs1-9-g8ffc2ab25eaa
[54/418] cd /home/jandy/workspace/ncs/build/modules/openthread/build && /usr/bin/cmake -DLIST="OPENTHREA...NFIG_FILE="nrf-config-user.h"" -P /home/jandy/workspace/ncs/modules/lib/openthread/etc/cmake/print.cmake
OPENTHREAD_CONFIG_ASSERT_ENABLE=1
OPENTHREAD_CONFIG_BACKBONE_ROUTER_ENABLE=1
OPENTHREAD_CONFIG_BACKBONE_ROUTER_DUA_NDPROXYING_ENABLE=1
OPENTHREAD_CONFIG_BACKBONE_ROUTER_MULTICAST_ROUTING_ENABLE=1
OPENTHREAD_CONFIG_BORDER_AGENT_ENABLE=1
OPENTHREAD_CONFIG_BORDER_ROUTER_ENABLE=1
OPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS=0
OPENTHREAD_CONFIG_ENABLE_BUILTIN_MBEDTLS_MANAGEMENT=0
OPENTHREAD_CONFIG_CHILD_SUPERVISION_ENABLE=1
OPENTHREAD_CONFIG_COAP_API_ENABLE=1
OPENTHREAD_CONFIG_COAP_SECURE_API_ENABLE=1
OPENTHREAD_CONFIG_COMMISSIONER_ENABLE=1
OPENTHREAD_CONFIG_MAC_CSL_RECEIVER_ENABLE=1
[... 43 lines cut ...]
NRF_802154_ACK_TIMEOUT_ENABLED=1
NRF_802154_ENCRYPTION_ENABLED=1
NRF_802154_SECURITY_WRITER_ENABLED=1
NRF_802154_IE_WRITER_ENABLED=1
NRF_802154_INTERNAL_RADIO_IRQ_HANDLING=0
NRF_802154_ECB_PRIORITY=-1
NRF_802154_SWI_PRIORITY=1
MBEDTLS_CONFIG_FILE=nrf-config.h
MBEDTLS_USER_CONFIG_FILE=nrf-config-user.h
[408/418] Linking CXX executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/jandy/workspace/ncs/build/zephyr/zephyr_pre0.map 
: && ccache /home/jandy/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/jandy/workspace/ncs/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/newlib/liblib__libc__newlib.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/boards/boards/arm/nrf52840dongl [... cut ...]
/home/jandy/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: dtoa.c:(.text._dtoa_r+0x598): undefined reference to `__aeabi_d2iz'
/home/jandy/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: dtoa.c:(.text._dtoa_r+0x646): undefined reference to `__aeabi_d2iz'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/jandy/workspace/ncs/build
```

rlubos, 2022-07-06:
Try to use Zephyr SDK 0.14.1, that's the version officially supported by NCS. It seems that something changed in 0.14.2 that made out binaries incompatible with this version.

napent, 2022-10-17:
just add CONFIG_CBPRINTF_FP_SUPPORT=y