zephyrproject-rtos/zephyr #47002: Migration and road map from the deprecated testing API to the new ZTEST API.

aaronemassey, 2022-06-28:
# Migration and road map from deprecated testing API to the current API.

## Each test suite should be refactored to the [new ZTEST API](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/testsuite/ztest/include/ztest_test_new.h).

## Some example API migrations

* [Bluetooth AT](https://github.com/zephyrproject-rtos/zephyr/pull/44761)
* [CMSIS DSP](https://github.com/zephyrproject-rtos/zephyr/pull/44855)
* [Condition Variables](https://github.com/zephyrproject-rtos/zephyr/pull/46662)
* [Posix](https://github.com/zephyrproject-rtos/zephyr/pull/46809)

## Migration status

The list below was initially generated by finding tests without a `ZTEST_SUITE`
macro usage and roughly categorizing them with respect to their directories and
tested features. For example, `tests/application_development` and
`tests/arch/arm` contain unmigrated tests.

- [ ] application_development
- [ ] arch/arm
- [ ] arch/arm64
- [ ] arch/x86
- [ ] benchmarks
- [ ] bluetooth
- [ ] boards/altera_max10
- [ ] boards/espressif_esp32
- [ ] boards/intel_adsp
- [ ] boards/mec15xxevb_assy6853
- [ ] boards/native_posix
- [ ] boot/uefi
- [ ] cmake/config_dir
- [ ] crypto
- [ ] drivers
- [ ] kernel
- [ ] lib
- [ ] misc
- [ ] net
- [ ] net/lib
- [ ] net/socket
- [ ] subsys
- [ ] subsys/fd
- [ ] subsys/logging
- [ ] subsys/mgmt
- [ ] subsys/pm
- [ ] subsys/portability
- [ ] subsys/settings
- [ ] subsys/shell
- [ ] subsys/storage
- [ ] subsys/usb
- [ ] unit
- [ ] ztest

## Simple migration workflow

This does not cover all migration cases nor the entirety of the ZTEST API. This
is just a simple workflow get started on simple tests.

* Replace `ztest_test_suite()` and its containing function with a
   `ZTEST_SUITE()` [defined here with documented
   parameters](https://github.com/zephyrproject-rtos/zephyr/blob/69d153cd3d72fd0f50ab662e83ea3af98bfc0049/subsys/testsuite/ztest/include/ztest_test_new.h#L114).
* Replace prior test functions invoking `zassert` with `ZTEST` or `ZTEST_USER`.
* Add `CONFIG_ZTEST_NEW_API=y` to the test's `prj.conf`.

## Difficulty or questions

Usually, a single test's migration shouldn't require more than 10 minutes. If
you're having difficulty porting a test or understanding how to use the new
ZTEST API, then make a comment on this issue.

nashif, 2022-06-30:
for all of those working on this, do not have a test_ prefix in the testsuite name. This is just confusing and redundant.

hakehuang, 2022-07-01:
@aaronemassey , I have a problem here, there used to be some test fucntions are registerred as shall command see below code
```
extern void test_dma_m2m_chan0_burst8(void);
extern void test_dma_m2m_chan1_burst8(void);
extern void test_dma_m2m_chan0_burst16(void);
extern void test_dma_m2m_chan1_burst16(void);

#ifdef CONFIG_SHELL
TC_CMD_DEFINE(test_dma_m2m_chan0_burst8)
TC_CMD_DEFINE(test_dma_m2m_chan1_burst8)
TC_CMD_DEFINE(test_dma_m2m_chan0_burst16)
TC_CMD_DEFINE(test_dma_m2m_chan1_burst16)

SHELL_CMD_REGISTER(test_dma_m2m_chan0_burst8, NULL, NULL,
			TC_CMD_ITEM(test_dma_m2m_chan0_burst8));
SHELL_CMD_REGISTER(test_dma_m2m_chan1_burst8, NULL, NULL,
			TC_CMD_ITEM(test_dma_m2m_chan1_burst8));
SHELL_CMD_REGISTER(test_dma_m2m_chan0_burst16, NULL, NULL,
			TC_CMD_ITEM(test_dma_m2m_chan0_burst16));
SHELL_CMD_REGISTER(test_dma_m2m_chan1_burst16, NULL, NULL,
			TC_CMD_ITEM(test_dma_m2m_chan1_burst16));
#endif
```

shall we create a marco for ZTEST_SH which looks like below
```
#define Z_TEST_WITH_SH(suite, fn, t_options, use_fixture)                                                  \
	static void _##suite##_##fn##_wrapper(void *data);                                         \
	static void suite##_##fn(                                                                  \
		COND_CODE_1(use_fixture, (struct suite##_fixture *fixture), (void)));              \
	static STRUCT_SECTION_ITERABLE(ztest_unit_test, z_ztest_unit_test_##suite##_##fn) = {      \
		.test_suite_name = STRINGIFY(suite),                                               \
		.name = STRINGIFY(fn),                                                             \
		.test = (_##suite##_##fn##_wrapper),                                               \
		.thread_options = t_options,                                                       \
	};                                                                                         \
	static void _##suite##_##fn##_wrapper(void *data)                                          \
	{                                                                                          \
		COND_CODE_1(use_fixture, (suite##_##fn((struct suite##_fixture *)data);),          \
			    (ARG_UNUSED(data); suite##_##fn();))                                   \
	}                                                                                          \
       TC_CMD_DEFINE(suite##_##fn)                                                           \
       SHELL_CMD_REGISTER(suite##_##fn, NULL, NULL, TC_CMD_ITEM(suite##_##fn));  \
	static inline void suite##_##fn(                                                           \
		COND_CODE_1(use_fixture, (struct suite##_fixture *fixture), (void)))

```

aaronemassey, 2022-07-06:
@hakehuang 

> @aaronemassey , I have a problem here, there used to be some test fucntions are registerred as shall command see below code

Looking at this still, but I'm a bit confused as to actually how the SHELL_COMMAND_REGISTER macro is actually being leveraged in this test. It looks like the test is simply running the test functions themselves without respect to the shell command macro. Maybe we can discuss this at the next working group meeting (tomorrow).

simhein, 2022-07-11:
@aaronemassey subsys/fs is wrong checked in the issue.  See: https://github.com/zephyrproject-rtos/zephyr/tree/main/tests/subsys/fs. I started working on it can you add my name to the "task"?

JordanYates, 2023-07-07:
I feel like this can be closed with the deprecation of the old API in v3.4

closed 2023-07-07 by JordanYates

shuai1x, 2023-07-10:
47002

Closed #47002 as completed.