zephyrproject-rtos/zephyr #47014: tests canbus failed with twister on nucleo_g474re

FRASTM, 2022-06-29:
**Describe the bug**
Running the _**tests/subsys/canbus/isotp/implementation**_  with twister on the nucleo_g474re fails

The same test passed when running after west build and west flash and when flashing with west and running `twister --no-update
`

**To Reproduce**
Steps to reproduce the behavior:
1. twister --device-testing --device-serial /dev/ttyACM0 --disable-suite-name-check -p nucleo_g474re -T tests/subsys/canbus/isotp/implementation/
2. See error

**Expected behavior**
test passed 

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.1.0-913-g6ede43b0789d  ***
Running TESTSUITE isotp
===================================================================
START - test_bind_unbind
PASS - test_bind_unbind in 2.509 seconds
===================================================================
START - test_send_receive_net_sf
PASS - test_send_receive_net_sf in 0.6 seconds
===================================================================
START - test_send_receive_net_blocks
PASS - test_send_receive_net_blocks in 1.20 seconds
===================================================================
START - test_send_receive_net_single_blocks
E: Reception of next FC has timed out
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:289: test_send_receive_net_single_blocks: (ret not equal to 0)
recv returned -14
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:55: send_complette_cb: (error_nr not equal to ISOTP_N_OK)
Sending failed (-2)
FAIL - test_send_receive_net_single_blocks in 1.65 seconds
===================================================================
START - test_send_receive_sf
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:91: get_sf: (ret not equal to 1)
recv returned -14
FAIL - test_send_receive_sf in 1.14 seconds
===================================================================
START - test_send_receive_blocks
E: No buffer for FF left
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:261: test_send_receive_blocks: (ret not equal to 0)
Binding failed (-11)
FAIL - test_send_receive_blocks in 0.18 seconds
===================================================================
START - test_send_receive_single_block
E: No buffer for FF left
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:313: test_send_receive_single_block: (ret not equal to 0)
Binding failed (-11)
FAIL - test_send_receive_single_block in 0.18 seconds
===================================================================
START - test_buffer_allocation
E: No buffer for FF left
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:390: test_buffer_allocation: (ret not equal to 0)
Binding failed (-11)
FAIL - test_buffer_allocation in 0.18 seconds
===================================================================
START - test_buffer_allocation_wait
E: No buffer for FF left
Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:406: test_buffer_allocation_wait: (ret not equal to 0)
Binding failed (-11)
FAIL - test_buffer_allocation_wait in 0.18 seconds
===================================================================
TESTSUITE isotp failed.
===================================================================
RunID: 0ca4e10ffc023fffa65cc26e77dc6785
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain : zephyr-sdk-0.14.0
 - Commit SHA : 6ede43b0789d973d041bab0b61245248dcaf2f59

henrikbrixandersen, 2022-07-13:
Confirmed. @alexanderwachter, @martinjaeger Any of you able to take a look at this? You are more familiar with the internal workings of ISO-TP than I am.

henrikbrixandersen, 2022-07-16:
The same test passes on several other boards, including the `lpcxpresso55s16` which uses the same Bosch M_CAN driver backend:

```
INFO    - 10705/10710 frdm_k64f                 tests/subsys/canbus/isotp/implementation/canbus.isotp.implementation PASSED (device 22.295s)
INFO    - 10706/10710 mimxrt1024_evk            tests/subsys/canbus/isotp/implementation/canbus.isotp.implementation PASSED (device 34.228s)
INFO    - 10707/10710 twr_ke18f                 tests/subsys/canbus/isotp/implementation/canbus.isotp.implementation PASSED (device 22.617s)
INFO    - 10708/10710 lpcxpresso55s16           tests/subsys/canbus/isotp/implementation/canbus.isotp.implementation PASSED (device 19.644s)
INFO    - 10709/10710 stm32f3_disco             tests/subsys/canbus/isotp/implementation/canbus.isotp.implementation PASSED (device 21.155s)
INFO    - 10710/10710 nucleo_g474re             tests/subsys/canbus/isotp/implementation/canbus.isotp.implementation FAILED Failed (device 9.651s)
```

henrikbrixandersen, 2022-08-14:
@FRASTM, @alexanderwachter, @martinjaeger, @karstenkoenig, @str4t0m: I am tagging this issue as "future" as I do not foresee having the bandwidth to solve this before Zephyr v3.2. If any of you would like to attempt to address this for v3.2, please reach out.

martinjaeger, 2022-08-14:
Don't have enough time at the moment to take care of this either, so I agree to postpone.

FRASTM, 2022-08-17:
recent PR #48995 and #48993 makes the tests/subsys/canbus/isotp/implementation PASSED on nucleo_g474re:

 twister --device-testing --device-serial /dev/ttyACM0 --disable-suite-name-check -p nucleo_g474re -T tests/subsys/canbus/isotp/implementation/

```
*** Booting Zephyr OS build zephyr-v3.1.0-3684-gf3017aa787e7  ***
Running TESTSUITE isotp_implementation
===================================================================
START - test_bind_unbind
PASS - test_bind_unbind in 2.509 seconds
===================================================================
START - test_buffer_allocation
PASS - test_buffer_allocation in 1.551 seconds
===================================================================
START - test_buffer_allocation_wait
PASS - test_buffer_allocation_wait in 10.151 seconds
===================================================================
START - test_send_receive_blocks
PASS - test_send_receive_blocks in 0.455 seconds
===================================================================
START - test_send_receive_net_blocks
PASS - test_send_receive_net_blocks in 1.024 seconds
===================================================================
START - test_send_receive_net_sf
PASS - test_send_receive_net_sf in 0.006 seconds
===================================================================
START - test_send_receive_net_single_blocks
PASS - test_send_receive_net_single_blocks in 0.166 seconds
===================================================================
START - test_send_receive_sf
PASS - test_send_receive_sf in 0.006 seconds
===================================================================
START - test_send_receive_single_block
PASS - test_send_receive_single_block in 0.166 seconds
===================================================================
TESTSUITE isotp_implementation succeeded
------ TESTSUITE SUMMARY START ------
SUITE PASS - 100.00% [isotp_implementation]: pass = 9, fail = 0, skip = 0, total = 9 duration = 16.034 seconds
- PASS - [isotp_implementation.test_bind_unbind] duration = 2.509 seconds
- PASS - [isotp_implementation.test_buffer_allocation] duration = 1.551 seconds
- PASS - [isotp_implementation.test_buffer_allocation_wait] duration = 10.151 seconds
- PASS - [isotp_implementation.test_send_receive_blocks] duration = 0.455 seconds
- PASS - [isotp_implementation.test_send_receive_net_blocks] duration = 1.024 seconds
- PASS - [isotp_implementation.test_send_receive_net_sf] duration = 0.006 seconds
- PASS - [isotp_implementation.test_send_receive_net_single_blocks] duration = 0.166 seconds
- PASS - [isotp_implementation.test_send_receive_sf] duration = 0.006 seconds
- PASS - [isotp_implementation.test_send_receive_single_block] duration = 0.166 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 5b4c4cfd7f0ac3e07fce7a5192a2892b
PROJECT EXECUTION SUCCESSFUL
```

FRASTM, 2022-08-17:
testcase PASSED on nucleo_g474re

henrikbrixandersen, 2022-08-18:
I can still reproduce it on b6fadd16001b9680918cf897af2d3a7d52fd73b6.

henrikbrixandersen, 2023-01-26:
I have been digging some more into this issue. The `twr_ke18f` consistently fail the ISO-TP conformance tests, but running the tests with `-DCONFIG_MCUX_LPTMR_TIMER=n -DCONFIG_CORTEX_M_SYSTICK=y` makes the test pass.

This looks like an issue related to the resolution of the system timer. Whether it only effects the tests or also the actual ISO-TP implementation is yet unknown.

FRASTM, 2023-01-27:
True, I see the tests PASSED with  latest main 0934f705ebc2dc9bd6d6e1ce712504ef7269c76d

henrikbrixandersen, 2023-01-31:
Ok. I cannot reproduce failure in the `tests/subsys/canbus/isotp/implementation/` tests, only the `tests/subsys/canbus/isotp/conformance/` tests seem to be affected by the ticks per second setting. 

I have opened #54254 for that instead. Closing this one.

closed 2023-01-31 by henrikbrixandersen