zephyrproject-rtos/zephyr #47018: net: http: reset timeout each time data is received

jeppenodgaard, 2022-06-29:
**Is your enhancement proposal related to a problem? Please describe.**
`http_client_req` times out if the whole response is not received before the specified `timeout`.

**Describe the solution you'd like**
`timeout` kicks in when there has been no activity for the `timeout` period.

**Describe alternatives you've considered**
Creating another timeout which is reset in the callback function, but this seems sub optimal.

**Additional context**
N/A

jeppenodgaard, 2022-06-29:
I've considered two options:
1. Use `zsock_poll` in `http_wait_data` and remove `k_work..`.
2. Call `k_work_reschedule` in `http_wait_data`

mniestroj, 2022-07-01:
I actually vote for the alternative, so having two timeout options:
- timeout for the whole transfer (what we got now),
- timeout for inactivity (what you need and what I have found useful already and implemented that on the application layer when needed).