zephyrproject-rtos/zephyr #47025: mimxrt1050_evk: reset cause

jeppenodgaard, 2022-06-29:
**Is your feature request related to a problem? Please describe.**
Get reset cause for debugging purpose.

**Describe the solution you'd like**
Possible to read the reset cause.

**Describe alternatives you've considered**
N/A

**Additional context**
Adding `HAS_MCUX_SRC` to `zephyr/soc/arm/nxp_imx/rt/Kconfig.soc` `config SOC_MIMXRT1052` builds, but I always get the same output:
```
uart:~$ hwinfo reset_cause 
reset caused by:
- pin
```
I tried reset with j-link, power cycle, CPU exception.

pdgendt, 2022-07-07:
Have you cleared the reset cause before a new test?
```
hwinfo reset_cause clear
```

jeppenodgaard, 2022-07-07:
> Have you cleared the reset cause before a new test?
> 
> ```
> hwinfo reset_cause clear
> ```

I do not think I did. I tried it now and it seems to be the same behavior.

I expected `exception` and `power` to be a reset cause but `hwinfo reset_cause supported` reports:
```
supported reset causes:
- pin
- watchdog
- debug
- security
- unknown
- unknown
```

It might work after all. I just did not manage to hit the non-pin causes.

closed 2022-07-12 by carlescufi