zephyrproject-rtos/zephyr #47046: Packet socket sample fails on STM32F7 (Nucleo F746ZG)

jaybuckeye06, 2022-06-29:
**Describe the bug**
Packet socket sample fails on STM32 Nucleo-F746ZG.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 Nucleo-746ZG.
 - What have you tried to diagnose or workaround this issue?
 - n/a

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b nucleo_f746zg samples/net/socket/packet
2. west flash
3. observe the uart
4. See error

**Expected behavior**
Sample should run on the board.

**Impact**

**Logs and console output**
[image]

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK
 - zephyr-v3.1.0-1-g44358e62ecd6

**Additional context**
Default dts and prj.conf are used.

erwango, 2022-06-30:
@jaybuckeye06 Thanks for raising this, I can also reproduce this on frdm_k64f, this is likely related to the upper stack.
```
uart:~$ *** Booting Zephyr OS build zephyr-v3.1.0-922-g69e71cedf8a2  ***

[00:00:00.006,000] <inf> net_pkt_sock_sample: Packet socket sample is running
[00:00:00.006,000] <inf> net_pkt_sock_sample: Waiting for packets ...
[00:00:00.006,000] <err> net_pkt_sock_sample: Failed to send, errno 115
[00:00:00.006,000] <inf> net_pkt_sock_sample: Stopping...
[00:00:03.001,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:39.353,000] <err> os: ***** BUS FAULT *****
[00:00:39.353,000] <err> os:   Imprecise data bus error
[00:00:39.353,000] <err> os: r0/a1:  0x00000001  r1/a2:  0x0000000a  r2/a3:  0x200012d8
[00:00:39.353,000] <err> os: r3/a4:  0x00000000 r12/ip:  0x0000e149 r14/lr:  0x0000e163
[00:00:39.353,000] <err> os:  xpsr:  0x41000000
[00:00:39.353,000] <err> os: Faulting instruction address (r15/pc): 0x0000e16e
[00:00:39.353,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:39.353,000] <err> os: Current thread: 0x20002838 (rx_q[0])
[00:00:39.382,000] <err> os: Halting system
```

@rlubos, can you have a check ?

rlubos, 2022-07-11:
I've submitted a PR fix fixes, please have a look: https://github.com/zephyrproject-rtos/zephyr/pull/47636

PR #47636 (2022-08-03): net: sockets: packet: Fix socket packet sample crash
Fix the core issue within the net_context that caused the crash (network connection was not deregistered properly).

Additionally, fix the sample initialization and teardown procedure:
* Make sure the network interface is up before running the sample, otherwise the initial send could fail, causing the sample to stop.
* Fix socket closure on exit.

Fixes #47046
closed 2022-08-03 by carlescufi