zephyrproject-rtos/zephyr #47049: [backport v2.7-branch] backport of #46699 failed

zephyrbot, 2022-06-29:
This issue was created automatically because the backport of #46699 to `v2.7-branch` failed.

The backport to `v2.7-branch` failed:
```
The process '/usr/bin/git' failed with exit code 1
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v2.7-branch v2.7-branch
# Navigate to the new working tree
cd .worktrees/backport-v2.7-branch
# Create a new branch
git switch --create backport-46699-to-v2.7-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick -x 4e20669aa97c9d8aac1e7492c4dfe307856e544f
# Push it to GitHub
git push --set-upstream origin backport-46699-to-v2.7-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v2.7-branch
```
Then, create a pull request where the `base` branch is `v2.7-branch` and the `compare`/`head` branch is `backport-46699-to-v2.7-branch`.

cfriedt, 2022-08-02:
@nordicjm - was this resolved via a manual backport?

nordicjm, 2022-08-02:
@cfriedt Nope, however, created a manual backport now at https://github.com/zephyrproject-rtos/zephyr/pull/48573

closed 2022-08-16 by nordicjm