zephyrproject-rtos/zephyr #47061: pipes: Usage between task and ISR results in corrupted pipe state

cmorganBE, 2022-06-29:
Using a pipe to marshal data between a uart isr callback and a task we ran into a case where the pipe appears to be corrupted after a call to k_pipe_get() in the ISR.

```
uart_isr_handler() {
...
   k_pipe_get(&pipe, 1)
   // at this point the pipe bytes_used == 0 but read_index > write_index, expected that read_index == write_index when bytes_used == 0
...
```

Result is that we have some corrupted bytes in the pipe that are sent via the uart.

Using zephyr 140837ce3ac6ba6ac01d804ebd40c8aaf00a3011 (we plan to upgrade to v3.1.0 shortly), there have been no kernel/pipes.c changes of consequence since this ~v3.0 git version.

Running on stm32h743zi

See pull request #47050 for additional context.

**To Reproduce**
We don't have an easily reproducible test case yet. We reproduce by running some code that puts data into a pipe for the isr to transmit. Adding a task/isr test to the ztests for pipes would be one idea that came to mind.

**Expected behavior**

The documentation for pipes indicates they are not allowed to be used from ISRs, however @andyross believes they should work in that context. I've been kicking myself as I've used similar approaches to pass data in other oses, like embOS, and didn't spot the "no isr" note until we've been using pipes for a few months. 

While the note says no ISR, and the pipes don't appear to work correctly in ISR/task context, it would be great if they did. We are going to use MessageQueue as a work around until more is known here or pipes can work in isr.

**Impact**
We had to swap Pipes -> MessageQueue
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
None captured yet but we can attach some if it would be helpful.

andyross, 2022-06-29:
Had a chance to take a quick look.  Indeed, k_pipe is using k_sched_lock()!  I had no idea.  It has a spinlock around most of the operations, but the actual copies being done in the put() and get() internal utilities are done outside of it, it will actually release the lock and take the scheduler lock instead.

That's actually quite bad, as it means that not only is it not safe vs. interrupts, it's also not usable at all in SMP contexts.

It's almost certainly possible to fix this by replacing the use of  sched_lock with the existing spinlock.  But my guess is that the reason this was done was a concern about interrupt latency (i.e. not wanting to hold the lock for very large block copies).  So there might need to be a fallback layer that does a quick checkpoint of state than then unlocks/locks to permit interrupts to fire.  It might be as simple as putting a little for loop in at the top of k_pipe_put/get() that does the requested operation in 64 byte chunks or whatever.

But then, the whole pipes.c file seems just wildly complicated for what it does (834 lines for what amounts to a dressed up ringbuffer?), so maybe a broader rewrite is in order.

Maybe @peter-mitsis wants to take a shot at this?

carlescufi, 2022-06-30:
I've marked this as a high-prio bug given that it actually describes a real issue in the pipe functionality inside the kernel itself.

peter-mitsis, 2022-07-05:
Just saw this.  I'll take a closer look.  When pipes were first written, it was assumed that Zephyr was only going to be for single core systems. With that assumption in mind, the schedule lock was used for exactly the reason that @andryross identified--reduce interrupt latency on large copies. Since that single core assumption no longer holds, it is time to revisit the implementation details.

PR #48024 (2022-08-17): Extend pipes to support use from ISR
This set of commits updates the pipes implementation and its documentation.

It has a few notable differences as compared to the previous implementation.

1. The use of schedule locking has been eliminated in favour of spinlocks. This both makes it safer for SMP and allows pipes to be used from the context of an ISR. This comes at the price of increasing interrupt latency while copying large amounts of data to/from the pipe.
2. Instead of using custom copy loops depending upon the location of source and destination buffers, I've tried to unify things more s [... cut]
commit f86027ff (2022-08-17): kernel: pipes: rewrite pipes implementation

This new implementation of pipes has a number of advantages over the
previous.
  1. The schedule locking is eliminated both making it safer for SMP
     and allowing for pipes to be used from ISR context.
  2. The code used to be structured to have separate code for copying
     to/from a wating thread's buffer and the pipe buffer. This had
     unnecessary duplication that has been replaced with a simpler
     scatter-gather copy model.
  3. The manner in which the "working list" is generated has also been
     simplified. It no longer tries to use [... cut]
closed 2022-08-17 by carlescufi