zephyrproject-rtos/zephyr #47081: on x86, k_is_in_isr() returns false in execption context

inteljiangwe1, 2022-06-30:
For k_is_in_isr() description, it checks whether the code is running at interrupt level. And it is used by caller to customize the behavior, like avoiding take semaphare in isr handler.
But on execption context, it returns false in execption context on x86

inteljiangwe1, 2022-07-22:
@peter-mitsis any comments?

peter-mitsis, 2022-07-22:
@inteljiangwe1 - Apologies for the late reply. My attention had been focussed on other issues tagged as higher priority. However, I now have a small window to take a look at this before I jump onto the next priority item.

peter-mitsis, 2022-07-22:
To fix or not to fix, that is the question.

I've spent the last couple of hours getting somewhat re-acquainted with this code as a fair bit has happened since I last looked at this code (over 5 years ago).  If I am following things correctly, this appears to be by design, and this will probably fall under "do not fix". (To be clear, this is not a final verdict; it is just an early indicator.)  I will still need to dig a little further, consult and confirm that I have the right mental model before I can say for sure. Either way, I expect to add a comment that will explain why.

peter-mitsis, 2022-07-25:
So I did some more digging. Exceptions, unlike interrupts, are handled at thread level. They may do things that interrupts are not allowed to do such as block on a resource.

I am certain now that this issue qualifies as a "Do Not Fix".

closed 2022-07-25 by peter-mitsis

peter-mitsis, 2022-07-25:
Although there are similarities between exceptions and interrupts, on x86 they are sufficiently different such that k_is_in_isr() returns false inside an exception handler (unless the exception occurred during the handling of an interrupt).

inteljiangwe1, 2022-07-26:
thanks Mitsis for explaination
@kwd-doodling for awareness