zephyrproject-rtos/zephyr #47082: drivers: modem: AT commands sent before OK from previous is received

jeppenodgaard, 2022-06-30:
**Describe the bug**
The gsm_ppp.c driver send the first `setup_modem_info_cmds` before the `OK` is received from the last `setup_cmds`.
This is due to no delay after the last `setup_cmds`:
https://github.com/zephyrproject-rtos/zephyr/blob/265394e30efd3f31fdbece6e49fae73a60eec8d9/drivers/modem/modem_cmd_handler.c#L572

**Expected behavior**
Delay between all AT commands.

jeppenodgaard, 2022-07-01:
I had NO CARRIER issues on a BG95 modem and found this inconsistency, and was not sure what the impact was, therefore I opened the issue.
With a logic analyzer there was a small overlap where the next AT command was sent before both `\r` and `\n` was received.

However I wondered where the 50 ms came from in the first place, but was not able to find anything in 3GPP standards.

mniestroj, 2022-07-01:
The only reason of a delay is when modem needs some time after executed command in order to prepare for the next one. I've seen such requirements stated in modem datasheets some time ago, though I don't remember which ones they were. I guess that this 50ms delay is just an arbitrary number. Maybe it was a requirement of the first modem that was tested with that framework, who knows.

I think that we still need such delay mechanism for modems that require that. Such delay should be configurable in Device Tree, so that by default there is no delay (because there is no reason to have one in general, we are just operating on AT commands).

Delay should not only be part of `modem_cmd_handler_setup_cmds()`, but it should rather be enforced between every AT command, so that there is the same delay between commands from subsequent `modem_cmd_handler_setup_cmds()` calls (which is what you try to achieve, right?). Question is, what to do with commands that operate on sockets? Are they affected as well?

> With a logic analyzer there was a small overlap where the next AT command was sent before both \r and \n was received.

This is related, but a bit different issue from delaying between sent commands. It might be worth to wait for `\r\n` characters before assuming that AT command execution was finished.

jeppenodgaard, 2022-07-01:
Very good points.

Can you elaborate on the socket issue? Do you think there is a need to add delays there as well?

>Delay should not only be part of modem_cmd_handler_setup_cmds(), but it should rather be enforced between every AT command, so that there is the same delay between commands from subsequent modem_cmd_handler_setup_cmds() calls (which is what you try to achieve, right?)

Yes.

These are the issues I see:

1. Fix bug where delay is not the same between the AT commands (which is what I attempt to fix [here](https://github.com/zephyrproject-rtos/zephyr/pull/47088)
2. Possible to set delay between AT commands via DTS.
3. Maybe an issue on commands on sockets.
4. Await "\r\n" before issuing another AT command.

mniestroj, 2022-07-01:
>Can you elaborate on the socket issue? Do you think there is a need to add delays there as well?

Sure. I though about alternative approach to have delays between issued AT commands. The idea was to add a delay in `modem_cmd_send*()` family of functions, so that delay is enforced between **each**, no matter from where it is issued (right now delays are only part of `modem_cmd_handler_setup_cmds*()`. The biggest issue with that alternative approach is that there would be delay between each packet transmission on socket, so network performance would suffer. Maybe we will need that at some point, maybe not. It would be better to have a hardware that requires such behavior before implementing it.

So in summary, I have commented about socket operations because of two reasons:
- maybe we need to add delay between socket operations as well (it is likely that modem firmware requires delay between each AT command, no matter if that is "setup phase" or "networking/socket communication"),
- I have concerns about networking performance if we decide to add delays between each AT command (not only between AT commands in "setup phase").

closed 2022-07-02 by carlescufi