zephyrproject-rtos/zephyr #47105: drivers: clock_control: stm32 common: wrong PLLCLK rate returned

str4t0m, 2022-06-30:
**Describe the bug**
The clock_control_get_subsys_rate() calls returns the wrong frequency value for the PLLCLK clock source on the affected STM32 targets. This is a result of using the wrong PLL source frequency in the calculation.

**Expected behavior**
As long as the PLLCLK is enabled, the correct frequency value should be returned otherwise 0.

**Impact**
The PLLCLK frequency can't be queried on most SOCs of the stm32_ll_common driver.
Affected: STM32F2, STM32F4, STM32F7, STM32G0, STM32G4, STM32L0, STM32L1, STM32L4, STM32L5, STM32WB, STM32WL.

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - Zephyr main (a1ef368094)

str4t0m, 2022-06-30:
I will soon open a PR resolving this as a followup to #47102.

FRASTM, 2022-07-22:
@str4t0m do you have a PR for this ?

str4t0m, 2022-07-22:
@FRASTM Yes I had prepared something, I think it should still be applicable without change.

Update:
- STM32G0, STM32G4, STM32L4, STM32L5, STM32WB, and STM32WL.are already fixed, and PLLCLK was dropped after the PLL_X sources have been introduced.
-  STM32F2, STM32F4, STM32F7: have the same error, but it is not exposed as PLLCLK was not added as clock source, but it will be needed once PLL sources are added for these SoCs.
- STM32L0, STM32L1: were still affected.

Will be closed in #48171

PR #48171 (2022-07-27): drivers/clock_control: stm32 L0,L1 fix STM32_SRC_PLLCLK calculation
Some Series were calculating the pll output frequency from an
clock source index instead of the clock source frequency.

This commit resolves this issue for l0, l1 and removes an unused function for f2, f4, f7.

fixes #47105
closed 2022-07-27 by carlescufi