zephyrproject-rtos/zephyr #47120: shell uart: busy wait for DTR in ISR

aurel32, 2022-06-30:
**Describe the bug**
With `CONFIG_SHELL_BACKEND_SERIAL_CHECK_DTR=y`, the shell uart backend waits for DTR before sending data to the UART. However it does so in the ISR via `uart_callback()` -> `uart_tx_handle()` -> `uart_dtr_wait()`

The consequences are very different depending on how the UART driver is implemented. In my case I use this USB CDC ACM UART, so it only blocks the whole USB workqueue, and thus the whole USB stack.

**To Reproduce**
- Take a USB sample
- Set `CONFIG_USB_COMPOSITE_DEVICE=y`
- Enable the shell on the USB CDC ACM UART
- The other USB device interfaces are unresponsive until a connection is make to the CDC ACM interface.

**Expected behavior**
Using a shell on USB CDC ACM UART should not impact the other interfaces of a composite USB device.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.14.2
 - Commit SHA: 74afc2df0667532315d01430dd9d4631e0c350fa

mmahadevan108, 2022-09-09:
@aurel32 does the suggestion from @jfischer-no work for you? If so kindly close this issue.

jakub-uC, 2023-01-11:
@aurel32 please let me know if this is still an issue or if we can close it.

jfischer-no, 2023-01-19:
> @aurel32 please let me know if this is still an issue or if we can close it.

It is still an issue. Maybe we can not init/enable uart backend per default if CDC ACM UART is used and enable/uninit it on DTR state change?

PR #56510 (2023-04-06): shell: uart: Fix wait for DTR before TX
Fail compilation when incorrect configuration is detected, i.e. when SHELL_BACKEND_SERIAL_CHECK_DTR is set but UART_LINE_CTRL is not set.

Use periodic timer to wait for DTR instead of waiting in uart callback to prevent blocking caller workqueue and/or sleeping in ISR.

DTR check was only ever supported with interrupt driven backend so add appropriate depends on to Kconfig.

Fixes: e9f238889b1a ("shell: uart: Add waiting on DTR signal before sending data")
Fixes: #47120
Fixes: #54705
commit 59f33a60 (2023-04-06): shell: uart: Fix wait for DTR before TX

Fail compilation when incorrect configuration is detected, i.e. when
SHELL_BACKEND_SERIAL_CHECK_DTR is set but UART_LINE_CTRL is not set.

Use periodic timer to wait for DTR instead of waiting in uart callback
to prevent blocking caller workqueue and/or sleeping in ISR.

DTR check was only ever supported with interrupt driven backend so add
appropriate depends on to Kconfig.

Fixes: e9f238889b1a ("shell: uart: Add waiting on DTR signal before
sending data")
Fixes: #47120
Fixes: #54705
closed 2023-04-06 by carlescufi