zephyrproject-rtos/zephyr #47127: twister : frdm_k64f ：Non-existent tests appear and fail on tests/lib/cmsis_dsp/transform

shaoanxli, 2022-07-01:
**Describe the bug**
Some Non-existent tests seem to be created and every time seems to be the different cases. The issue seems hard to reproduce. 
In twister.json the result shows below:
[image]

The following are some of the cases that do not exist:
```
libraries.cmsis_dsp.transform.cq31.arm_rfft_f32_real_step_64
libraries.cmsis_dsp.transform.cq31.arm_rfft_f16_real_noisy_4096
libraries.cmsis_dsp.transform.cq31.arm_rifft_f32_real_step_128
libraries.cmsis_dsp.transform.cq31.arm_cfft_f32_cmplx_step_2048
libraries.cmsis_dsp.transform.cq31.arm_rifft_q31_step_2048
libraries.cmsis_dsp.transform.cq31.arm_rifft_f64_real_step_2048
```

**Impact**
The Non-existent cases show test failed and lead to a drop in the pass rate.

**Environment :**
OS: Linux,
Toolchain Zephyr SDK 0.14.1
Commit SHA or Version: dba3d31dddac

nashif, 2022-07-07:
I am not sure what you mean., this test is defined and runs if the platform supports running it, if not, then it right now captured as not tested.

shaoanxli, 2022-07-07:
> I am not sure what you mean., this test is defined and runs if the platform supports running it, if not, then it right now captured as not tested.

But in this situation , the test is marked as blocked , and it does not actually run. This should not happen.

shaoanxli, 2022-07-08:
This is not happening now, so it should be closed.

shaoanxli, 2022-07-12:
The failure arise again.

Take **libraries.cmsis_dsp.matrix.binary_f16.fpu.op1_arm_mat_scale_q15** as an example.

The case is on this path **tests/lib/cmsis_dsp/matrix/src/unary_q15.c**  , but the error shows :
```
            "name":"tests/lib/cmsis_dsp/matrix/libraries.cmsis_dsp.matrix.binary_f16.fpu",
            "arch":"arm",
            "platform":"frdm_k64f",
            "run_id":"6a4b696a8efa48acb1cc5edc923ce2b3",
            "runnable":true,
            "status":"passed",
            "execution_time":"11.72",
            "testcases":[
                {
                    "identifier":"libraries.cmsis_dsp.matrix.binary_f16.fpu.op1_arm_mat_scale_q15",
                    "execution_time":"0.00",
                    "status":"blocked",
                    "reason":"No Console Output(Timeout)"
                },

```
Then if we check testsuite **matrix_binary_f16** :
```
void test_matrix_binary_f16(void)
{
        ztest_test_suite(matrix_binary_f16,
                ztest_unit_test(test_op2_arm_mat_mult_f16),
                ztest_unit_test(test_op2c_arm_mat_cmplx_mult_f16)
                );

        ztest_run_test_suite(matrix_binary_f16);
}

```
We can not find the test case **op1_arm_mat_scale_q15** here.
So maybe twister mismatch the testsuite to the test case.

mengxianglinx, 2022-07-12:
That means twister wrongly concaterate the test suite name and test case name. In test suite matrix_binary_f16 there is no such a test case as test_op1_arm_mat_scale_q15, that case is belong to test suite matrix_unary_q15.

mengxianglinx, 2022-07-12:
There is no mismatch. In libraries.cmsis_dsp.matrix.binary_f16.fpu.op1_arm_mat_scale_q15, "libraries.cmsis_dsp.matrix.binary_f16.fpu" is defined in testcases.yml, and "op1_arm_mat_scale_q15" in defined in c file with prefix "test_"

nashif, 2022-07-27:
not a bug in twister, the dsp tests themselves are a mess and this can only be resolved once we start getting testcase lists from binaries.

closed 2022-07-27 by nashif