zephyrproject-rtos/zephyr #47197: BLE latency decreasing and increasing over time (possibly GPIO issue)

CodingGhost, 2022-07-02:
I am trying to measure I/O latency over BLE by toggling an LED, doing a write without response and toggle an led on the receiving side once it got the data. Then I measure the LED pulses with an oscilloscope to get the latency.
I would expect the latency to be between 7.5ms and 15ms for a connection interval of 7.5ms
but I measure something different: the latency starts at around 60ms, then decreases over time. It eventually hits about 4ms (which should be impossible) and then jumps back to 60ms.
I recorded the oscilloscope to give a better impression of the problem:
https://youtu.be/v8DMdQ2Ga4o

I have made a small example to reproduce the Issue:
[Uploading zephyrtest.zip…]()
configure PIN_LED to 13 on both programs if you want to use 2 DKs, then just load the 2 programs on 2 devices and watch the LEDs, it is noticable even without an oscilloscope if you watch closely.

Im using Zephyr 2.7.1 (which should be LTS thus I guess its still important)

Thalley, 2022-07-04:
@CodingGhost do you by any chance have the wireshark sniff you can share?

If the hosts think that the connectional interval is properly updated, then there's something going on in the controller.

CodingGhost, 2022-07-04:
[driftsniff.txt](https://github.com/zephyrproject-rtos/zephyr/files/9039643/driftsniff.txt)
you will have to rename the file ending to .pcapng because github doesnt allow that extension.
@cvinayak I wouldnt call it solved, Its more of a workaround. my Program already does a connection parameter update and you can see it in the sniff. they agree on 7.5ms connection interval but still I get latency up to 60ms. if I add the kconfig parameters the latency is where it should be. the sniff I sent was without the kconfig parameter, I will try to get one with it this evening

CodingGhost, 2022-07-04:
damn. I missed that second parameter update in the sniff. thanks for clearing that up, now everything makes sense!