zephyrproject-rtos/zephyr #47238: SD Card init issue when CONFIG_SPEED_OPTIMIZATIONS=y

asarbs, 2022-07-04:
**Describe the bug**

When I'm turning on `CONFIG_SPEED_OPTIMIZATIONS=y` I'm getting  `sd:Card never left busy state` error.  

**To Reproduce**
Steps to reproduce the behavior:
prj.conf
```
CONFIG_STDOUT_CONSOLE=y

CONFIG_CPLUSPLUS=y
CONFIG_STD_CPP20=y
CONFIG_NEWLIB_LIBC=y
CONFIG_LIB_CPLUSPLUS=y

CONFIG_MAIN_STACK_SIZE=90000

CONFIG_DEBUG=y
CONFIG_PRINTK=y
CONFIG_LOG=y
CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y

CONFIG_DISK_ACCESS=y
CONFIG_FILE_SYSTEM=y
CONFIG_FAT_FILESYSTEM_ELM=y
CONFIG_DISK_DRIVER_SDMMC=y
CONFIG_SPI=y

CONFIG_LOG_MODE_IMMEDIATE=y
CONFIG_PWM_LOG_LEVEL_DBG=y

CONFIG_FS_LITTLEFS_BLK_DEV=y
CONFIG_APP_LITTLEFS_STORAGE_BLK_SDMMC=y
CONFIG_FILE_SYSTEM_LITTLEFS=y

CONFIG_TIMING_FUNCTIONS=y

CONFIG_SPEED_OPTIMIZATIONS=y
CONFIG_FS_FATFS_NUM_FILES=10
```

```
            if (disk_access_init(disk_pdrv) != 0) {
                LOG_ERR("Storage init ERROR!");
                break;
            }
```

1. west flash 

I don't have this problem when i setup comment out `CONFIG_SPEED_OPTIMIZATIONS=y` from my config file. 

**Expected behavior**
I'm able to read from SD card files with `CONFIG_SPEED_OPTIMIZATIONS=y`

**Impact**
I can't use `CONFIG_SPEED_OPTIMIZATIONS=y`

**Logs and console output**
[image]

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 3.1.99

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

asarbs, 2022-07-05:
```
static int sdmmc_send_ocr(struct sd_card *card, int ocr_arg)
{
	struct sdhc_command cmd;
	int ret;
	int retries;

	cmd.opcode = SD_APP_SEND_OP_COND;
	cmd.arg = ocr_arg;
	cmd.response_type = (SD_RSP_TYPE_R3 | SD_SPI_RSP_TYPE_R1);
	cmd.timeout_ms = CONFIG_SD_CMD_TIMEOUT;
	/* Send initialization ACMD41 */
	for (retries = 0; retries < CONFIG_SD_OCR_RETRY_COUNT; retries++) {
		**LOG_INF("retries=%d", retries);**
		ret = sdmmc_app_command(card, 0U);
		if (ret == SD_RETRY) {
			/* Retry */
			continue;
		} else if (ret) {
			return ret;
		}
		ret = sdhc_request(card->sdhc, &cmd, NULL);
		if (ret) {
			/* OCR failed */
			return ret;
		}
		if (ocr_arg == 0) {
			/* Just probing, don't wait for card to exit busy state */
			return 0;
		}
		/*
		 * Check to see if card is busy with power up. PWR_BUSY
		 * flag will be cleared when card finishes power up sequence
		 */
		if (card->host_props.is_spi) {
			if (!(cmd.response[0] & SD_SPI_R1IDLE_STATE)) {
				break;
			}
		} else {
			if ((cmd.response[0U] & SD_OCR_PWR_BUSY_FLAG)) {
				break;
			}
		}
		sd_delay(10);
	}
	if (retries >= CONFIG_SD_OCR_RETRY_COUNT) {
		/* OCR timed out */
		LOG_ERR("Card never left busy state");
		return -ETIMEDOUT;
	}
	LOG_DBG("SDMMC responded to ACMD41 after %d attempts", retries);
	if (!card->host_props.is_spi) {
		/* Save OCR */
		card->ocr = cmd.response[0U];
	}
	return 0;
}
```

After add of this LOG_INF line problem stop to occur.

carlescufi, 2022-07-05:
@asarbs what board are you building for?

asarbs, 2022-07-06:
> 

I'm using mimxrt1020_evk

asarbs, 2022-07-13:
I have do some more investigation and  i have information that if i set CONFIG_SPEED_OPTIMIZATIONS=y, mount point flags suggest that this is READ ONLY File system.

danieldegrasse, 2022-07-13:
@asarbs  Interesting that this fixes the issue- based on your initial description I expected the issue was that CMD55 (app command) was being sent to the card too soon after CMD8. If you replace the `LOG_INF` statement with `k_busy_wait(100)` is the issue also resolved?

asarbs, 2022-07-14:
@danieldegrasse sorry last comment was my mistake, if we are talking about initial issue I still don't have solution.

danieldegrasse, 2022-08-01:
@asarbs, sorry, missed this notification and was on PTO this past week. Will changing the `LOG_INF` statement out for `k_busy_wait` work to fix the issue?

asarbs, 2022-08-03:
@danieldegrasse yes.

PR #48790 (2022-08-18): sd: sdmmc: wait before sending second OCR
introduce a small wait between probing and initialization OCR. This
delay fixes a failure to initialize seen on some SD cards

Fixes #47238
commit ba9220c5 (2022-08-18): sd: sdmmc: wait before sending second OCR

introduce a small wait between probing and initialization OCR. This
delay fixes a failure to initialize seen on some SD cards

Fixes #47238
closed 2022-08-18 by carlescufi