zephyrproject-rtos/zephyr #47240: net: tcp: Correctly handle overlapped TCP retransmits

ssharks, 2022-07-04:
**Describe the bug**
When a fast re-transmit overlaps with data that is already received, the stack does not accept the new incoming data to create one contiguous data stream and acknowledge out of order packets as well.

To reproduce the bug, take the updated test from:
https://github.com/zephyrproject-rtos/zephyr/pull/47115/commits/5885b790dc6f71b0bb89a15b0edf443607f786e9

Take the test example:
static struct out_of_order_check_struct out_of_order_check_list[] = {
	{ 30, 10, 0, 0}, /* First packet will be out-of-order */
	{ 20, 10, 0, 0},
	{ 10, 10, 0, 0},
	{ 0,  10, 40, 0}, /* First sequence complete */
	{ 60, 10, 40, 0},
	{ 70, 10, 40, 0},
	{ 40, 20, 80, 0}, /* Some bigger data */
};

And replace the 20 in the last line to 25:

static struct out_of_order_check_struct out_of_order_check_list[] = {
	{ 30, 10, 0, 0}, /* First packet will be out-of-order */
	{ 20, 10, 0, 0},
	{ 10, 10, 0, 0},
	{ 0,  10, 40, 0}, /* First sequence complete */
	{ 60, 10, 40, 0},
	{ 70, 10, 40, 0},
	{ 40, **25**, 80, 0}, /* Some bigger data */
};

Next run the test using ./scripts/twister -T tests/net/tcp -p qemu_x86

With in that situation, the last 15 bytes are not acknowledged although they are properly received.

**Expected behavior**
The duplicate data should simply discard and constructed into a contiguous stream.

**Impact**
Although PR https://github.com/zephyrproject-rtos/zephyr/pull/47115 implement fast re-transmit and reception for out of order reception to work, in practice it's benefit might be limited.

**Environment (please complete the following information):**
 - OS: Linux

ssharks, 2022-07-04:
@jukkar: FYI

rlubos, 2022-09-02:
Should be fixed by https://github.com/zephyrproject-rtos/zephyr/pull/47115

closed 2022-09-02 by rlubos