zephyrproject-rtos/zephyr #47275: Twister builds are broken with gnuarmemb toolchain, due to picolibc tests/configuration

pfalcon, 2022-07-04:
**Describe the bug**
Building Picolibc-enabled configurations with a toolchain other than Zephyr SDK (reported against GnuArmEmb toolchain) leads to Kconfig warnings due to conflicting config symbol definitions, which in case of running with Twister leads to an error.

**To Reproduce**
1. Download https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-rm/9-2019q4/gcc-arm-none-eabi-9-2019-q4-major-x86_64-linux.tar.bz2 and uncompress it, e.g. to ~/opt
2. Set in the environment:
```
export ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
export GNUARMEMB_TOOLCHAIN_PATH=~/opt/gcc-arm-none-eabi-9-2019-q4-major
```
3. Run `./scripts/twister --no-clean -s tests/kernel/common/kernel.common.picolibc -p frdm_k64f -v`
4. Leads to error:
```
warning: THREAD_LOCAL_STORAGE (defined at kernel/Kconfig:930) has direct dependencies ARCH_HAS_THREAD_LOCAL_STORAGE && TOOLCHAIN_SUPPORTS_THREAD_LOCAL_STORAGE with value n, but is currently being y-selected by the following symbols:
 - PICOLIBC (defined at lib/libc/Kconfig:39), with value y, direct dependencies !NATIVE_APPLICATION && <choice LIBC_IMPLEMENTATION> (value: y), and select condition ARCH_HAS_THREAD_LOCAL_STORAGE && !NATIVE_APPLICATION && <choice LIBC_IMPLEMENTATION> (value: y)

error: Aborting due to Kconfig warnings

CMake Error at /home/pfalcon/projects-work/Zephyr/zephyr/cmake/modules/kconfig.cmake:290 (message):
```

**Expected behavior**

Building the testsuite with Twiswter using toolchains other than Zephyr SDK should not fail because Picolibc was integrated.

**Impact**

This would, and does, fail any CI which verifies that Zephyr testsuite is buildable with other toolchains but Zephyr SDK. E.g., it fails Linaro CI which verifies buildability with gnuarmemb: https://ci.linaro.org/job/zephyr-upstream/6891/

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gnuarmemb 9-2019-q4-major , https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-rm/9-2019q4/gcc-arm-none-eabi-9-2019-q4-major-x86_64-linux.tar.bz2
 - Commit SHA or Version used: d2ac7b4835ea3f4a32bed74db191a51dfafcf4d2 (latest main as of writing)

pfalcon, 2022-07-04:
@keith-packard: I hope that assigning this to you makes sense.

pfalcon, 2022-07-04:
Analysis:

lib/libc/Kconfig has:
~~~
config PICOLIBC
        bool "Picolibc library"
        select THREAD_LOCAL_STORAGE if ARCH_HAS_THREAD_LOCAL_STORAGE
~~~

kernel/Kconfig has:
~~~
config THREAD_LOCAL_STORAGE
        bool "Thread Local Storage (TLS)"
        depends on ARCH_HAS_THREAD_LOCAL_STORAGE && TOOLCHAIN_SUPPORTS_THREAD_LOCAL_STORAGE
~~~

`select`s are known to cause problem like this (conflicting symbol definition) and there's a known best practice to avoid them in general. Here my humble guess is that 'config PICOLIBC's select is simply incorrect, because to the best of my knowledge, thread-local storage requires support from the compiler to generate the right code.

An alternative to fixing the `select` might be to predicate PICOLIBC_SUPPORTED on Zephyr SDK, but that looks like a rather crude workaround.

pfalcon, 2022-07-04:
Have been failing for a while in Linaro CI, e.g. https://ci.linaro.org/job/zephyr-upstream/6891/ . cc: @microbuilder

keith-packard, 2022-07-04:
This was added before picolibc could be used as a module -- at that point, we needed to know if the toolchain had picolibc built with TLS support. And, at that point, picolibc required TLS support to work correctly. Now, picolibc can be built to use Zephyr's errno implementation, but that requires picolibc be compiled with different options, which is easy if it's built as a module, and depends on the toolchain to be configured correctly if not.

Ideally, we'd autodetect whether picolibc was built with TLS support when it isn't built as a module, if not, then autodetect whether picolibc has the correct errno support, and if both are false, fail the configuration. Those autodetection operations require checking in the picolibc header file for various definitions. I know how to do that with meson, but am unsure how to manage that with cmake.

When built as a module, it would be best if we enabled TLS support and used that when supported. That sounds like
we should add a `default y if PICOLIBC` to the `config THREAD_LOCAL_STORAGE` block?

When picolibc is not built as a module, we need to *require* `THREAD_LOCAL_STORAGE` if picolibc is built to support it. Could that be solved with a `depends on THREAD_LOCAL_STORAGE && !PICOLIBC_MODULE && TOOLCHAIN_SUPPORTS_THREAD_LOCAL_STORAGE` in the `config PICOLIBC` block?

pfalcon, 2022-07-04:
@keith-packard: Sadly, I can't answer those specific questions, beyond the generic: Picolibc support was of course a great endeavour, and expectably brings various complexities on its own, like supporting "builtin" picolibs, or as a module, etc. That's why it took so much time to get merged, and thanks for patiently leading that in. On the other hand, last time I heard, support for different toolchains (including not open source) was also a goal for Zephyr, in this regard, gnuarmemb is a benchmark "another" toolchain whose support was there right from the beginning, and whose support makes sense to maintain (that's why we have it in CI for example).

So, if it's the case that Picolibc support wasn't really tested with other toolchains but Zephyr SDK, and there's no easy way to fix it right now in a generic way due to those "as-module" vs "not" issues, then maybe indeed it would make sense to just short-circuit Picolibc support on Zephyr SDK, and let someone else interested to use Picolibc with another toolchain to deal with those issues later. Thanks.

keith-packard, 2022-07-04:
Sounds like we could use help from someone able to answer these questions, especially questions about getting cmake to detect the configuration of a toolchain-provided picolibc? Any idea who that might be?

keith-packard, 2022-07-04:
Also, it would be great to have picolibc included in gnuarmemb at some point -- that's the only way we could get full C++ support using picolibc. Right now, the only toolchain with picolibc and C++ support is crosstool-ng. Building picolibc as a module means no C++ support, alas.

stephanosio, 2022-07-05:
> Sounds like we could use help from someone able to answer these questions, especially questions about getting cmake to detect the configuration of a toolchain-provided picolibc? Any idea who that might be?

@tejlmand is a CMake expert and would be the right person to answer this question.

> This was added before picolibc could be used as a module -- at that point, we needed to know if the toolchain had picolibc built with TLS support.

I am not sure if the toolchain-embedded picolibc has any practical applications given that we already support the picolibc as a module. The module approach is superior in every aspect I can imagine (especially, the build time configurability aspect) and there is no practical reason for one to choose the toolchain-embedded one over the module.

libstdc++ support with the picolibc, as you noted, would require the toolchain-embedded approach; however, for any practical purposes, the newlib can serve the same purpose especially given that the C++ support was never meant to be for the highly resource constrained devices (where the picolibc is actually needed).

Nevertheless, the libc config autodetection in the CMake would be something very valuable and useful for both the toolchain-embedded picolibc and the newlib, so it would be interesting to investigate how we can implement this (maybe it would make more sense to write a Python script to do this?).

keith-packard, 2022-07-05:
> libstdc++ support with the picolibc, as you noted, would require the toolchain-embedded approach; however, for any practical purposes, the newlib can serve the same purpose especially given that the [...]

I'd love to see Zephyr use picolibc on all targets; newlib is a poor fit for embedded platforms as it cannot fix longstanding API issues caused by compatibility requirements for Cygwin. And, gcc does support picolibc in libstdc++, so there's no need for newlib even in that case. I've heard rumors that libstdc++ may get split out from gcc; if that happens, then creating a module for that would be easy. Until then, creating a module for libstdc++ would dramatically increase compile times for Zephyr as building it today involves building most of gcc ...

> Nevertheless, the libc config autodetection in the CMake would be something very valuable and useful for both the toolchain-embedded picolibc and the newlib, so it would be interesting to investigate how we can implement this (maybe it would make more sense to write a Python script to do this?).

That would be cool. Could this be done as part of the Kconfig bits? Or would it need to be done afterwards in cmake?

PR #48900 (2022-08-11): picolibc: Don't select TLS without toolchain support
If the architecture has TLS support, but the toolchain doesn't, then
don't enable Zephyr TLS support when selecting picolibc.

Closes #47275
commit 8865d4d7 (2022-08-11): picolibc: Don't select TLS without toolchain support

If the architecture has TLS support, but the toolchain doesn't, then
don't enable Zephyr TLS support when selecting picolibc.

Closes: #47275.
closed 2022-08-11 by carlescufi