zephyrproject-rtos/zephyr #47300: intel_adsp_cavs25: Failure in tests/lib/spsc_pbuf

shuai1x, 2022-07-05:
**Describe the bug**
On intel_adsp_cavs25 platform, tests/lib/spsc_pbuf/ failed. 

Please also mention any information which could help others to understand
the problem you're facing:
 - intel_adsp_cavs25
 - tests/lib/spsc_pbuf  No Console Output(Timeout)

**To Reproduce**
Steps to reproduce the behavior:

1. twister -W --hardware-map /home/ztest/cavs/cavs.map --device-testing -x=CONFIG_BOOT_DELAY=500 -T tests/lib/spsc_pbuf/ --no-skipped-report -vv
2. See error

**Logs and console output**
```
START - test_stress
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/kernel/sched.c:1764
aborted _current back from dead
E:  ** FATAL EXCEPTION
E:  ** CPU 0 EXCCAUSE 63 (zephyr exception)
E:  **  PC 0xbe0124ee VADDR (nil)
E:  **  PS 0x60b20
E:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:11 CALLINC:2)
E:  **  A0 0xbe0120b8  SP 0xbe01cf00  A2 0x4  A3 0x9e022b5c
E:  **  A4 0x1  A5 0x60b20  A6 0x1f  A7 0x1
E:  **  A8 0x1  A9 (nil) A10 0x1 A11 0x9e022a58
E:  ** A12 0x9e0224c8 A13 0x21 A14 0xbe01cef0 A15 0x4
E:  ** LBEG 0xbe013071 LEND 0xbe013087 LCOUNT (nil)
E:  ** SAR 0x4
Backtrace:0xBE0124EB:0xBE01CF00 0xBE0120B5:0xBE01CF10 0xBE016FF8:0xBE01CF30 0xBE011773:0xBE01CF70 
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x9e0224c8 (ztress_0)
E: Halting system
0% remaining:4000 ms
0% remaining:2999 ms
0% remaining:1999 ms
0% remaining:999 ms
0% remaining:0 ms
0% remaining:0 ms
0% remaining:0 ms
0% remaining:0 ms
0% remaining:0 ms
0% remaining:0 ms
0% remaining:0 ms
0% remaining:0 ms
0% remaining:0 ms
0% remaining:0 ms
0% remaining:0 ms
0% remaining:0 ms
0% remaining:0 ms
0% remaining:0 ms
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain - SDK 14.1
 - Commit SHA: fa055f743fd870809ed2a22dc41f6fb34a6eea46

smrtos, 2022-07-07:
@nordic-krch  Could you take a look at this issue?

smrtos, 2022-07-12:
@nordic-krch  Could you help check this issue? Thx!

laurenmurphyx64, 2022-07-18:
Also observing other similar tests failing with patches on top of commit for CAVS18 in addition to CAVS25:

```
tests/lib/spsc_pbuf/lib.spsc_pbuf_utilization
tests/lib/spsc_pbuf/lib.spsc_pbuf_cache
tests/lib/spsc_pbuf/lib.spsc_pbuf_nocache
tests/lib/spsc_pbuf/lib.spsc_pbuf
```

smrtos, 2022-08-02:
This issue has been reproduced on cavs25. Investigating it.

smrtos, 2022-08-03:
Saw two changes related to spsc_pbuf tests.
https://github.com/zephyrproject-rtos/zephyr/commit/3a369c47b3398878bc5112fd9107f0c1dda1de1e
https://github.com/zephyrproject-rtos/zephyr/commit/09b41829a8aad8ca91db87a2f175aa2e8cbb0a75

smrtos, 2022-08-04:
Tried to reproduce with below command on cavs25:
`twister --hardware-map ~/daily_qa/cavs25.map --device-testing -vv -s tests/lib/spsc_pbuf/lib.spsc_pbuf`

Saw below log
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.1.0-3042-g8e55e59c5917  ***
DEBUG   - DEVICE: Running TESTSUITE test_spsc_pbuf
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_0cpy
DEBUG   - DEVICE: PASS - test_0cpy in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_0cpy_corner1
DEBUG   - DEVICE: PASS - test_0cpy_corner1 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_0cpy_corner2
DEBUG   - DEVICE: PASS - test_0cpy_corner2 in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_0cpy_discard
DEBUG   - DEVICE: PASS - test_0cpy_discard in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_0cpy_smaller
DEBUG   - DEVICE: PASS - test_0cpy_smaller in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_largest_alloc
DEBUG   - DEVICE: PASS - test_largest_alloc in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_spsc_pbuf_ut
DEBUG   - DEVICE: PASS - test_spsc_pbuf_ut in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_spsc_pbuf_ut_cache
DEBUG   - DEVICE: PASS - test_spsc_pbuf_ut_cache in 0.1 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_stress
DEBUG   - DEVICE: 2% remaining:3999 ms
DEBUG   - DEVICE: 4% remaining:2999 ms
DEBUG   - DEVICE: 8% remaining:1999 ms
DEBUG   - DEVICE: 11% remaining:999 ms
DEBUG   - DEVICE: 14% remaining:0 ms
DEBUG   - DEVICE: Ztress execution report:
DEBUG   - DEVICE: context 0:
DEBUG   - DEVICE: - executed:22633, preempted:1798
DEBUG   - DEVICE: - ticks initial:4, optimized:4
DEBUG   - DEVICE: context 1:
DEBUG   - DEVICE: - executed:22330, preempted:143
DEBUG   - DEVICE: - ticks initial:4, optimized:4
DEBUG   - DEVICE: Average CPU load:55%, measurements:49
DEBUG   - DEVICE: 2% remaining:3999 ms
DEBUG   - DEVICE: 4% remaining:2999 ms
DEBUG   - DEVICE: 6% remaining:1999 ms
DEBUG   - DEVICE: 6% remaining:999 ms
DEBUG   - DEVICE: 7% remaining:0 ms
DEBUG   - DEVICE: Ztress execution report:
DEBUG   - DEVICE: context 0:
DEBUG   - DEVICE: - executed:22722, preempted:614
DEBUG   - DEVICE: - ticks initial:4, optimized:4
DEBUG   - DEVICE: context 1:
DEBUG   - DEVICE: - executed:22452, preempted:75
DEBUG   - DEVICE: - ticks initial:4, optimized:4
DEBUG   - DEVICE: Average CPU load:65%, measurements:49
DEBUG   - DEVICE: PASS - test_stress in 10.23 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_stress_0cpy
DEBUG   - DEVICE: 0% remaining:3999 ms
DEBUG   - DEVICE: 2% remaining:2999 ms
DEBUG   - DEVICE: 3% remaining:1999 ms
DEBUG   - DEVICE: 4% remaining:999 ms
DEBUG   - DEVICE: 6% remaining:0 ms
DEBUG   - DEVICE: Ztress execution report:
DEBUG   - DEVICE: context 0:
DEBUG   - DEVICE: - executed:22776, preempted:702
[... 29 lines cut ...]
DEBUG   - DEVICE:
DEBUG   - DEVICE:
DEBUG   - DEVICE: Backtrace:
DEBUG   - DEVICE:
DEBUG   - DEVICE: Backtrace:0xBE01263F:0xBE01CEF0 0xBE01263F:0xBE01D6F0 0xBE0120C1:0xBE01CF00 0xBE0120C1:0xBE01D700 0xBE01224E:0xBE01CF20 0xBE01224E:0xBE01D720 0xBE010B5D:0xBE01CF40 0xBE010B5D:0xBE01D740 0xBE013A1B:0xBE01CF70 0xBE013A1B:0xBE01D770 0xBE01177F:0xBE01CFB0 0xBE01177F:0xBE01D7B0
DEBUG   - DEVICE:
DEBUG   - DEVICE:
DEBUG   - DEVICE:
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 1
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 3
DEBUG   - DEVICE: E: Current thread: 0x9e022648 (ztress_0)
DEBUG   - DEVICE: E: Current thread: 0x9e0226f8 (ztress_1)
DEBUG   - DEVICE: E: Halting system
DEBUG   - DEVICE: E: Halting system
DEBUG   - DEVICE: 0% remaining:3999 ms
DEBUG   - DEVICE: 0% remaining:2999 ms
DEBUG   - DEVICE: 0% remaining:1999 ms
DEBUG   - DEVICE: 0% remaining:999 ms
DEBUG   - DEVICE: 0% remaining:0 ms
DEBUG   - DEVICE: 0% remaining:0 ms
DEBUG   - DEVICE: 0% remaining:0 ms
DEBUG   - DEVICE: 0% remaining:0 ms
DEBUG   - DEVICE: 0% remaining:0 ms
DEBUG   - DEVICE: 0% remaining:0 ms
```

smrtos, 2022-08-08:
Without any change, the 2 tests can sometimes pass on cavs25:

```
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_stress
DEBUG   - DEVICE: 14% remaining:4000 ms
DEBUG   - DEVICE: 28% remaining:3000 ms
DEBUG   - DEVICE: 46% remaining:2000 ms
DEBUG   - DEVICE: 63% remaining:1000 ms
DEBUG   - DEVICE: 82% remaining:0 ms
DEBUG   - DEVICE: Ztress execution report:
DEBUG   - DEVICE: context 0:
DEBUG   - DEVICE: - executed:23703, preempted:1227
DEBUG   - DEVICE: - ticks initial:4, optimized:4
DEBUG   - DEVICE: context 1:
DEBUG   - DEVICE: - executed:23425, preempted:821
DEBUG   - DEVICE: - ticks initial:4, optimized:4
DEBUG   - DEVICE: Average CPU load:46%, measurements:50
DEBUG   - DEVICE: 10% remaining:4000 ms
DEBUG   - DEVICE: 23% remaining:3000 ms
DEBUG   - DEVICE: 33% remaining:2000 ms
DEBUG   - DEVICE: 42% remaining:1000 ms
DEBUG   - DEVICE: 54% remaining:0 ms
DEBUG   - DEVICE: Ztress execution report:
DEBUG   - DEVICE: context 0:
DEBUG   - DEVICE: - executed:24135, preempted:576
DEBUG   - DEVICE: - ticks initial:4, optimized:4
DEBUG   - DEVICE: context 1:
DEBUG   - DEVICE: - executed:23556, preempted:540
DEBUG   - DEVICE: - ticks initial:4, optimized:4
DEBUG   - DEVICE: Average CPU load:71%, measurements:50
DEBUG   - DEVICE: PASS - test_stress in 10.023 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_stress_0cpy
DEBUG   - DEVICE: 12% remaining:4000 ms
DEBUG   - DEVICE: 28% remaining:3000 ms
DEBUG   - DEVICE: 42% remaining:2000 ms
DEBUG   - DEVICE: 54% remaining:1000 ms
DEBUG   - DEVICE: 67% remaining:0 ms
DEBUG   - DEVICE: Ztress execution report:
DEBUG   - DEVICE: context 0:
DEBUG   - DEVICE: - executed:24155, preempted:719
DEBUG   - DEVICE: - ticks initial:4, optimized:4
DEBUG   - DEVICE: context 1:
DEBUG   - DEVICE: - executed:23628, preempted:676
DEBUG   - DEVICE: - ticks initial:4, optimized:4
DEBUG   - DEVICE: Average CPU load:78%, measurements:50
DEBUG   - DEVICE: 9% remaining:4000 ms
DEBUG   - DEVICE: 18% remaining:3000 ms
DEBUG   - DEVICE: 26% remaining:2000 ms
DEBUG   - DEVICE: 34% remaining:1000 ms
DEBUG   - DEVICE: 44% remaining:0 ms
DEBUG   - DEVICE: Ztress execution report:
DEBUG   - DEVICE: context 0:
DEBUG   - DEVICE: - executed:24008, preempted:910
DEBUG   - DEVICE: - ticks initial:4, optimized:4
DEBUG   - DEVICE: context 1:
DEBUG   - DEVICE: - executed:23562, preempted:441
DEBUG   - DEVICE: - ticks initial:4, optimized:4
DEBUG   - DEVICE: Average CPU load:79%, measurements:50
DEBUG   - DEVICE: PASS - test_stress_0cpy in 10.023 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_utilization
DEBUG   - DEVICE: PASS - test_utilization in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
```

smrtos, 2022-08-08:
Some failed logs indicate that alignment is related:

```
START - test_stress
E:  ** FATAL EXCEPTION
E:  ** CPU 0 EXCCAUSE 9 (load/store alignment)  <=================== HERE
E:  **  PC 0x9e022558 VADDR 0x9
E:  **  PS 0x60b20
E:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:11 CALLINC:2)
E:  **  A0 0xbe011756  SP 0xbe01d7b0  A2 0xbe01c788  A3 0x1
E:  **  A4 0x9e022d64  A5 0x9e022638  A6 0x1  A7 0x1
E:  **  A8 0xbe013996  A9 0x9e022d78 A10 0x9e022db4 A11 0x1
E:  ** A12 (nil) A13 0x1 A14 0x4 A15 0x60a22
E:  ** LBEG (nil) LEND (nil) LCOUNT (nil)
E:  ** SAR 0x6
```

smrtos, 2022-08-08:
Some failed logs indicate it is a PIF data error:

```
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_stress_0cpy
DEBUG   - DEVICE: E:  ** FATAL EXCEPTION
DEBUG   - DEVICE: E:  ** CPU 1 EXCCAUSE 12 (instr PIF data error) <=================== HERE
DEBUG   - DEVICE: E:  **  PC 0x4 VADDR (nil)
DEBUG   - DEVICE: E:  **  PS 0x60020
DEBUG   - DEVICE: E:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:0 CALLINC:2)
DEBUG   - DEVICE: E:  **  A0 0xbe01177a  SP 0xbe01d0f0  A2 0xbe01c8c0  A3 (nil)
DEBUG   - DEVICE: E:  **  A4 0x9e022e3c  A5 0x9e022630  A6 (nil)  A7 (nil)
DEBUG   - DEVICE: E:  **  A8 0xbe013bfe  A9 0x9e022e54 A10 (nil) A11 (nil)
DEBUG   - DEVICE: E:  ** A12 0x1 A13 (nil) A14 0xbe01d0ec A15 0x60722
DEBUG   - DEVICE: E:  ** LBEG (nil) LEND (nil) LCOUNT (nil)
DEBUG   - DEVICE: E:  ** SAR 0x20
```

According to [here](https://mirrobo.ru/wp-content/uploads/2016/11/Cadence_Tensillica_Xtensa_LX6_ds.pdf), PIF = Processor Interface

smrtos, 2022-08-08:
Some failed logs indicate it is an assertion failure:

```
ASSERTION FAIL [bytes_stored >= (len + sizeof(uint32_t))] @ WEST_TOPDIR/zephyr/lib/os/spsc_pbuf.c:296
ASSERTION FAIL [bytes_stored >= (len + sizeof(uint32_t))] @ WEST_TOPDIR/zephyr/lib/os/spsc_pbuf.c:296
```

smrtos, 2022-08-08:
Some failed logs indicate it is related to schedule:

```
START - test_stress
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/kernel/sched.c:1764  <============= HRER
aborted _current back from dead
E:  ** FATAL EXCEPTION
E:  ** CPU 1 EXCCAUSE 63 (zephyr exception)
E:  **  PC 0xbe01276a VADDR (nil)
E:  **  PS 0x60f20
E:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:15 CALLINC:2)
E:  **  A0 0xbe0121a8  SP 0xbe01cfc0  A2 0x4  A3 0x9e022cdc
E:  **  A4 0x1  A5 0x60f20  A6 0x1  A7 (nil)
E:  **  A8 0x1  A9 (nil) A10 0x1 A11 0x9e022c08
E:  ** A12 0x9e022649 A13 0x21 A14 0xbe01cfb0 A15 0x4
E:  ** LBEG (nil) LEND (nil) LCOUNT (nil)
E:  ** SAR 0x4
Backtrace:0xBE012767:0xBE01CFC0 0xBE0121A5:0xBE01CFD0 0xBE017258:0xBE01CFF0 0xBE011863:0xBE01D030 
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 1
E: Current thread: 0x9e022648 (ztress_0)
E: Halting system
```

smrtos, 2022-08-11:
Change the title since all cavs platforms are affected by this issue.

nordic-krch, 2022-08-11:
@smrtos  `0cpy` means special mode where data is not copied into the buffer but rather first space within the buffer is allocated. Data is written directly to that space and then buffer is commited.

smrtos, 2022-08-23:
On CAVS25 with `8403866e39b3e0`.

Below command can pass:
`twister --hardware-map ~/daily_qa/cavs25.map --device-testing -vv -s tests/lib/spsc_pbuf/lib.spsc_pbuf`

Below commands failed.
The failed case is always `test_stress_0cpy`.
The `test_stress` can pass.

```
twister --hardware-map ~/daily_qa/cavs25.map --device-testing -vv -s tests/lib/spsc_pbuf/lib.spsc_pbuf_cache (sometimes)
twister --hardware-map ~/daily_qa/cavs25.map --device-testing -vv -s tests/lib/spsc_pbuf/lib.spsc_pbuf_nocache (sometimes)
twister --hardware-map ~/daily_qa/cavs25.map --device-testing -vv -s tests/lib/spsc_pbuf/lib.spsc_pbuf_utilization (seems always)
```

smrtos, 2022-08-27:
I tried 1986591beb68e59c8a8e52f0a3f3e9762cf2a11d which first introduced the `test_stress_0cpy` test on Jun. 30.
It still failed on CAVS25.

So this test never passed on CAVS boards.

smrtos, 2022-08-27:
I tried https://github.com/zephyrproject-rtos/zephyr/commit/1986591beb68e59c8a8e52f0a3f3e9762cf2a11d which first introduced the test_stress_0cpy test on Jun. 30.
It still failed on CAVS25.

So this test has never passed on CAVS boards.

smrtos, 2022-08-27:
In the ztress framework, if I don't call the handler as below line 265. The test never fail.

https://github.com/zephyrproject-rtos/zephyr/blob/5108c4f21d3d1ac3ca143cfaefe33f80e738d5d5/subsys/testsuite/ztest/src/ztress.c#L264-L266

So there must be something wrong with the handler invoking.
Then I add logic to compare the `context_data->handler` value to the 2 handlers `stress_claim_free` and `stress_alloc_commit`.

https://github.com/zephyrproject-rtos/zephyr/blob/5108c4f21d3d1ac3ca143cfaefe33f80e738d5d5/tests/lib/spsc_pbuf/src/main.c#L488

https://github.com/zephyrproject-rtos/zephyr/blob/5108c4f21d3d1ac3ca143cfaefe33f80e738d5d5/tests/lib/spsc_pbuf/src/main.c#L512

I add below logic right before the above line 265 

```
	 	// added start
		extern uint32_t handler1, handler2;
		if((uint32_t)context_data->handler != handler1 && (uint32_t)context_data->handler != handler2 ) {
			printk("\n >>> context_data->handler = %p, expected: %x or %x\n", context_data->handler, handler1, handler2);
		}
		// added end

	   cont = context_data->handler(context_data->user_data, cnt, !cont_test, priority);
```

And I saw the handler data can some unexpected data when the test failed:

```
DEBUG   - DEVICE: sm:--- [stress_claim_free] = be010b5c
DEBUG   - DEVICE: sm:--- [stress_alloc_commit] = be011414

...<omitted>

DEBUG   - DEVICE: >>> context_data->handler = 0x1, expected: be010b5c or be011414  <========= HERE!

DEBUG   - DEVICE: E:  ** FATAL EXCEPTION
DEBUG   - DEVICE: E:  ** CPU 0 EXCCAUSE 12 (instr PIF data error)
DEBUG   - DEVICE: E:  **  PC 0x1 VADDR (nil)
DEBUG   - DEVICE: E:  **  PS 0x60620
DEBUG   - DEVICE: E:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:6 CALLINC:2)
DEBUG   - DEVICE: E:  **  A0 0xbe0115f2  SP 0xbe01d8f0  A2 0xbe01c8d8  A3 0x1
DEBUG   - DEVICE: E:  **  A4 0x9e022dbc  A5 0x9e0225b8  A6 0x1  A7 (nil)
DEBUG   - DEVICE: E:  **  A8 0xbe013a88  A9 0xbe01d8a0 A10 (nil) A11 (nil)
DEBUG   - DEVICE: E:  ** A12 (nil) A13 0x1 A14 0xbe01d8ec A15 0x60522
DEBUG   - DEVICE: E:  ** LBEG (nil) LEND (nil) LCOUNT (nil)
DEBUG   - DEVICE: E:  ** SAR 0x5
```

This should be the cause of failure. But not sure why this is happening. Still investigating.

(Many thanks to @mengxianglinx )

smrtos, 2022-08-28:
Found another issue through code review. It's about progress stats, trying to fix it with https://github.com/zephyrproject-rtos/zephyr/pull/49604

But it is not fixing this issue.

edersondisouza, 2022-08-29:
If you suspect some stack overflow is happening, you can try some things, like a debugger and watch when the stack variable gets overwritten, or enabling CONFIG_STACK_SENTINEL/CONFIG_STACK_CANARIES, to catch when it's happening. Increasing stack size is another valid test. If none of those helps, it maybe, as you also mentioned that this only happen with more than one core, some context switch issue.

edersondisouza, 2022-08-30:
Hmm, I think that this is falling in the cache incoherence trap. And based on some fixes for this issue (like 4796037cf721c6f6416d446e99ec02521542b949), it seems that having `static data` in the `ZTRESS_EXECUTE` macro is the way to go. Can you send a PR for that?

smrtos, 2022-08-30:
> Hmm, I think that this is falling in the cache incoherence trap. And based on some fixes for this issue (like [4796037](https://github.com/zephyrproject-rtos/zephyr/commit/4796037cf721c6f6416d446e99ec02521542b949)), it seems that having `static data` in the `ZTRESS_EXECUTE` macro is the way to go. Can you send a PR for that?

Yeah, the `CONFIG_KERNEL_COHERENCE=y` does exist in the final `.config` file.  It seems very likely! 
BTW, I discussed this with @enjiamai  yesterday and he also doubted about the cache coherence.

I can send a PR for this test. But it seems there are still some missing pieces of the puzzle. 
Why the scheduler doesn't ensure the coherence when switching contexts? 
As I understand, all access to memory go through the cache, why the data in `.bss` section doesn't suffer from this?

edersondisouza, 2022-08-30:
Stack is not made coherent - that's the whole issue. Note that while the config option is called "kernel_coherence", it's is actually there to deal with incoherent cache: https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_KERNEL_COHERENCE.
It's illegal to share data on the stack among the CPUs, as stack is in the cache (which is incoherent). `.bss` section is put on uncached memory, thus, doesn't suffer the incoherence issue.

smrtos, 2022-08-30:
> Stack is not made coherent - that's the whole issue. Note that while the config option is called "kernel_coherence", it's is actually there to deal with incoherent cache: [docs.zephyrproject.org/latest/kconfig.html#CONFIG_KERNEL_COHERENCE](https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_KERNEL_COHERENCE). It's illegal to share data on the stack among the CPUs, as stack is in the cache (which is incoherent). `.bss` section is put on uncached memory, thus, doesn't suffer the incoherence issue.

It seems to be an architecture specific thing. Some memories are cacheable while some are not.
And it seems only specific to the Xtensa because I didn't see such issue on other platforms so far.

A naïve guess maybe I can get some info from the memory map.  Or is it configurable through the MMU?
I got a Xtensa data book. Not sure if I can find answer in it.

smrtos, 2022-08-31:
Other issues related to incoherent cache:

- https://github.com/zephyrproject-rtos/zephyr/pull/49707
- https://github.com/zephyrproject-rtos/zephyr/pull/41932

It's interesting that such issues are emerging recently.

andyross, 2022-08-31:
Xtensa tests on intel_adsp have been dealing with fixes like this consistently since KERNEL_COHERENCE landed, actually.  It's, heh, just the staff that changed. :)

But yes, the rules are: when enabled (which only happens now for SMP Xtensa platforms, but in principle other cache-incoherent architectures might behave similarly), thread stack memory is cached/incoherent and must be treated as local to the currently executing CPU.   The kernel will handle flushing appropriately for you when the thread context switches.  In general you shouldn't share it with other threads at all, but it's possible to do so as long as you pad and align your data to a full cache line (which is variable-sized, check core-isa.h, though right now all affected devices are 64 bytes) and use the Xtensa cache API (don't use the HAL, use our code, it's better) to flush changes and invalidate before use.  There is automatic support in the kernel that detects obvious mistakes like putting spinlocks or waitq's (i.e. typical IPC primitives) on the stack.

closed 2022-09-01 by carlescufi