zephyrproject-rtos/zephyr #47307: tests: kernel: fatal: exception: failed on multiple platforms

LixinGuoX, 2022-07-05:
**Describe the bug**
kernel.common.stack_protection_arm_fpu_sharing failed on nrf9160dk_nrf9160_ns, lpcxpresso55s69_ns, bl5340_dvk_cpuapp_ns.

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p nrf9160dk_nrf9160_ns -p lpcxpresso55s69_ns -p bl5340_dvk_cpuapp_ns -T tests/kernel/fatal/
2. See error

**Expected behavior**
Tests should pass.

**Impact**
CI got blocked.

**Logs and console output**
```
Loading Zephyr default modules (Zephyr base).
-- Application: /home/guo/zephyrproject/zephyr/tests/kernel/fatal/exception
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /home/guo/.cache/zephyr
-- Zephyr version: 3.1.99 (/home/guo/zephyrproject/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: nrf9160dk_nrf9160_ns, Revision: 0.7.0
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.14.2 (/home/guo/zephyr-sdk-0.14.2)
-- Found dtc: /home/guo/zephyr-sdk-0.14.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.14.2 (/home/guo/zephyr-sdk-0.14.2)
-- Found BOARD.dts: /home/guo/zephyrproject/zephyr/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160_ns.dts
-- Generated zephyr.dts: /home/guo/zephyrproject/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/kernel/fatal/exception/kernel.common.stack_protection_arm_fpu_sharing/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/guo/zephyrproject/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/kernel/fatal/exception/kernel.common.stack_protection_arm_fpu_sharing/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/guo/zephyrproject/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/kernel/fatal/exception/kernel.common.stack_protection_arm_fpu_sharing/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/guo/zephyrproject/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/kernel/fatal/exception/kernel.common.stack_protection_arm_fpu_sharing/zephyr/dts.cmake
Parsing /home/guo/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/guo/zephyrproject/zephyr/boards/arm/nrf9160dk_nrf9160/nrf9160dk_nrf9160_ns_defconfig'
Merged configuration '/home/guo/zephyrproject/zephyr/tests/kernel/fatal/exception/prj_arm_fpu_sharing.conf'
Configuration saved to '/home/guo/zephyrproject/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/kernel/fatal/exception/kernel.common.stack_protection_arm_fpu_sharing/zephyr/.config'
Kconfig header saved to '/home/guo/zephyrproject/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/kernel/fatal/exception/kernel.common.stack_protection_arm_fpu_sharing/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/guo/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/guo/zephyrproject/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/kernel/fatal/exception/kernel.common.stack_protection_arm_fpu_sharing
[  0%] Generating ../../tfm/CMakeCache.txt
-- The C compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/guo/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Error at config/check_config.cmake:19 (message):
  INVALID CONFIG: ( NOT TFM_PSA_API ) AND CONFIG_TFM_FP = hard
Call Stack (most recent call first):
  config/check_config.cmake:59 (tfm_invalid_config)
  CMakeLists.txt:70 (include)

-- Configuring incomplete, errors occurred!
See also "/home/guo/zephyrproject/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/kernel/fatal/exception/kernel.common.stack_protection_arm_fpu_sharing/tfm/CMakeFiles/CMakeOutput.log".
See also "/home/guo/zephyrproject/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/kernel/fatal/exception/kernel.common.stack_protection_arm_fpu_sharing/tfm/CMakeFiles/CMakeError.log".
make[2]: *** [modules/trusted-firmware-m/CMakeFiles/tfm_cmake.dir/build.make:73: tfm/CMakeCache.txt] Error 1
make[2]: *** Deleting file 'tfm/CMakeCache.txt'
make[1]: *** [CMakeFiles/Makefile2:4095: modules/trusted-firmware-m/CMakeFiles/tfm_cmake.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

stephanosio, 2022-07-05:
probably related to https://github.com/zephyrproject-rtos/zephyr/pull/45906

closed 2022-07-06 by microbuilder

joerchan, 2022-07-18:
FP hard ABI is only supported with IPC model in TF-M. This was a missing dependency. Thanks for the bug-report.