zephyrproject-rtos/zephyr #47341: net: sockets: dtls: define behavior of connect() regarding DTLS handshake

mniestroj, 2022-07-05:
## Introduction

Right now it is not clear what **should** be the behavior of `connect()` when used with DTLS sockets, created with https://github.com/zephyrproject-rtos/zephyr/blob/ec8a6c83fa988a933e9dcd2d6c73f1fb496c4463/include/zephyr/net/net_ip.h#L79-L80

### Problem description

`connect()` on DTLS sockets just assigns server address, so that subsequent `recv*()` and `send*()` calls know which address to use. This is the behavior of UDP sockets. The difference between UDP and DTLS is that with UDP there is no notion of connection, but with DTLS there is a connection (also described as association or session in [RFC 6347](https://datatracker.ietf.org/doc/html/rfc6347)).

According to `connect()` documentation in POSIX specification:

>The connect() function shall attempt to make a connection on a connection-mode
socket or to set or reset the peer address of a connectionless-mode socket.

and

>If the initiating socket is connection-mode, then connect() shall attempt to
establish a connection to the address specified by the address argument. If the
connection cannot be established immediately and O_NONBLOCK is not set for the
file descriptor for the socket, connect() shall block for up to an unspecified
timeout interval until the connection is established.

which means that for connection-mode sockets the connection should be established as part of (successful) `connect()` call.

### Proposed change

Define behavior of `connect()` when used with DTLS sockets. Document that `connect()` on DTLS sockets should result with initiating handshake and successful `connect()` (0 return code) means that handshake was done.

## Detailed RFC

TODO

### Proposed change (Detailed)

TODO

### Concerns and Unresolved Questions

TODO

## Alternatives

TODO

mniestroj, 2022-07-07:
>We should retain the current behaviour however, so that in case the socket is in an unconnected state sendto()/recv() would also initiate the handshake, depending on the DTLS role.

I've investigated many (D)TLS libraries (including openssl and mbedtls) and most (if not all) of them do implicit handshake as part of send/recv. So in theory there is not much work to do on Zephyr TLS socket layer implementation. The rationale behind it is that handshake can be retriggered by TLS layer at some point, so it would be too complex to bother application layer, at least to what documentation of those (D)TLS libraries say. So overall I agree with your statement.

>In any case, if connect() was not called or the handshake was disabled on connect(), the handshake would still be initiated automatically from within sendto()/recv()

**Currently** if `connect` was not called, then there is no information about server address, so handshake cannot be initiated. However if we would introduce an option to:
1) inject transport layer socket, e.g. with `setsockopt(dtls_sock, ..., SET_TRANSPORT_SOCKET, udp_sock);`,
2) "connect" just to transport layer, e.g. with `setsockopt(dtls_sock, ..., CONNECT_TRANSPORT, (struct addrinfo *) {"ip", port});`,

then calling `connect()` would not be needed in those cases.

rlubos, 2022-07-07:
> **Currently** if `connect` was not called, then there is no information about server address, so handshake cannot be initiated. 

This is only true if you call `send()`, then indeed there's no address information. You can still call `sendto()` however in such case to provide the address and then handshake **is** initiated.

boaks, 2022-07-14:
In my experience, there are many using UDP without "connect" in order to use a single socket to communicate with a couple of other peers. Therefore I don't think, that couple DTLS to a "UDP connect" makes too much sense. The DTLS implementations usually keeps track of the security context in map based on `ip-address:port` or `Connection ID` and doesn't require that "connect".
So, I would prefer, if the "UDP connect" stay as option for UDP, and doesn't get mandatory for DTLS.

mniestroj, 2022-08-10:
@boaks Sure. As @rlubos suggested, it will still be possible to use `sendto()` and pass address directly, so `connect()` won't be mandatory.

PR #66695 (2024-01-03): tests: net: sockets: Extend TLS test suite (part 2)
Second batch of TLS/DTLS sockets tests, covering `poll()` functionality. This time new tests revealed some issues related to poll() processing with TLS/DTLS sockets, so the tests are accompanied with several fixes/improvements. For details, please see commit descriptions.

Resolves #47341
closed 2024-01-03 by fabiobaltieri