zephyrproject-rtos/zephyr #47407: stm32l5: tfm: Build error on tests/arch/arm/arm_thread_swap_tz

erwango, 2022-07-06:
**Describe the bug**
A clear and concise description of what the bug is.

```
/local/mcu/zephyrproject/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/stm/common/hal/accelerator/aes_alt.c:825:6: warning: no previous prototype for 'mbedtls_aes_encrypt' [-Wmissing-prototypes]
  825 | void mbedtls_aes_encrypt(mbedtls_aes_context *ctx,
      |      ^~~~~~~~~~~~~~~~~~~
/local/mcu/zephyrproject/modules/tee/tf-m/trusted-firmware-m/platform/ext/target/stm/common/hal/accelerator/aes_alt.c:832:6: warning: no previous prototype for 'mbedtls_aes_decrypt' [-Wmissing-prototypes]
  832 | void mbedtls_aes_decrypt(mbedtls_aes_context *ctx,
      |      ^~~~~~~~~~~~~~~~~~~

```
Note that exact same warnings were present on release v3.1.0, but the build was successful.

**To Reproduce**
```
twister -b -n -p stm32l562e_dk_ns -T tests/arch/arm/arm_thread_swap_tz
```

**Expected behavior**
Should build

**Impact**
Block CI

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Linux
 - Toolchain (e.g Zephyr SDK, ...): 0.14.0
 - Commit SHA or Version used: zephyr-v3.1.0-1605-g4c9f1ca877

erwango, 2022-07-06:
@microbuilder Any idea what could have changed in TFM compilation recently that can change the fact that warnings are now generating build error (not complaining, but I'd like to understand ;-) ) ?

microbuilder, 2022-07-06:
> @microbuilder Any idea what could have changed in TFM compilation recently that can change the fact that warnings are now generating build error (not complaining, but I'd like to understand ;-) ) ?

I'm not sure off the top of my head, but I'll reproduce and see if I spot a reason.

microbuilder, 2022-07-06:
Caused by https://github.com/zephyrproject-rtos/zephyr/pull/45906 ... reverting this to disable FP hard ABI resolves this.

erwango, 2022-07-06:
> Caused by #45906 ... reverting this to disable FP hard ABI resolves this.

Thank Kevin, so it explains why we get this issue now.
Handing over to @jamike

closed 2022-07-08 by erwango

joerchan, 2022-07-18:
The problem here appears to be the FPU support being enabled without the dependent PR to not build the NS application merged first.
Merging https://github.com/zephyrproject-rtos/zephyr/pull/43599 and reverting #47423 again this builds without issues.

The bug description here is inaccurate, the issue is this: `error: tf-m-tests/app/CMakeFiles/tfm_ns.dir/main_ns.o uses VFP register arguments, bin/tfm_ns.axf does not`.

Maybe a different issues should be opened (or this one reopened) to track the ST-M missing prototype warning?