zephyrproject-rtos/zephyr #47428: samples: bbc_microbit: pong fails to build

galak, 2022-07-06:
when trying to build samples/boards/bbc_microbit/pong

I get the following error:

```
/opt/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr_pre0.elf section `noinit' will not fit in region `SRAM'
/opt/zephyr-sdk-0.14.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: region `SRAM' overflowed by 1079 bytes
collect2: error: ld returned 1 exit status
```

Doing a git bisect, tracked it down to:

commit 3c5bf5089ad8465b82e59c57fb1ad8b78b48d971 (refs/bisect/bad)
Author: Jonathan Rico <jonathan.rico@nordicsemi.no>
Date:   Mon May 30 16:08:11 2022 +0200

    Bluetooth: host: add dedicated WQ for long-running tasks

mmahadevan108, 2022-07-06:
@jori-nordic, can you help take a look at this issue.

jori-nordic, 2022-07-11:
@carlescufi do you know why this wasn't caught in CI ? Aren't we building all the samples in the oncommit ?

PR #47591 (2022-07-11): samples: boards: fix bbc_microbit pong build
The GATT caching feature isn't needed, and was making the memory
consumption blow up because of the recent addition of `long_wq` for
deferred work.

Fixes #47428 .
commit 1f226e0c (2022-07-11): samples: boards: fix bbc_microbit pong build

The GATT caching feature isn't needed, and was making the memory
consumption blow up because of the recent addition of `long_wq` for
deferred work.

Fixes #47428 .
closed 2022-07-11 by carlescufi