zephyrproject-rtos/zephyr #47461: Unable to build the flash_shell samples with board cc1352r1_launchxl

ngboonkhai, 2022-07-07:
I was trying to check in a flash driver, however the twister test caught failing when building the sample "samples/drivers/flash_shell" with the board cc1352r1_launchxl. This has blocked me from getting the PR https://github.com/zephyrproject-rtos/zephyr/pull/46443 approved. Please fix the bug ASAP

**Board**
cc1352r1_launchxl

**What I did to resolve the issue**
1) Tried to grep the the whole zephyr, not seeing the 'NOROM_FlashProgram' reference being defined
2) Grep at outside the zephyr, module folder notice that the FlashProgram reference is defined to 'NOROM_FlashProgram' but still the 'NOROM_FlashProgram is not pointing to anywhere.
hal/ti/simplelink/source/ti/devices/cc13x2_cc26x2/driverlib/flash.c:    

**To Reproduce**
Steps to reproduce the behavior:
1.  west build -b cc1352r1_launchxl samples/drivers/flash_shell --pristine
2.  See error

**Expected behavior**
It should be build successfully.

**Impact**
This is blocking the PR https://github.com/zephyrproject-rtos/zephyr/pull/46443 

**Logs and console output**
west build -b cc1352r1_launchxl samples/drivers/flash_shell --pristine
WARNING: can't find the zephyr repository
  - no --zephyr-base given
  - ZEPHYR_BASE is unset
  - west config contains no zephyr.base setting
  - no manifest project has name or path "zephyr"

  If this isn't a Zephyr workspace, you can   silence this warning with something like this:
    west config zephyr.base not-using-zephyr
-- west build: making build dir /nfs/png/disks/swuser_work_boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr repository).
-- Application: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/samples/drivers/flash_shell
-- Found Python3: /build/zephyrproject/.venv/bin/python3.6 (found suitable exact version "3.6.15") found components: Interpreter 
-- Cache files will be written to: /nfs/site/home/boonkhai/.cache/zephyr
-- Zephyr version: 3.1.99 (/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: cc1352r1_launchxl
-- Found host-tools: zephyr 0.13.2 (/build/zephyr-sdk-0.13.2)
-- Found dtc: /build/zephyr-sdk-0.13.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.13.2 (/build/zephyr-sdk-0.13.2)
-- Found BOARD.dts: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/boards/arm/cc1352r1_launchxl/cc1352r1_launchxl.dts
-- Generated zephyr.dts: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/dts.cmake
Parsing /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/samples/drivers/flash_shell/Kconfig
Loaded configuration '/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/boards/arm/cc1352r1_launchxl/cc1352r1_launchxl_defconfig'
Merged configuration '/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/samples/drivers/flash_shell/prj.conf'
Configuration saved to '/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/.config'
Kconfig header saved to '/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /build/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build
-- west build: building application
[1/192] Preparing syscall dependency handling

[2/192] Generating include/generated/version.h
-- Zephyr version: 3.1.99 (/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr), build: zephyr-v3.1.0-1090-gfa48dbd1dbbf
[182/192] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map 
: && ccache /build/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/ti/simplelink/source/ti/devices/cc13x2_cc26x2/lib..__modules__hal__ti__simplelink__source__ti__devices__cc13x2_cc26x2.a  modules/ti/simplelink/lib..__modules__hal__ti__simplelink.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -L"/build/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/thumb/v7e-m/nofp"  -L/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr && /usr/local/bin/cmake -E echo
/build/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(soc_flash_cc13xx_cc26xx.c.obj): in function `flash_cc13xx_cc26xx_erase':
/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/drivers/flash/soc_flash_cc13xx_cc26xx.c:143: undefined reference to `NOROM_FlashSectorErase'
/build/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(soc_flash_cc13xx_cc26xx.c.obj): in function `flash_cc13xx_cc26xx_write':
/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/drivers/flash/soc_flash_cc13xx_cc26xx.c:203: undefined reference to `NOROM_FlashProgram'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /nfs/png/disks/swuser_work_boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build

**Environment (please complete the following information):**
 - Linux
 - Zephyr SDK
 - zephyr-sdk-0.13.2

cfriedt, 2022-07-07:
Have you tried `source zephyr-env.sh`?

ngboonkhai, 2022-07-07:
> Have you tried `source zephyr-env.sh`?
Hi cfriedt,  It's not about the environment issue, you see,  the definition of NOROM_FlashProgram were not found, and I couldn't find it elsewhere. it is easily replicated when you west init a new project and try to build it. I wonder how the driver can be checkin at the first place and doubt if the test are done correctly.

/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/drivers/flash/soc_flash_cc13xx_cc26xx.c:203: undefined reference to NOROM_FlashProgram'

Even if i tried to source the zephyr-env.sh it is still the same below is the output
```
[07/07/2022 - 06:50.35] [boonkhai.boonkhai-container-55d96b56d8-fgzrj]
/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr
$ source zephyr-env.sh

[07/07/2022 - 06:50.41] [boonkhai.boonkhai-container-55d96b56d8-fgzrj]
/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr
$ west build -b cc1352r1_launchxl samples/drivers/flash_shell --pristine
-- west build: making build dir /nfs/png/disks/swuser_work_boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/samples/drivers/flash_shell
-- Found Python3: /build/zephyrproject/.venv/bin/python3.6 (found suitable exact version "3.6.15") found components: Interpreter 
-- Cache files will be written to: /nfs/site/home/boonkhai/.cache/zephyr
-- Zephyr version: 3.1.99 (/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: cc1352r1_launchxl
-- Found host-tools: zephyr 0.13.2 (/build/zephyr-sdk-0.13.2)
-- Found dtc: /build/zephyr-sdk-0.13.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.13.2 (/build/zephyr-sdk-0.13.2)
-- Found BOARD.dts: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/boards/arm/cc1352r1_launchxl/cc1352r1_launchxl.dts
-- Generated zephyr.dts: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/dts.cmake
Parsing /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/samples/drivers/flash_shell/Kconfig
Loaded configuration '/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/boards/arm/cc1352r1_launchxl/cc1352r1_launchxl_defconfig'
Merged configuration '/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/samples/drivers/flash_shell/prj.conf'
Configuration saved to '/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/.config'
Kconfig header saved to '/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /build/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build
-- west build: building application
[1/192] Preparing syscall dependency handling

[2/192] Generating include/generated/version.h
-- Zephyr version: 3.1.99 (/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr), build: zephyr-v3.1.0-1090-gfa48dbd1dbbf
[182/192] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map 
: && ccache /build/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc   zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  zephy [... cut ...]
/build/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/drivers/flash/libdrivers__flash.a(soc_flash_cc13xx_cc26xx.c.obj): in function `flash_cc13xx_cc26xx_write':
/p/psg/swip/w/boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/drivers/flash/soc_flash_cc13xx_cc26xx.c:203: undefined reference to `NOROM_FlashProgram'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /nfs/png/disks/swuser_work_boonkhai/git/boonkhai_github/build_zephyr/qspi_zephyr_project_rebased/zephyr/build

```

ngboonkhai, 2022-07-07:
> * zephyr-sdk-0.13.2

Hi @cfriedt,

Do you got a chance to try to build the board with
west build -b cc1352r1_launchxl samples/drivers/flash_shell --pristine

And replicate the issue?

cfriedt, 2022-07-07:
Awesome! Thanks for the fix @pepe2k !

ngboonkhai, 2022-07-08:
Thanks for the prompt action @pepe2k